marcone/teslausb #605: Empty folders on samba share

SergioRius, 2021-01-29:
Some folders appear empty on the samba share. If I log into TeslaUsb and go to the corresponding folder in "mutable" it shows links to the files in the snapshot.

[image]
*The highlighted folder is empty.
```
root@teslausb:/mutable/TeslaCam/SavedClips# ll 2020-12-30_16-43-26/
total 49K
drwxr-xr-x 2 root root 3.0K Dec 30 16:55 .
drwxr-xr-x 5 root root 1.0K Jan 26 15:11 ..
lrwxrwxrwx 1 root root  108 Dec 30 16:55 2020-12-30_16-32-26-back.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-32-26-back.mp4
lrwxrwxrwx 1 root root  109 Dec 30 16:55 2020-12-30_16-32-26-front.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-32-26-front.mp4
lrwxrwxrwx 1 root root  117 Dec 30 16:55 2020-12-30_16-32-26-left_repeater.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-32-26-left_repeater.mp4
lrwxrwxrwx 1 root root  118 Dec 30 16:55 2020-12-30_16-32-26-right_repeater.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-32-26-right_repeater.mp4
lrwxrwxrwx 1 root root  108 Dec 30 16:55 2020-12-30_16-33-27-back.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-33-27-back.mp4
lrwxrwxrwx 1 root root  109 Dec 30 16:55 2020-12-30_16-33-27-front.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-33-27-front.mp4
lrwxrwxrwx 1 root root  117 Dec 30 16:55 2020-12-30_16-33-27-left_repeater.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-33-27-left_repeater.mp4
lrwxrwxrwx 1 root root  118 Dec 30 16:55 2020-12-30_16-33-27-right_repeater.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-33-27-right_repeater.mp4
lrwxrwxrwx 1 root root  108 Dec 30 16:55 2020-12-30_16-34-27-back.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-34-27-back.mp4
lrwxrwxrwx 1 root root  109 Dec 30 16:55 2020-12-30_16-34-27-front.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-34-27-front.mp4
lrwxrwxrwx 1 root root  117 Dec 30 16:55 2020-12-30_16-34-27-left_repeater.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-34-27-left_repeater.mp4
lrwxrwxrwx 1 root root  118 Dec 30 16:55 2020-12-30_16-34-27-right_repeater.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-34-27-right_repeater.mp4
lrwxrwxrwx 1 root root  108 Dec 30 16:55 2020-12-30_16-35-27-back.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-35-27-back.mp4
lrwxrwxrwx 1 root root  109 Dec 30 16:55 2020-12-30_16-35-27-front.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-35-27-front.mp4
lrwxrwxrwx 1 root root  117 Dec 30 16:55 2020-12-30_16-35-27-left_repeater.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-35-27-left_repeater.mp4
lrwxrwxrwx 1 root root  118 Dec 30 16:55 2020-12-30_16-35-27-right_repeater.mp4 -> /backingfiles/snapshots/snap-000081/mnt/TeslaCam/SavedClips/2020-12-30_16-43-26/2020-12-30_16-35-27-right_repeater.mp4
...
```
Then there is the first folder in the image, that apparently is corrupted, I suppose because a sudden poweroff in the car.
Contents of the samba share are readonly. But if I change samba config, nothing changes. Is there a way I can clean those things up from a remote computer without logging and rm them? Is it made automatically?

marcone, 2021-01-29:
If you use `ls -lL` instead of `ll` on that directory, what does it say?

SergioRius, 2021-02-01:
Thanks marcone. It's not that I've forgotten to check what you told me. My M3 is in the service center and I forgot the TUSB there.

SergioRius, 2021-02-03:
Well, it seems that I'm late and all the folders that where present in the day had been recycled.
```
root@teslausb:/mutable/TeslaCam/SavedClips# ls -lL 2020-12-30_16-43-26
ls: cannot access '2020-12-30_16-43-26': No such file or directory
root@teslausb:/mutable/TeslaCam/SavedClips# ll
total 9.0K
drwxr-xr-x 4 root root 1.0K Jan 30 01:08 .
drwxr-xr-x 5 root root 1.0K Jan 28 23:33 ..
drwxr-xr-x 2 root root 4.0K Feb  1 14:37 2021-01-26_12-28-36
drwxr-xr-x 2 root root 3.0K Jan 27 09:16 2021-01-26_15-04-35
```

marcone, 2021-02-04:
Teslausb's samba share is read-only. Teslausb manages and removes old entries when it needs to free space.

The reason I asked about `ls -lL` is because that will show you the size of the files. Teslausb will not archive files that are smaller than 100KB, since those are generally not playable, however on older versions of teslausb it would still create the containing folder in the archive, so if you had a Sentry folder full of truncated files, that would result in an empty folder being created in the archive.

closed 2021-02-04 by marcone

THX723, 2023-01-26:
@marcone Sorry to dig this out of the grave. What causes those funky '~' folders? I've been getting a ton of that. Is this an issue with my teslusb? or from the car?

marcone, 2023-01-27:
They are created by the filesystem check that is performed on the disk image. Most likely they are previously-deleted files, i.e. duplicates of files that were already archived.

THX723, 2023-01-27:
> They are created by the filesystem check that is performed on the disk image. Most likely they are previously-deleted files, i.e. duplicates of files that were already archived.

Gotcha. Thanks!