marcone/teslausb #607: Raspberry pi Zero test pads

smemobox, 2021-02-03:
Hi. 
I noticed that your prebuilt image disable the usb test pads on the zero.
Would you know what specifically turns them off?
They work fine on a vanilla raspbian install. Thanks.

marcone, 2021-02-03:
I'm not sure what you mean. As far as I know the Pi Zero has only one USB port, and the test pads provide access to that one port. Teslausb most definitely does not turn off the USB port, since it needs it to communicate with the car.

smemobox, 2021-02-04:
I connected a hub to the data pads. 
Can't see devices I connect to the hub. 
I can if switch to raspbian.

marcone, 2021-02-04:
teslausb acts as a USB drive, and for that it has to switch the USB port from host mode to gadget mode. Hubs only work in host mode.
If you want both host and gadget at the same time, you'll need a Raspberry Pi 4, which can do gadget mode on its USB-C port while also doing host mode on the USB-A ports.

closed 2021-02-04 by marcone