marcone/teslausb #608: "Dashcam USB drive is too small"

rmyadsk, 2021-02-05:
I have two installs on 64GB SDCards, each with 16GB CAM_SIZEs. Both vehicles are running firmware 2020.48.35.5 and both constantly display alert UI_a109 as below. The alert presumably suggests dedicating 32GB or greater to CAM_SIZE, but the documentation states:

> In fact, it is better to use a relatively small size, so that teslausb has space to preserve recordings that are older than 1 hour, which would otherwise be discarded by the car. As an example, if your normal use case is driving to work in the morning, enabling Sentry while parked, and going back home in the evening, with the car reporting up to 10 Sentry events, then 16 GB is a good size to use.

Has this allocation been reconsidered? My drive sizing is below. 

>pi@teslausb-blue:~ $ df -h
Filesystem         Size  Used Avail Use% Mounted on
/dev/root          1.8G  1.2G  549M  69% /
devtmpfs           184M     0  184M   0% /dev
tmpfs              216M     0  216M   0% /dev/shm
tmpfs              216M  5.8M  211M   3% /run
tmpfs              5.0M     0  5.0M   0% /run/lock
tmpfs              216M     0  216M   0% /sys/fs/cgroup
tmpfs              216M     0  216M   0% /var/tmp
tmpfs              216M     0  216M   0% /var/spool
tmpfs              216M  8.0K  216M   1% /var/log
tmpfs              216M  4.0K  216M   1% /tmp
/dev/mmcblk0p1     253M   47M  206M  19% /boot
/dev/mmcblk0p4      93M  4.0M   82M   5% /mutable
/dev/mmcblk0p3      56G   43G   14G  77% /backingfiles
tmpfs               44M     0   44M   0% /run/user/1000

[image]

marcone, 2021-02-05:
yes, the documentation needs to be updated. The sample config file has the right info.

marcone, 2021-02-09:
Updated the readme

closed 2021-02-09 by marcone