marcone/teslausb #610: how to access snapshots

radhoo2k10, 2021-02-12:
I have tried to read me up on the issue. As far as I can tell the snapshots is accessible via Samba. And the snapshots is where the 1 hour + clips are located.
So I enter the ip-adress and I see the TeslaCam folder and enter user and pass. There is 3 folders. RecentClips, SavedClips and SentryClips. 
Is the snapshots in the RecentClips?

I have this in the setupfile: 

# uncomment to enable a Samba/CIFS server that makes the recordings
# available as a network share
export SAMBA_ENABLED=true
# uncomment to enable guest access to the Samba server
#export SAMBA_GUEST=true

radhoo2k10, 2021-02-12:
`# uncomment to enable a Samba/CIFS server that makes the recordings
# available as a network share
export SAMBA_ENABLED=true
# uncomment to enable guest access to the Samba server
#export SAMBA_GUEST=true`

marcone, 2021-02-12:
What you see in Samba is a merged view of all the snapshots

closed 2021-02-12 by marcone