marcone/teslausb #611: Way to detect difference between driving (RecentClips) and parked (SentryClips)?

Jughead135, 2021-02-18:
I find my TeslaUSB rig is continuously recording whenever the car is awake--which means any time Sentry Mode is enabled, the "dashcam" is constantly working (and overwriting previous vids).  Sentry Mode recordings are still copied into the SentryClips folder, so that bit is normal.

I don't know what may have changed (since that didn't seem to be happening before--perhaps I somehow have missed it??).  Is there a way to get TeslaUSB to recognize that the car is parked and stop feeding the incoming vid to the RecentClips folder?  My card holds a bit over 20 hours of video--but almost all of that will get taken up with just two days of being out & about.  I'd find it a much more useful archive to strictly record driving time (with SavedClips & SentryClips being saved separately as they are now).

Do I need to choose between using Sentry Mode and keeping only driving vids in RecentClips?  Or is there a way to make TeslaUSB distinguish driving time vs. Senty Mode time?  I'm thinking along the lines that if it can tell that Sentry Mode is active via the app, then it could stop archiving the cameras?

marcone, 2021-02-18:
> I find my TeslaUSB rig is continuously recording whenever the car is awake

This is normal, expected, and how it has always worked.

> Is there a way to get TeslaUSB to recognize that the car is parked and stop feeding the incoming vid to the RecentClips folder?

TeslaUSB doesn't feed video anywhere. TeslaUSB simply presents the car with a USB drive, and the car records whatever it wants to it.

closed 2021-02-18 by marcone

Jughead135, 2021-02-18:
Perhaps I’ve used the wrong terminology (likely).

It’s not the recording per se that I’d like to stop (since, as you say, it’s the car doing that to the “flash drive” it sees in the USB).

Rather, it’s the archiving of that recording—the copying of the “last hour” that the car stores into the larger storage—which I’d like to suspend when the car is parked.

marcone, 2021-02-18:
It doesn't copy the recordings, it takes an hourly snapshot of the drive image. In order to stop taking those snapshots while parked, TeslaUSB would need to know that the car is parked. I can't think of a way to do that except by using the Tesla API, and that will only work if you're connected to wifi.

Jughead135, 2021-02-18:
The API angle was what I was musing about.  I wasn’t thinking through the Wi-Fi angle for the Pi to be able to talk to the API.  Duh. 

As I’ve fessed up before, your program is well beyond my meager coding abilities (or even understanding, in many cases).  My thoughts are along the lines of (assuming it’s possible):
- take snapshot (as done now)
- compare to previous (as now)
- discard if same (as now)
- discard if Sentry enabled (new)

Probably would need a flag for the “first time” it sees Sentry enabled:  keep the first snapshot with the “partial hour” of driving up until parking (Sentry enabled), then discard all future ones until Sentry no longer active.  Reset the flag & start over. 

Anyway, I see your point re the API not being accessible.  I suppose if one has a hotspot in the car, that could be surmountable, but that’s not me (yet?).

Thanks for the discussion.  As always, thanks for the great project!

Jughead135, 2021-03-17:
> In order to stop taking those snapshots while parked, TeslaUSB would need to know that the car is parked. I can't think of a way to do that except by using the Tesla API, and that will only work if you're connected to wifi.

I've been chewing on this....

Instead of directly detecting the car is parked via the API (requiring a wifi connection), would it be reasonable to detect instead whether Sentry is recording?  Say, a test at some interval to see if there are any new files in the SentryClips directory?  Since Sentry saves a minute's worth of video (x4 to cover each camera), if there's something in there less than two minutes old, then Sentry is doing its thing (so, car is parked, stop archiving the RecentClips snapshots).  Once the latest SentryClips file is more than two minutes old, Sentry has stopped (so, assume car is active/driving, resume archiving).  If Sentry is turned off, then the car won't be recording anything (and the MCU shuts down), so the current logic would be unchanged.

Not perfect (I'm sure there are failure modes where Sentry stops recording, the MCU stays on, and this logic would conclude "car is driving" & start archiving), but it seems like it could work...?

marcone, 2021-03-17:
> Since Sentry saves a minute's worth of video (x4 to cover each camera), if there's something in there less than two minutes old, then Sentry is doing its thing (so, car is parked, stop archiving the [...]

That's not quite how it works. Whenever the car is "on" (whether because you're driving or because Sentry mode is on), it will write 1-minute clips to RecentClips, every minute. When a Sentry event is detected, it moves the last 10 minutes worth of recordings from RecentClips to a folder under SentryClips.
So from TeslaUSB's point of view, there is no difference between "just driving around" and "parked with Sentry mode on, but no Sentry event has been triggered yet".

Jughead135, 2021-03-17:
> from TeslaUSB's point of view, there is no difference between "just driving around" and "parked with Sentry mode on, but no Sentry event has been triggered yet".

Ah.  Knew I was probably missing something....

Thanks, as always!