marcone/teslausb #613: Where are permissions assigned when the car saves a file?

Jughead135, 2021-02-25:
This is a bit of a follow-up to #480 , wherein I was looking for how to write via SMB to the shared `TeslaCam` directory on the Pi.  Unsurprisingly, the issue there was my own ignorance, and @marcone set me straight (thanks again!).

I finally figured out that at least part of my problem was the permissions on the files themselves, vs. the SMB permissions.  The normal permissions I see (at least in the `SentryClips` directory) are:
Directory level:  `drwxr-xr-x`
Files (links):  `lrwxrwxrwx`
Files (actual):  `-rwxrwxrwx`

For all of those, `root` is always the owner and `root` is always the group.

What I'd **like** to do is make the directories (`TeslaCam` & all subs) be `drwxrwxr-x` and change the group to `pi`.  That's easy enough to do via chmod & chgrp, and I find it works as I'd expect once that's done (i.e., I can write or delete files over the SMB connection).  However, any new files recorded after the chmod/chgrp revert to the defaults, including the no-write privs for the group at the directory level, and all new files are group `root`.

Is there a way to change the way the files--specifically the directories--get created with regard to their permissions & group?  Is this done by the car or by TeslaUSB?

WHY do I want to do this?  Nothing critical--I'm playing around with different archiving techniques & trying to see just how integrated I can get things.  Also trying to learn a bit of Linux while I'm at it (I'm a noob, if that's not obvious).

As a workaround, I found that adding user `pi` to group `root` accomplishes what I want--but, it seems sloppy to do it that way (effectively a global add, when all I want user `pi` to have write privs in `TeslaCam`).  I'm not terribly worried about it, since the car is the only "user" of the TeslaUSB and security of my dashcam footage isn't a big concern--but, again, I'm trying to learn here, so I want to do it "right"....

marcone, 2021-02-25:
You probably need to look at smb.conf options like "force group", "force user", "create mask", etc.

marcone, 2021-02-25:
Though with regard to the question you ask in the title: "Where are permissions assigned when the car saves a file?"
That's entirely up to the car. Teslausb only present a block device to the car. Samba is not involved in that in any way.

marcone, 2021-02-25:
And one more comment: the permissions you see in the SentryClips folder (assuming you're looking at /mnt/cam/TeslaCam/SentryClips) are actually DOS permissions mapped to unix permissions, since the CAM drive is a FAT32 filesystem mounted on linux. You should be able to change the owner and permissions however you want with the appropriate mount options in /etc/fstab, e.g. the uid/gid/umask/dmask/fmask options. See `man mount`

Jughead135, 2021-02-25:
I was afraid that’s what you were going to say re the car assigning permissions.  I guess I’d hoped that at some point in the process of going from the symbolic linked file on the SD card to the “flash drive” the car sees was imposing some sort of order on it (and would be tweakable).

NB, I’m looking at the permissions via the Linux command line (over SSH), not through SMB.  The directories created on the shared TeslaCam directory are all owned by root, in the root group, with no group or other write prices.

marcone, 2021-02-25:
OK, and by "shared TeslaCam directory" you mean /mutable/TeslaCam, which is also exported via Samba?
The car doesn't actually write to that. The car writes to the "/backingfiles/cam_disk.bin" disk image which is mounted on /mnt/cam on the Pi during archiving.
/mutable/TeslaCam contains symbolic links to snapshots taken of cam_disk.bin. Those symbolic links are owned by root, but Samba on the Pi is configured to follow those links to their targets, so the owner of the links shouldn't really matter. The snapshots containing the link targets are mounted as needed by the automounter, using the same options as those specified for cam_disk.bin in /etc/fstab. So if you set different user/group/permissions for cam_disk.bin in /etc/fstab, that will affect the snapshots too (you'll have to `rm /backingfiles/snapshots/*/*.opts` after changing fstab, since teslausb caches the mount options). It won't change the owner or permissions of the symbolic links though. For that, you'd have to modify `/root/bin/make_snapshot.sh` to chmod/chown as needed.
 
It's not clear to me what you're actually trying to achieve, but it sounds like you want to be able to write to or delete from the snapshots. Note that that will probably not achieve what you want this way. For one thing, a given recording may exist in more than one snapshot, but the symbolic link of course only points to one of them. The data for such recordings will be shared between snapshots, so there's no space overhead, and you won't free up any space by deleting the target of the symbolic link, since the other copy/copies of the recording will continue to exist in other snapshots. You will actually use *more* space, since deleting a file means it has to write to the disk image, which because of copy-on-write semantics means it will use extra space. In order to truly free up space you would have to delete all copies of the file in all snapshots, and them run fstrim on all the mounted snapshots.

Jughead135, 2021-02-25:
> OK, and by "shared TeslaCam directory" you mean /mutable/TeslaCam, which is also exported via Samba?

Yes.  Sorry for the imprecise phrasing, my confusion shows through again.

> The car doesn't actually write to that. The car writes to the "/backingfiles/cam_disk.bin" disk image which is mounted on /mnt/cam on the Pi during archiving.

OK, this is apparently the source of my confusion!  You're saying `/mutable/TeslaCam` is **not** the FAT32-formatted "flash drive" the car "sees"?  I clearly don't understand something here.  AIUI (and I think you've emphasized), the car only understands a FAT32-formatted drive connected via USB (so, that's what TeslaUSB makes it think it's looking at).  On that USB drive, it uses a folder named "TeslaCam," with "RecentClips," "SavedClips," and "SentryClips" sub-folders.  If the car is actually writing to the disk image, does that imply that TeslaUSB is intercepting a write intended for a FAT32 folder and putting it there?  Or are you saying the "real" FAT32 folder lives in the disk image (which is what the car sees), and `/mutable/TeslaCam` is just a copy (links) to give SMB something to look at (but is **not**, in fact, the FAT32 share that the car uses)?

> /mutable/TeslaCam contains symbolic links to snapshots taken of cam_disk.bin. Those symbolic links are owned by root, but Samba on the Pi is configured to follow those links to their targets, so the [...]

This leads me to think my last statement is what you've been trying to get through my thick skull all along:  the "TeslaCam" folder I can see via SMB is the same directory I see in `/mutable/TeslaCam`...  but it is **not** the "/TeslaCam" on the "USB drive" that the car sees.  Assuming I've finally got that right, that explains where I was lost in all this..!

marcone, 2021-02-25:
> OK, this is apparently the source of my confusion! You're saying `/mutable/TeslaCam` is **not** the FAT32-formatted "flash drive" the car "sees"?

Correct.

> If the car is actually writing to the disk image, does that imply that TeslaUSB is intercepting a write intended for a FAT32 folder and putting it there?

No, teslausb doesn't intercept anything. It publishes the disk image as a USB drive. Teslausb doesn't know anything about the writes that the car does, that's all handled by the USB gadget driver. Teslausb doesn't see the result of those writes until it mounts the diskimage, or a snapshot of the diskimage.

> Or are you saying the "real" FAT32 folder lives in the disk image (which is what the car sees), and `/mutable/TeslaCam` is just a copy (links) to give SMB something to look at (but is **not**, in fact, the FAT32 share that the car uses)?

Correct. The links in /mutable only point to the snapshots, not to the "main" disk image. The main disk image is disconnected from the car and then mounted on the Pi during archiving, after which it is connected to the car again. The snapshots are both used to keep a history of what was recorded, and to allow seeing the current recordings (you don't want to mount the main disk image on the Pi while it is also connected to the car, so Teslausb creates a snapshot and then mounts that for access by Samba)

Jughead135, 2021-02-26:
OK, thank you!  As usual, you've taken a lot of your time to describe something I'm sure you consider basic--that's not really in your job description.  I appreciate your help (along with a great project!).

That really explains a lot re why I couldn't get it truly "working."  For the types of things I was trying to play with re archiving, it's simply not manageable (at least with my skill set) to achieve.  If ever I need to play with this, the workaround of setting user `pi` to group `root` will get the job done.

Seriously, thank you for your patience & help.  Cheers!

closed 2021-02-26 by marcone