marcone/teslausb #614: CAM drive showing but no Music?

nickjupp, 2021-02-26:
Hi, I have a new clean install
I can see and access the CAM drive from my macbook, but there is no MUSIC drive

running diagnostics I get 
```
lun0 connected, from file /backingfiles/cam_disk.bin
1 snapshots mounted
USB state (23.9 seconds ago): connected to host and host mounted drive(s)
====== disk / images ======
Model: SD SC64G (sd/mmc)
Disk /dev/mmcblk0: 63.9GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      273MB   2181MB  1908MB  primary  ext4
 3      2181MB  63.8GB  61.6GB  primary  xfs
 4      63.8GB  63.9GB  105MB   primary  ext4
cam disk image has 11 extents
Model:  (file)
Disk /backingfiles/cam_disk.bin: 32.2GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  32.2GB  32.2GB  primary  fat32        lba
**no music disk image found**
```

snip from config - I have tried without quotes (6G rather than "6G") as well as percentage

```
#   * MUSIC_SIZE should be large enough to hold your music library, obviously.
#   * If no music size is specified, no music drive will be created.
export CAM_SIZE="30G"
export MUSIC_SIZE="6G"

```

marcone, 2021-02-26:
Did you initially not specify a music size and then add one later and rerun setup?

marcone, 2021-02-26:
Per discussion in the Discord, this happened because no music size was specified initially, and when rerunning setup teslausb doesn't update the image(s). I will either put a note that effect in the config file, or make setup more robust.

nickjupp, 2021-02-26:
Thanks for your help. can confirm m`usic_disk.bin` is there now after deleting `cam_disk.bin `and re-running setup

jhtx2023, 2023-01-07:
Thanks for this. I was wondering how to resize the music_disk.bin after I initially sized it as 4G and now would like to expand it to 16G.

chrisalmy, 2023-06-29:
To piggyback off this thread, I'm experiencing a similar issue. Music drive just won't be created (missing music_disk.bin). I've even deleted the cam_disk.bin and reran the setup, but I get the same results. Setup logs show it sees the music size but won't create:
```
Thu 29 Jun 14:38:17 PDT 2023 : create-backingfiles: starting
Thu 29 Jun 14:38:17 PDT 2023 : create-backingfiles: cam: 52G, music: 4G, boombox: 0 mountpoint: /backingfiles, exfat: false
Thu 29 Jun 14:38:18 PDT 2023 : create-backingfiles: Allocating 48942552K for /backingfiles/cam_disk.bin...
Thu 29 Jun 14:38:19 PDT 2023 : create-backingfiles: Creating filesystem with label 'CAM'
Thu 29 Jun 14:38:21 PDT 2023 : create-backingfiles: created camera backing file
Thu 29 Jun 14:38:22 PDT 2023 : create-backingfiles: `done`
```
Here is my config file (all commented out lines have been removed):
```
export ARCHIVE_SYSTEM=cifs
export ARCHIVE_SERVER=192.168.1.x
export SHARE_NAME=NAS2/Cam/Tesla
export SHARE_USER=redacted
export SHARE_PASSWORD=redacted
export CIFS_VERSION="3.0"
export CAM_SIZE=52G
export MUSIC_SIZE=4G
export MUSIC_SHARE_NAME=Backup/CarMP3/Tesla
export SSID="redacted"
export WIFIPASS="redacted"
export TIME_ZONE="America/Los_Angeles"
export TESLAUSB_HOSTNAME=teslausb-ModelY
```

Any idea where to start? It's a 64Gig SD card, so there should be plenty of space for both partitions. The Cam drive otherwise works great in the car. I simply noticed shortly after installing my music wasn't there. Took me by surprise as I've been running another Pi Zero W with this system for over 4 years now in my Model 3 with no issues.

marcone, 2023-07-03:
@chrisalmy 64 GB is the bare minimum that will work. I highly recommend using a 128 or 256 GB sd card instead.
I suspect the problem in your case is that you specified `CAM_SIZE=52G`, and after accounting for OS overhead that doesn't leave enough space for the 4GB music drive. Try reducing CAM_SIZE to 40G or so (you will need to reflash the sd card and start over, or delete the existing /backingfiles/cam_disk.bin and rerun setup)

chrisalmy, 2023-07-05:
@marcone Yep, that appears to be the issue. I would have thought 8Gigs of space would be more then enough for the OS, but I was wrong. I thought my existing 4+ year old TeslaUSB setup with 64Gig SD card also had about 52Gig of CAM partition, but I just pulled it up and sure enough, it was 42Gig. I reduced the CAM size and I now have a music folder!

Now on to the next issue. While on a short road trip a few days ago, the drive stopped working a few times and needed a reboot. The car also said the drive was too slow (a message I haven't seen in years). This could be because my new Model Y is a AP4 system which saves about 180MB of data per minute versus my AP3 which does about 105MB per minute. I'll keep monitoring the setup, as for all I know the poor Zero W may not be able to keep up. If I continue to have issues, I have a spare Pi 4 I can try instead.

oopskoule, 2023-10-22:
I am getting same log entry:  Music archive not configured or unreachable

teslausb_setup_variables.conf:

export ARCHIVE_SYSTEM=rclone
export RCLONE_DRIVE="backup"
export RCLONE_PATH="backup"
...
export MUSIC_SHARE_NAME="Music"
...

rclone lsd backup:
          -1 2023-05-05 18:31:15        -1 Music
          -1 2023-10-22 12:59:55        -1 backup
rclone lsd backup:Music
          -1 2023-05-05 18:31:15        -1 Best_of
          -1 2023-03-31 13:43:05        -1 pop
/backingfiles# ls -lh
total 21G
-rw-r--r--  1 root root 20G Oct 22 22:22 cam_disk.bin
-rw-r--r--  1 root root 20G Oct 22 21:59 music_disk.bin
drwxr-xr-x 10 root root 158 Oct 22 22:01 snapshots

I tried to define export MUSIC_SHARE_NAME with "Music", "backup:Music"  etc. without success.
cam is working fine, how to define music using rclone? 
(I assume its a syntax issue...)
thanks a lot!

marcone, 2023-10-22:
> I am getting same log entry: Music archive not configured or unreachable

Music *syncing* is only supported for CIFS, not for rsync and rclone. You can still have a music drive, you'll just have to copy to it manually.

marcone, 2024-10-23:
> Per discussion in the Discord, this happened because no music size was specified initially, and when rerunning setup teslausb doesn't update the image(s). I will either put a note to that effect in the config file, or make setup more robust.

The current code now lets you add/remove drives after the fact by changing the corresponding sizes in the config file.

closed 2024-10-23 by marcone