marcone/teslausb #615: Token refresh failed: %TeslaApi.Error

rmyadsk, 2021-03-03:
Seems persistent, but may not impact data collection.

How can I best help troubleshoot further?

Here is my `docker-compose telsamate logs`, reasonably truncated as it is most repetitive:

> teslamate_1  | 2021-03-02 16:10:02.039 car_id=2 [info] Derived efficiency factor: 145.41000000000003 Wh/km (18x confirmed)
teslamate_1  | 2021-03-02 16:10:02.040 car_id=2 [info] Charging / Disconnected / 41.77 kWh – 413 min
teslamate_1  | 2021-03-02 16:10:02.044 car_id=2 [info] Start / :online
teslamate_1  | 2021-03-02 16:10:02.089 car_id=2 [info] Connecting ...
teslamate_1  | 2021-03-02 16:10:31.685 car_id=2 [info] Driving / Start
teslamate_1  | 2021-03-02 16:24:50.153 [info] Refreshing access token ...
teslamate_1  | 2021-03-02 16:24:50.360 [error] POST https://auth.tesla.com/oauth2/v3/token -> 400 (205.971 ms)
teslamate_1  | 2021-03-02 16:24:50.364 [warn] Token refresh failed: %TeslaApi.Error{
teslamate_1  |   env: %Tesla.Env{
teslamate_1  |     __client__: %Tesla.Client{adapter: nil, fun: nil, post: [], pre: []},
teslamate_1  |     __module__: TeslaApi.Auth,
teslamate_1  |     body: %{
teslamate_1  |       "error" => "server_error",
teslamate_1  |       "error_description" => "Internal server error",
teslamate_1  |       "error_uri" => "https://auth.tesla.com/error/reference/460053ab-2601-4a0a-8d77-a461cde1785a-1614720290307"
teslamate_1  |     },
teslamate_1  |     headers: [
teslamate_1  |       {"server", "nginx"},
teslamate_1  |       {"content-type", "application/json"},
teslamate_1  |       {"x-dns-prefetch-control", "off"},
teslamate_1  |       {"x-frame-options", "DENY"},
teslamate_1  |       {"strict-transport-security", "max-age=15552000; includeSubDomains"},
teslamate_1  |       {"x-download-options", "noopen"},
teslamate_1  |       {"x-content-type-options", "nosniff"},
teslamate_1  |       {"x-xss-protection", "1; mode=block"},
teslamate_1  |       {"x-request-id", "460053ab-2601-4a0a-8d77-a461cde1785a"},
teslamate_1  |       {"x-correlation-id", "460053ab-2601-4a0a-8d77-a461cde1785a"},
teslamate_1  |       {"access-control-allow-origin", "*"},
teslamate_1  |       {"x-response-time", "3.318ms"},
teslamate_1  |       {"date", "Tue, 02 Mar 2021 21:24:50 GMT"},
teslamate_1  |       {"connection", "close"}
teslamate_1  |     ],
teslamate_1  |     method: :post,
teslamate_1  |     opts: [],
teslamate_1  |     query: [],
teslamate_1  |     status: 400,
teslamate_1  |     url: "https://auth.tesla.com/oauth2/v3/token"
teslamate_1  |   },
teslamate_1  |   message: nil,
teslamate_1  |   reason: :token_refresh
teslamate_1  | }
teslamate_1  | 2021-03-02 16:24:50.364 [warn] Retrying in 1 hour...
teslamate_1  | 2021-03-02 16:40:23.454 car_id=2 [info] Driving / Ended / 32 km – 29 min
teslamate_1  | 2021-03-02 16:40:23.458 car_id=2 [info] Start / :online
teslamate_1  | 2021-03-02 16:43:41.224 car_id=2 [info] Suspending logging
teslamate_1  | 2021-03-02 17:11:05.871 [info] GET /
teslamate_1  | 2021-03-02 17:11:05.889 [info] Sent 200 in 17ms
teslamate_1  | 2021-03-02 17:12:46.948 car_id=2 [info] Resuming logging
teslamate_1  | 2021-03-02 17:12:49.525 car_id=2 [info] Received stream data: :inactive
teslamate_1  | 2021-03-02 17:12:57.263 [info] GET https://owner-api.teslamotors.com/api/1/vehicles/40149755796008394/vehicle_data -> 408 (9313.108 ms)
teslamate_1  | 2021-03-02 17:13:07.573 car_id=2 [info] Start / :offline
teslamate_1  | 2021-03-02 17:13:07.604 car_id=2 [info] Disconnecting ...
teslamate_1  | 2021-03-02 17:13:20.145 [warn] GET https://owner-api.teslamotors.com/api/1/vehicles/40149755796008394/vehicle_data -> 540 (2410.264 ms)
teslamate_1  | 2021-03-02 17:13:20.146 [warn] TeslaApi.Error / https://mothership-api.vn.teslamotors.com:443/vehicles/174286848/user_command/vehicle_data => vehicle error: it is online but not responsive
teslamate_1  | 2021-03-02 17:13:20.146 car_id=2 [error] Error / :unknown
teslamate_1  | 2021-03-02 17:13:50.609 car_id=2 [info] Start / :online
teslamate_1  | 2021-03-02 17:13:50.617 car_id=2 [info] Logged missing software udpate: 2021.4.11 8af6a4f84c84
teslamate_1  | 2021-03-02 17:13:50.686 car_id=2 [info] Connecting ...
teslamate_1  | 2021-03-02 17:16:53.375 car_id=2 [info] Suspending logging
teslamate_1  | 2021-03-02 17:24:50.365 [info] Refreshing access token ...
teslamate_1  | 2021-03-02 17:24:50.646 [error] POST https://auth.tesla.com/oauth2/v3/token -> 400 (279.988 ms)
teslamate_1  | 2021-03-02 17:24:50.650 [warn] Token refresh failed: %TeslaApi.Error{
teslamate_1  |   env: %Tesla.Env{
teslamate_1  |     __client__: %Tesla.Client{adapter: nil, fun: nil, post: [], pre: []},
[... 338 lines cut ...]
teslamate_1  |       {"x-xss-protection", "1; mode=block"},
teslamate_1  |       {"x-request-id", "0626697f-40f3-4b4c-9734-e14b63830664"},
teslamate_1  |       {"x-correlation-id", "0626697f-40f3-4b4c-9734-e14b63830664"},
teslamate_1  |       {"access-control-allow-origin", "*"},
teslamate_1  |       {"x-response-time", "7.046ms"},
teslamate_1  |       {"date", "Wed, 03 Mar 2021 05:24:52 GMT"},
teslamate_1  |       {"connection", "close"}
teslamate_1  |     ],
teslamate_1  |     method: :post,
teslamate_1  |     opts: [],
teslamate_1  |     query: [],
teslamate_1  |     status: 400,
teslamate_1  |     url: "https://auth.tesla.com/oauth2/v3/token"
teslamate_1  |   },
teslamate_1  |   message: nil,
teslamate_1  |   reason: :token_refresh
teslamate_1  | }
teslamate_1  | 2021-03-03 00:24:52.314 [warn] Retrying in 1 hour...
teslamate_1  | 2021-03-03 00:26:53.863 [info] GET /
teslamate_1  | 2021-03-03 00:26:53.882 [info] Sent 200 in 19ms
teslamate_1  | 2021-03-03 00:26:54.649 [info] GET /
teslamate_1  | 2021-03-03 00:26:54.666 [info] Sent 200 in 17ms
teslamate_1  | 2021-03-03 00:53:46.450 [info] GET https://owner-api.teslamotors.com/api/1/vehicles/5853617637379554/vehicle_data -> 408 (9137.932 ms)
teslamate_1  | 2021-03-03 00:53:46.450 car_id=1 [error] Error / :vehicle_unavailable
teslamate_1  | 2021-03-03 00:54:22.024 car_id=1 [info] Vehicle is still online. Falling asleep for: 33 min

raushel, 2021-03-03:
Err, this is the TeslaUSB github, not TeslaMate, but basically you need to update your docker for TeslaMate for the latest API Auth changes from awhile back and then Log Out / Log Back in.

rmyadsk, 2021-03-03:
D'oh! My mistake. Please close with indignity pointed here: I can confirm the teslausb-20201211 image works terrificly absent a `apt-get update; apt-get upgrade -y; apt autoremove -y` for Raspian things. Thanks @raushel!

closed 2021-03-03 by marcone