marcone/teslausb #617: rsync archival method does not seem to follow export trigger_file*

rmyadsk, 2021-03-05:
After several months as a Mac user (as a person hosting a Mac archival server, to two teslausb clients) suffering along with Apple's implementation of SMB/CIFS/whatevs, and ultimately finding it broken with a recent Big Sur release, I decided to implement the rsync archival method instead.

Great success. Though the ARCHIVE_UPLOADED files are no longer to be seen.

I worked from the instructions at https://github.com/cimryan/teslausb/blob/master/doc/SetupRSync.md but had preloaded my system to be ready for it. Specifically, in lieu of a CIFS section, the first lines of my config file are:

export ARCHIVE_SYSTEM=rsync
export RSYNC_USER=piuser
export RSYNC_SERVER=0.0.420.69
export RSYNC_PATH=/Volumes/To/The/Moon/

Anyway, I can assure you that rsync archiving succeeds from first run, but the associated below does not: 

> export trigger_file_saved=ARCHIVE_UPLOADED
> export trigger_file_sentry=ARCHIVE_UPLOADED
> export trigger_file_any=ARCHIVE_UPLOADED

Can we make teslausb write these files upon successful rsync write?

marcone, 2021-03-05:
The best way to do this is probably to have archiveloop create the trigger files in a directory under /tmp, and then simply have archive-clips.sh copy them over. Then it would work with each archive method.

closed 2021-03-07 by marcone

rmyadsk, 2021-03-09:
Updated my installation and observed the desired file present following an archive, thanks!

commit 64956ee4 (2021-03-07): Rework archive triggers

Make the archive trigger files part of the archive operation
itself, so they work with cifs, rsync and rclone.
Fixes #617
commit 63ade8e5 (2021-03-09): Rework archive triggers

Make the archive trigger files part of the archive operation
itself, so they work with cifs, rsync and rclone.
Fixes #617