marcone/teslausb #618: Use 4G modem instead WiFi for uploading videos

azzimuth, 2021-03-18:
As far as I understood the documentation, the archiving of the Sentry clips is done via Wi-Fi. I wonder if it is possible to do that using a 4G modem. I have one with unlimited data plan and at the same time the car is parked in the area with no Wi-Fi coverage, so this would be a perfect solution for me (and I guess many others).

marcone, 2021-03-18:
Teslausb doesn't care what network connection you have, just that there is one. If you configure your Pi to make use of your 4G modem, it should just work.

closed 2021-03-18 by marcone

rmyadsk, 2021-03-18:
Caveat: there must be appropriate port forwarding open on the server (archive) side. The 4G modem will get your Pi on the Web, but you will need to open one or more inbound ports on your home router and direct them to your archive box via port forwarding. For example, tcp/22 for rsync via SSH.