marcone/teslausb #619: teslausb_setup_variables.conf file getting deleted upon boot

doug-u-agrave, 2021-03-22:
I've reimaged teslausb microsd a couple times now.  I have copied in a saved copy of teslausb_setup_variables.conf file and then booted the pi.  I wait at least 10 minutes but it never shows up on my network.  Then when I remount the microsd the teslausb_setup_variables.conf file is gone!  In the teslausb-headless-setup.log is an endless series of:

Sat 12 Dec 05:19:27 GMT 2020 : Detecting whether to update wpa_supplicant.conf
Sat 12 Dec 05:19:27 GMT 2020 : Grabbing main setup file.
Sat 12 Dec 05:19:37 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:19:51 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:20:04 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:20:17 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:20:31 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:20:44 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:20:57 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:21:11 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:21:24 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:21:37 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:21:50 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:22:04 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:22:17 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:22:30 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:22:44 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:22:57 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:23:10 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:23:23 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:23:37 GMT 2020 : get_script failed, retrying
Sat 12 Dec 05:23:50 GMT 2020 : get_script failed, retrying

marcone, 2021-03-22:
The setup process moves the .conf from /boot to /root, so that part is normal and expected.
For the "get_script failed" problem, double check that you've specified the correct wifi info in the .conf file.
You can also connect the Pi to a Windows PC, in which case it should be able to use a USB network connection to the PC, allowing you to ssh in from the PC and see in more detail what's going on.

closed 2021-03-22 by marcone

doug-u-agrave, 2021-03-22:
Thanks Marcone.  Still having major issues
First, when I inspect the pi I don't see a /root.  Is it on a different partition?  I see another partition ("G:" where /boot is "F:") but windows can access it.  Perhaps its a filesystem windows doesn't understand like ext3?
I triple checked the wifi info.  I have everything correct.
I tried ssh over usb using the directions https://desertbot.io/blog/headless-pi-zero-ssh-access-over-usb-windows but it wont connect.
Any help would be appreciated.

marcone, 2021-03-22:
/root is that G: drive, an ext4 partition that Windows doesn't understand.

You shouldn't need to do steps 1-7 on the page you linked. Just go straight to step 8 (or 9, if you already have Putty installed) and ssh to teslausb.local (not raspberrypi.local).

For wifi, make sure you're not trying to use a 5GHz network, since the Raspberry Pi 0 only supports 2.4 GHz wifi.

doug-u-agrave, 2021-03-22:
Thanks so much for your help and creating this branch.  I got it working.  It was a cable problem.  Was booting with power to the usb port not the power port.  Complete re-image and switched to power cord that came with the pi direct to power port.  Now i am ssh'd in.  THANKS

marcone, 2021-03-22:
Interesting. The Pi can be powered from either port, so it shouldn't really matter.