marcone/teslausb #621: Change timezone

doug-u-agrave, 2021-03-29:
1. Thanks Marcone
2. I was having issues early on that I thought might be caused by timezone setting but it turned out to be a different issue.  Now its working fine but has the wrong timezone.  Is there an easy way to change the timezone?

tried 
  sudo timedatectl set-timezone America/Los_Angeles 
but got
 Failed to set time zone: Read-only file system

Was thinking is could edit teslausb_setup_variables.conf over in /root and run setup_teslausb but /root too is read-only

Is there an easy way or do I need to update teslausb_setup_variables.conf and completely reflash

rmyadsk, 2021-03-29:
To make any changes you'll first need to set the filesystem to read/write:

> sudo sh
> /root/bin/remountfs_rw

closed 2021-03-29 by marcone

doug-u-agrave, 2021-03-30:
worked perfectly!  thanks for babysitting this newbie