marcone/teslausb #623: Error during archiving. Archived 0 event files in 8s

JohPePe, 2021-04-01:
TeslaUsb has lately started to fail in copying anything basically.

TeslaUsbBot, [Apr 1, 2021 at 19:19]
teslausb: Archiving 3 event folder(s) with 41 file(s) and 0 track mode file(s) starting at Thu  1 Apr 19:19:43 CEST 2021

TeslaUsbBot, [Apr 1, 2021 at 19:19]
teslausb: Error during archiving. Archived 0 event files in 8s

is there any way to get a more detailed log of what’s happening?

miles267, 2021-04-01:
Yes.  I've been experiencing this more and more recently and wouldn't be aware if it weren't for PushOver alerts.  There's no obvious indication of these failures without them.  I tend to find that they occur more frequently with > 200 files to sync.  Since connectivity isn't the issue in my home environment, I've yet to understand what it is about the process that is causing the failure.  Also, I've discovered a pattern.  1.) I'm using the archive of /RecentClips option and 2.) the /RecentClips portion of the sync session always completes in its entirety, but the moment it progresses to the next phase of sync (/SentryClips), the failure occurs.  This results in empty dated subfolders in the /RecentClips folder and the /teslausbtmp folder remains, is empty, and doesn't get deleted unless a 100% successful sync session completes.

marcone, 2021-04-01:
when you see a failure, log in to the Pi and check /tmp/rsynclog

JohPePe, 2021-04-01:
Hmm, tried that. I have no idea what’s going on

***@***.***:/tmp $ cat rsynclog
building file list ... done
SavedClips/2021-04-01_19-11-40/2021-04-01_19-11-09-front.mp4
SavedClips/2021-04-01_19-11-40/2021-04-01_19-11-09-left_repeater.mp4
***@***.***:/tmp $ cat rsynclog
building file list ... done

Number of files: 0
Number of created files: 0
Number of deleted files: 0
Number of regular files transferred: 0
Total file size: 0 bytes
Total transferred file size: 0 bytes
Literal data: 0 bytes
Matched data: 0 bytes
File list size: 0
File list generation time: 0.001 seconds
File list transfer time: 0.000 seconds
Total bytes sent: 29
Total bytes received: 12

sent 29 bytes  received 12 bytes  82.00 bytes/sec
total size is 0  speedup is 0.00

He’s building the list, then it’s empty?

/Johan

marcone, 2021-04-01:
Did you get an error from pushover (or whichever push notification system that was) that time though?

JohPePe, 2021-04-01:
This time I got the following, strangely enough!

TeslaUsbBot, [Apr 1, 2021 at 20:41]
teslausb: Archiving 3 event folder(s) with 41 file(s) and 0 track mode file(s) starting at Thu  1 Apr 20:41:08 CEST 2021

TeslaUsbBot, [Apr 1, 2021 at 20:41]
teslausb: Archiving completed successfully. Archived 4 event files in 41s

It does not make any sense considering the contents of the log file... 🤔

/Johan

JohPePe, 2021-04-01:
Tried it again, this time I got the 0 file message

cat rsynclog
building file list ... done

Number of files: 0
Number of created files: 0
Number of deleted files: 0
Number of regular files transferred: 0
Total file size: 0 bytes
Total transferred file size: 0 bytes
Literal data: 0 bytes
Matched data: 0 bytes
File list size: 0
File list generation time: 0.001 seconds
File list transfer time: 0.000 seconds
Total bytes sent: 29
Total bytes received: 12

sent 29 bytes  received 12 bytes  82.00 bytes/sec
total size is 0  speedup is 0.00

Telegram messages below

TeslaUsbBot, [Apr 1, 2021 at 22:50]
teslausb: Archiving 3 event folder(s) with 37 file(s) and 0 track mode file(s) starting at Thu  1 Apr 22:50:46 CEST 2021

TeslaUsbBot, [Apr 1, 2021 at 22:50]
teslausb: Archiving completed successfully. Archived 0 event files in 10s

/Johan

marcone, 2021-04-01:
Which archive method are you using? (cifs, rsync or rclone?)

JohPePe, 2021-04-01:
I’m using cifs!

/Johan

marcone, 2021-04-02:
You're running an older version of teslausb it seems, so try `/root/bin/setup-teslausb upgrade`

miles267, 2021-04-02:
Oddly I’m consistently getting errors syncing (cifs) with the latest build with the /RecentClips sync enabled and there are any /SentryClips. But was able to sync without error previously when using the “hack” above. Am not sure what is different about the new, official implementation than the previous method above.

marcone, 2021-04-02:
> @marcone should I be changing the export ARCHIVE_SYSTEM=cifs to =rclone?

Not sure I understand the question in this context. You'd use rclone if you want to start archiving to cloud storage instead of your own file server.

rmyadsk, 2021-04-02:
> > @marcone should I be changing the export ARCHIVE_SYSTEM=cifs to =rclone?
> 
> Not sure I understand the question in this context. You'd use rclone if you want to start archiving to cloud storage instead of your own file server.

Yes - rclone should be considered only for remote cloud. For local storage - say, a destination on the car's own subnet (or one easily routable to it), rsync is ideal if you have a Linux or Mac (or a NAS, which will likely be based on Linux) setup as your archive server. If your server is Windows, you'll likely want CIFS, unless you want to install cygwin and a sshd service and all that...

I have found the rsync/ssh combo to be exceptionally reliable.

JohPePe, 2021-04-02:
Actually i was running the axkg repo. 

export REPO=axkg
export BRANCH=matrix

I had no idea that could be the cause of my problems.
Going back to main dev to see if problems are solved.

JohPePe, 2021-04-02:
After going back to main dev and updating it seems to work ok.

file has vanished: "/tmp/cam/merged/RecentClips/2021-04-02/2021-04-02_11-29-56-left_repeater.mp4"
file has vanished: "/tmp/cam/merged/RecentClips/2021-04-02/2021-04-02_11-29-56-right_repeater.mp4"
RecentClips/2021-04-02/2021-04-02_11-30-57-back.mp4
RecentClips/2021-04-02/2021-04-02_11-30-57-front.mp4
RecentClips/2021-04-02/2021-04-02_11-30-57-left_repeater.mp4
RecentClips/2021-04-02/2021-04-02_11-30-57-right_repeater.mp4
RecentClips/event.json/
RecentClips/event.json/event.json

Number of files: 237 (reg: 227, dir: 10)
Number of created files: 186 (reg: 185, dir: 1)
Number of deleted files: 0
Number of regular files transferred: 185
Total file size: 8.44G bytes
Total transferred file size: 7.05G bytes
Literal data: 5.14G bytes
Matched data: 0 bytes
File list size: 0
File list generation time: 0.209 seconds
File list transfer time: 0.000 seconds
Total bytes sent: 5.14G
Total bytes received: 5.50K

sent 5.14G bytes  received 5.50K bytes  2.71M bytes/sec
total size is 8.44G  speedup is 1.64
rsync warning: some files vanished before they could be transferred (code 24) at main.c(1330) [sender=v3.2.3]

Thanks for your help!
(Sorry for any extra work a caused)

miles267, 2021-04-02:
I'm using CIFS and have experiencing frequent archival errors.  Seems I was getting them hourly overnight but not awake to capture the log that marcone requested yet.  I'll be at the ready to capture and share here today.

miles267, 2021-04-06:
@marcone was finally able to trap the sync errors when receiving this error message.  It references files vanishing before they could be transferred.  Perhaps this is normal but is being reported as an error?  Please advise.
[rsynclog.txt](https://github.com/marcone/teslausb/files/6264506/rsynclog.txt)

JohPePe, 2021-04-06:
Aha, got those messages now and then. Hope there is some way the script can ignore it and continue syncing. 😀

/Johan

miles267, 2021-04-06:
> Aha, got those messages now and then. Hope there is some way the script can ignore it and continue syncing. 😀 /Johan
> […](#)

Yes!  The occurrence of "vanished" files that are expected by teslausb is halting the sync session completely.  At that point, it isn't until the subsequent sync session (that is triggered by teslausb pi device leaving and returning to wifi) that the next sync session begins.  Between those 2 events, it seems as if additional files are added and accumulate making that next sync session even longer and at risk of interruption due to vanished files (as they are labeled in the log).

@marcone is it in any way to modify the process to flag and ignore vanished files?  I'm not certain how it identifies the files it expects to sync which are deleted, no longer exist, rendering them as vanished.

Thanks for consideration.

marcone, 2021-04-06:
When it complained about the vanished files, did it also say "Archived 0 event files"? (the rsync log you attached shows it archived 240 files, so I would expect it to say that).

There doesn't appear to be an option to make it ignore "vanished files", however it does emit a specific error code in that case, so I could make it so that specific error code is ignored.

miles267, 2021-04-06:
No, the first message says:
Archiving 0 event folder(s) with 240 file(s) starting at Tue  6 Apr 06:45:32 CDT 2021

The subsequent message, after the sync session has run before a period of time returns:
Error during archiving. Archived 240 event files in 6m34s

I believe since it says 0 event folders, these are all RecentClips if that helps.

marcone, 2021-04-06:
Can you also attach your archiveloop.log here? (no need for full diagnostics)
I'm trying to understand why these files disappeared. I suspect it's because a snapshot was taken during archiving, which could delete an old snapshot, but want to make sure.

JohPePe, 2021-04-06:
Hi Marcone!
Here’s my file

https://1drv.ms/u/s!AgjRNx_1R6lDg_suPzxunM3oJTocfg

Example Mon 5 Apr 10:40:22 CEST 2021: Error during archiving. Archived 0 event files in 8 s

marcone, 2021-04-06:
Thanks, @JohPePe, but that question was mainly for @miles267 :)
I took a look at your log too, and the last few failures were because teslausb couldn't ping the archive server anymore. This is usually a sign of poor wifi signal.
You do have a few failures that are like this:
```
Mon  5 Apr 10:40:14 CEST 2021: Archiving 2 event folder(s) with 202 file(s) starting at Mon  5 Apr 10:40:14 CEST 2021
Mon  5 Apr 10:40:14 CEST 2021: Sending Telegram message.
Mon  5 Apr 10:40:15 CEST 2021: Send Matrix message.
Mon  5 Apr 10:40:22 CEST 2021: Error during archiving. Archived 0 event files in 8s
```
That's the kind of failure where /tmp/rsynclog might be helpful.

JohPePe, 2021-04-06:
Ok @marcone.

I looked at the log as it was running and among other things I got :

SentryClips/2021-04-06_10-18-47/2021-04-06_10-13-13-left_repeater.mp4
sender file already removed: SentryClips/2021-04-06_06-40-42/2021-04-06_06-39-37-left_repeater.mp4
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-13-13-right_repeater.mp4"
sender file already removed: SentryClips/2021-04-06_06-40-42/2021-04-06_06-39-37-right_repeater.mp4
sender file already removed: SentryClips/2021-04-06_06-40-42/event.json
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-14-13-back.mp4"
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-14-13-front.mp4"
sender file already removed: SentryClips/2021-04-06_10-18-47/2021-04-06_10-12-28-back.mp4
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-14-13-left_repeater.mp4"
sender file already removed: SentryClips/2021-04-06_10-18-47/2021-04-06_10-12-28-front.mp4
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-14-13-right_repeater.mp4"
sender file already removed: SentryClips/2021-04-06_10-18-47/2021-04-06_10-12-28-left_repeater.mp4
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-15-13-back.mp4"
sender file already removed: SentryClips/2021-04-06_10-18-47/2021-04-06_10-12-28-right_repeater.mp4
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-15-13-front.mp4"
sender file already removed: SentryClips/2021-04-06_10-18-47/2021-04-06_10-13-13-back.mp4
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-15-13-left_repeater.mp4"
sender file already removed: SentryClips/2021-04-06_10-18-47/2021-04-06_10-13-13-front.mp4
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-15-13-right_repeater.mp4"
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-16-13-back.mp4"
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-16-13-front.mp4"
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-16-13-left_repeater.mp4"
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-16-13-right_repeater.mp4"
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-17-13-back.mp4"
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-17-13-front.mp4"
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-17-13-left_repeater.mp4"
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-17-13-right_repeater.mp4"
file has vanished: "/tmp/cam/merged/SentryClips/2021-04-06_10-18-47/2021-04-06_10-18-14-back.mp4"

marcone, 2021-04-06:
@JohPePe I suspect you have your cam size set to 100% or at least a relatively large value compared to the size of your card. I recommend you don't do that, it defeats the entire purpose of having snapshots and archiving recentclips.
In any case, there seems to be a bug in handling that: from your logs it looks like an old snapshot was removed, but the links that were created for it stayed behind. You can check if that's the case for you by doing
```
sudo -i
bin/setup-teslausb install tools/checksnapshotlinks.sh
bin/checksnapshotlinks.sh
```
if that runs for a bit and then exits without printing anything, you're good.
If it outputs anything about missing or bad links, run it again like so:
```
bin/checksnapshotlinks.sh repair
```

JohPePe, 2021-04-06:
@Marcone, yes had camsize set to 100%, changed it to 90%.

Thanks 👍😀

/Johan

marcone, 2021-04-06:
I would suggest setting it much, much lower (depending on the size of your sd card). Also note that just changing the config variable won't change anything. You'll need to delete /backingfiles/cam_disk.bin and re-run setup for the change to take effect.

miles267, 2021-04-06:
I'm using a 128GB Sandisk Endurance card, however my cam limit is set to a fixed 64G rather than a percentage.  Capacity hasn't been an issue.

miles267, 2021-04-07:
> Can you also attach your archiveloop.log here? (no need for full diagnostics)
> I'm trying to understand why these files disappeared. I suspect it's because a snapshot was taken during archiving, which could delete an old snapshot, but want to make sure.

@marcone providing archiveloop.log per your request.
[archiveloop.log](https://github.com/marcone/teslausb/files/6268394/archiveloop.log)

marcone, 2021-04-07:
@miles267 what size sd card do you use?
I see this in the log:
```
Tue  6 Apr 06:45:32 CDT 2021: There are 0 event folder(s) with 240 file(s) and 0 track mode file(s) to move.  0 short recording(s) will be skipped.
Tue  6 Apr 06:45:32 CDT 2021: Starting recording archiving
Tue  6 Apr 06:45:32 CDT 2021: Archiving 0 event folder(s) with 240 file(s) starting at Tue  6 Apr 06:45:32 CDT 2021
Tue  6 Apr 06:45:32 CDT 2021: Sending Pushover message.
Tue  6 Apr 06:51:49 CDT 2021: waiting up to 90 seconds for idle interval
Tue  6 Apr 06:51:50 CDT 2021: mass storage process not active, OK to write
Tue  6 Apr 06:51:51 CDT 2021: low space, deleting /backingfiles/snapshots/snap-000000
Tue  6 Apr 06:51:57 CDT 2021: warning: low space for snapshots
Tue  6 Apr 06:51:57 CDT 2021: taking snapshot of cam disk in /backingfiles/snapshots/snap-000000
Tue  6 Apr 06:52:06 CDT 2021: Error during archiving. Archived 240 event files in 6m34s
```
which explains the situation from your rsynclog file: it started archiving, then during archiving another snapshot was taken (it takes one every hour), and because your cam_size is relatively large compared to your total storage size, it had to delete the oldest (and only) snapshot.

miles267, 2021-04-07:
Thanks @marcone - my SD card is 128GB. My camsize is set to 64G. Am not syncing music.

miles267, 2021-04-09:
@marcone do you need any additional details above?  I've attached my recent achiveloop.log file per request.  Thanks.

miles267, 2021-04-13:
@JohPePe are you no longer receiving any errors during archiving of /RecentClips or anything else?  I continue to receive these sporadically with the latest teslausb and my RPi 0 since migrating from a hacked teslausb image to the official image with the archival of recentclips enabled.  Also, I notice that my RPi 0 with this new configuration also sometimes connects to wifi and begins archiving and other times not until/unless I reboot the RPi 0.  When I reimagine, disabling the archival of recentclips option, this issue doesn't occur any longer.

I had thought about trying a RPi 4 with 5 ghz just to benefit from faster throughput on the uploads back to my file share, but am first trying to figure out how to resolve archival errors due to "file has vanished" as you've indicated above.

marcone, 2021-04-15:
Have you updated in the last 5 days? I made a change that should ignore the vanished files error.

miles267, 2021-04-15:
> Have you updated in the last 5 days? I made a change that should ignore the vanished files error.

Thanks @marcone wasn't aware of this update.  I've just updated my RPi 0 and had my first successful sync of 227 RecentFiles.  Will let it run for another day to see whether I no longer encounter the errors during archiving teslausb was prone to earlier.

miles267, 2021-04-16:
> Thanks @marcone wasn't aware of this update. I've just updated my RPi 0 and had my first successful sync of 227 RecentFiles. Will let it run for another day to see whether I no longer encounter the [...]

@marcone unfortunately this did not address the issue.  That's not to say that it didn't improve the process.  Am going to test whether a RPi 4 might address these issues as was the case with others here.

marcone, 2021-04-16:
@miles267 It should be ignoring the "vanished files" now, so what problem is it still giving you?

JohPePe, 2021-04-16:
@marcone, I’m still getting those errors.

Error during archiving. Archived 0 event files in 10s
If I login and manually reboot, it usually works as expected.

I just had an idea...
I’ve set the archivedelay to 30 s, but the scripts seems to ignore it.
In archiveloop.log it starts archiveloop after just 22 secs.

Just my 10 c worth :-)

/Johan

marcone, 2021-04-16:
@JohPePe the delay [is still there](https://github.com/marcone/teslausb/blob/main-dev/run/archiveloop#L835), but it only delays after reconnecting to wifi, not on initial boot.

JohPePe, 2021-04-16:
Yes, that is how I thought it would work too.
But

I get home with my car. Raspberry connects to my WiFi. Starts to sync, and failes.
I reboot and before the last upgrade the sync would succeed.
I don’t understand why it fails initially, the WiFi signal is strong and I can ssh without any problems. Very strange...

/Johan

marcone, 2021-04-16:
> I don’t understand why it fails initially, the WiFi signal is strong and I can ssh without any problems

the answer will likely be in the rsync log files in /tmp

JohPePe, 2021-04-17:
@marcone
Yesterday when I got home, the pi started to sync, only to fail a few seconds later, as it usually does.

Instead of rebooting the pi, I tried to create a file, /tmp/archive_is_unreachable, and a few seconds later it started to sync. Many minutes later it had transferred 386 files without any problems.

So, I suspect that when it fails to sync initially it’s because the pi is not fully connected to my network. 🤔

/Johan

miles267, 2021-04-19:
> @miles267 It should be ignoring the "vanished files" now, so what problem is it still giving you?

I’ve since moved to a RPi4 and all problems seem to have gone away. Am no longer getting any archiving errors. And the boot up times and 5 ghz WiFi sync are about 5X faster. Massive improvement.

miles267, 2021-04-26:
Have upgraded to most current teslausb.  Am not receiving archiving errors any longer.  However now it appears /RecentClips are coming thru consistently.  The sync sessions often stop and I must manually trigger a resync.

Also I've noticed that my /SentryClip dated folders are created and contain either only a .json file or a series of small video clips that are 500K or less.  This is consistent and unusual.  Has anyone else encountered this?

marcone, 2021-04-26:
> RecentClips are coming thru consistently

I assume you meant "inconsistently"? If the sync stops and there's no message about it, that sounds like the Pi is powering off mid-sync.

> video clips that are 500K or less

That's an issue on the car's side that in my experience requires resetting the MCU to resolve.

JohPePe, 2021-04-29:
Hi @marcone

After the last update, which allowed us to see recordings via www (nicely done), no files gets synced at all.

TeslaUsbBot, [Apr 29, 2021 at 22:24]
teslausb: Archiving 7 event folder(s) with 1229 file(s) starting at Thu 29 Apr 22:24:17 CEST 2021

TeslaUsbBot, [Apr 29, 2021 at 22:26]
teslausb: Archiving 4 event folder(s) with 1055 file(s) starting at Thu 29 Apr 22:26:53 CEST 2021

TeslaUsbBot, [Apr 29, 2021 at 22:27]
teslausb: Error during archiving. Archived 0 event files in 9s

TeslaUsbBot, [Apr 29, 2021 at 22:28]
teslausb: Archiving 7 event folder(s) with 1229 file(s) starting at Thu 29 Apr 22:28:06 CEST 2021

TeslaUsbBot, [Apr 29, 2021 at 22:28]
teslausb: Error during archiving. Archived 0 event files in 8s

TeslaUsbBot, [Apr 29, 2021 at 22:29]
teslausb: Archiving 7 event folder(s) with 1229 file(s) starting at Thu 29 Apr 22:29:24 CEST 2021

TeslaUsbBot, [Apr 29, 2021 at 22:29]
teslausb: Error during archiving. Archived 0 event files in 9s

TeslaUsbBot, [Apr 29, 2021 at 22:30]
teslausb: Archiving 7 event folder(s) with 1229 file(s) starting at Thu 29 Apr 22:30:56 CEST 2021

TeslaUsbBot, [Apr 29, 2021 at 22:31]
teslausb: Error during archiving. Archived 0 event files in 8s

I’ve triggered a new sync via www several times but it always fails. Not sure what to do.
/Johan

marcone, 2021-04-29:
> TeslaUsbBot, [Apr 29, 2021 at 22:30]
> teslausb: Archiving 7 event folder(s) with 1229 file(s) starting at Thu 29 Apr 22:30:56 CEST 2021
> 
> TeslaUsbBot, [Apr 29, 2021 at 22:31]
> teslausb: Error during archiving. Archived 0 event files in 8s

That seems like something that you might see if there's a permission problem with the archive server.
If you look at the rsync logs in /tmp, it might show the problem (you'll have to ssh in to the Pi as this is not yet included in the diagnostics)

JohPePe, 2021-04-29:
I’ve tried that, but it does not create the rsync log file at all.

/Johan

miles267, 2021-04-29:
Oddly I am getting consistent syncs of /RecentClips.  Working great.  No dated folders with only .json files and no folders with files less than ~500K in size.  However, /SentryClips remains problematic despite rebooting the MCU repeatedly. The sync session completes successfully per PushOver and all /RecentClips files included are intact; however the corresponding /SentryClips dated folder now only includes a single .json file.  What would cause this?

marcone, 2021-04-30:
> Oddly I am getting consistent syncs of /RecentClips. Working great. No dated folders with only .json files and no folders with files less than ~500K in size. However, /SentryClips remains [...]

The only thing I can think of is that for some reason rsync decides to not archive your SentryClips files because the same files were already archived as part of RecentClips (in order to provide a continuous "RecentClips" record, TeslaUSB makes recordings that are part of a Sentry event also appear in RecentClips).
It doesn't do that for me though, so I don't see why it would do that for you.

CodeChief, 2021-04-30:
I had this issue and had to improve "archive_clips.sh" to see the problem (update on that later). In short, it was too difficult to see the failure because the log is not accessible or echoed to the terminal (can't even see the fault when testing via SSH). In my case the original RSync errors (currently invisible) told me straight away that I was having network disconnects and a disk full error. But I had no way to know it as a normal user. The current implementaiton just says "errors occured" which is about as useful as chocolate teapot :-)

So until that is improved, to help yourself... When the problem occurs you should login to the Pi with SSH and check the contents of "/tmp/archive-rsync-cmd.log" or "/tmp/rsynclog".

miles267, 2021-05-01:
I am going to reimage my RPi 0 (have been using a RPi 4) and see whether that helps.  Fortunately the Pi 4 with 802.11ac just syncs so much faster.  I have been synching 5k files per day whereas the RPi 0 wouldn't typically finish it's sync sessions which prompted me to upgrade it.  I suppose I'll go back to the Pi 0 now with /RecentClips enabled and a clean image.  This should ensure /SentryClips sync as they've not been at all in the last 2 weeks.

miles267, 2021-05-01:
Here we go again.  A fresh RPi 0 with latest image and updates.  Stable wifi directly overhead in garage.  Sentry mode on (to prevent any power outages).  No issues with the network file share.  Yet teslausb is reporting errors during sync from PushOver.  These errors were not occurring with the RPi 4 though it never synced any /SentryClips despite returning a successful sync message of ~5k files at a time in PushOver alerts.  Achiveloop attached if someone can please help me.  Thanks.
[archiveloop.log](https://github.com/marcone/teslausb/files/6409852/archiveloop.log)

marcone, 2021-05-01:
The archiveloop.log is not helpful here. Look at the rsync logs, as mentioned multiple times in comments above, or possibly the full diagnostics.

miles267, 2021-05-01:
> The archiveloop.log is not helpful here. Look at the rsync logs, as mentioned multiple times in comments above, or possibly the full diagnostics.

Apologies.  Have attached the full diagnostic log here.  Unfortunately I'm not very good at interpreting the rsync logs :-(
[diagnostics.txt](https://github.com/marcone/teslausb/files/6409938/diagnostics.txt)

marcone, 2021-05-01:
The diagnostics show that rsync crashed with what looks to be a jump to a bad address:
```
[Sat May  1 10:31:51 2021] Unable to handle kernel paging request at virtual address bfc730e0
[Sat May  1 10:31:51 2021] pgd = 50e98393
[Sat May  1 10:31:51 2021] [bfc730e0] *pgd=00000000
[Sat May  1 10:31:51 2021] Internal error: Oops: 80000005 [#1] ARM
[Sat May  1 10:31:51 2021] Modules linked in: overlay md4 md5 sha512_generic aes_arm aes_generic libaes cmac hmac nls_utf8 cifs gcm libarc4 8021q garp stp llc fuse xfs brcmfmac brcmutil sha256_generic libsha256 cfg80211 rfkill raspberrypi_hwmon bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) snd_bcm2835(C) v4l2_mem2mem videobuf2_vmalloc bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_pcm videobuf2_common snd_timer snd vc_sm_cma(C) videodev mc uio_pdrv_genirq fixed uio dwc2 udc_core ip_tables x_tables ipv6 [last unloaded: libcomposite]
[Sat May  1 10:31:52 2021] CPU: 0 PID: 12921 Comm: rsync Tainted: G         C        5.4.79+ #1373
[Sat May  1 10:31:52 2021] Hardware name: BCM2835
[Sat May  1 10:31:52 2021] PC is at 0xbfc730e0
[Sat May  1 10:31:52 2021] LR is at cifs_revalidate_dentry_attr+0xfc/0x208 [cifs]
[Sat May  1 10:31:52 2021] pc : [<bfc730e0>]    lr : [<bf4370d0>]    psr: 60000013
[Sat May  1 10:31:52 2021] sp : d65bdcb0  ip : d65bdcb0  fp : d65bdcec
[Sat May  1 10:31:52 2021] r10: 00000000  r9 : c0a97028  r8 : d56372a0
[Sat May  1 10:31:52 2021] r7 : d0318400  r6 : 00001079  r5 : 0000000b  r4 : 00000000
[Sat May  1 10:31:52 2021] r3 : 00000000  r2 : bf496c0c  r1 : dbfba020  r0 : 00001079
[Sat May  1 10:31:52 2021] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[Sat May  1 10:31:52 2021] Control: 00c5387d  Table: 14330008  DAC: 00000055
[Sat May  1 10:31:52 2021] Process rsync (pid: 12921, stack limit = 0xd55dc0c3)
[Sat May  1 10:31:52 2021] Stack: (0xd65bdcb0 to 0xd65be000)
[Sat May  1 10:31:52 2021] dca0:                                     00001079 00000000 d3470be0 0e0136c7
[Sat May  1 10:31:52 2021] dcc0: d5637e80 d3470be0 d3470be0 d65bdd74 c0a97028 fffffff6 d65bdd6c cf4fdf10
[Sat May  1 10:31:52 2021] dce0: d65bdd04 d65bdcf0 bf437228 bf436fe0 d6137990 d3470be0 d65bdd1c d65bdd08
[Sat May  1 10:31:52 2021] dd00: bf427ab4 bf437218 d65bde90 d6137990 d65bdd5c d65bdd20 c01d881c bf427a64
[Sat May  1 10:31:52 2021] dd20: d65bdd70 00000002 00000002 0e0136c7 60000013 d65bde90 00000000 c0a97028
[Sat May  1 10:31:52 2021] dd40: 00000003 00000000 0000000c e2e0e2ff d65bddac d65bdd60 c01da9e4 c01d857c
[Sat May  1 10:31:52 2021] dd60: d3470dc8 00000081 00000000 d3470be0 00000002 d65bdd80 c01d7094 0e0136c7
[Sat May  1 10:31:52 2021] dd80: c0bbf7cc d65bde90 fefefeff 61c88647 d48a401d 00000000 0000000c e2e0e2ff
[Sat May  1 10:31:52 2021] dda0: d65bddfc d65bddb0 c01daf9c c01da9a8 c01ccd30 c00616f8 c0bbf3b0 c0ac0e30
[Sat May  1 10:31:52 2021] ddc0: 000200c2 d331f880 d65bddfc 0e0136c7 c01cd104 d65bde90 d65bde90 ffffe000
[Sat May  1 10:31:52 2021] dde0: c0a97028 d65bdf50 d65bc000 d4b14d20 d65bde8c d65bde00 c01db850 c01dad14
[Sat May  1 10:31:52 2021] de00: 00000000 d65bde84 d65bde98 d65bde54 00000000 d3470810 d65bde44 d65bde28
[Sat May  1 10:31:52 2021] de20: c0014714 ffffe000 00000041 c0a97028 00000001 c01ccec8 d65bde7c d65bde48
[Sat May  1 10:31:52 2021] de40: c0013548 c00146f8 c0013548 00000000 c01ccec8 d65bdefc c0ab155c 0e0136c7
[Sat May  1 10:31:52 2021] de60: 00000001 00000001 d65bde90 d65bdf50 c0a97028 00000001 d65bc000 00000000
[Sat May  1 10:31:52 2021] de80: d65bdf44 d65bde90 c01ddef8 c01db7cc cf4fdf10 d622d5d8 d9888369 0000000c
[Sat May  1 10:31:52 2021] dea0: d48a4010 d65bdeb0 00000000 c00146f8 d3470dc8 00000011 00000002 00000696
[Sat May  1 10:31:52 2021] dec0: 00000000 00000000 00000000 d65bded0 d65bdf14 0e0136c7 c0013548 00000001
[Sat May  1 10:31:52 2021] dee0: d3017000 d3017014 00000000 00000400 000200c2 00000020 d65bdf34 d65bdf08
[Sat May  1 10:31:52 2021] df00: d48a4000 00000000 000200c2 000200c2 ffffff9c ffffff9c d48a4000 0e0136c7
[Sat May  1 10:31:52 2021] df20: d65bc000 00000001 c0a97028 ffffff9c d48a4000 fffff000 d65bdf94 d65bdf48
[Sat May  1 10:31:52 2021] df40: c01c89d4 c01dde84 00000000 bebf00d0 000200c2 c0008180 00000006 00000700
[Sat May  1 10:31:52 2021] df60: 00000001 0e0136c7 c00777a4 a0199adc b6f23968 bebf2284 00000142 c00091a4
[Sat May  1 10:31:52 2021] df80: d65bc000 00000000 d65bdfa4 d65bdf98 c01c8acc c01c8878 00000000 d65bdfa8
[Sat May  1 10:31:52 2021] dfa0: c0009000 c01c8abc a0199adc b6f23968 ffffff9c bebf2250 000200c2 00000180
[Sat May  1 10:31:52 2021] dfc0: a0199adc b6f23968 bebf2284 00000142 00000000 0003a2f8 bebf2250 00000000
[Sat May  1 10:31:52 2021] dfe0: b6cad000 bebf0088 b6bbc31c b6c282f0 60000010 ffffff9c 00000000 00000000
[Sat May  1 10:31:52 2021] Backtrace: 
[Sat May  1 10:31:52 2021] [<bf436fd4>] (cifs_revalidate_dentry_attr [cifs]) from [<bf437228>] (cifs_revalidate_dentry+0x1c/0x30 [cifs])
[Sat May  1 10:31:52 2021]  r10:cf4fdf10 r9:d65bdd6c r8:fffffff6 r7:c0a97028 r6:d65bdd74 r5:d3470be0
[Sat May  1 10:31:52 2021]  r4:d3470be0
[Sat May  1 10:31:52 2021] [<bf43720c>] (cifs_revalidate_dentry [cifs]) from [<bf427ab4>] (cifs_d_revalidate+0x5c/0xe4 [cifs])
[Sat May  1 10:31:52 2021]  r5:d3470be0 r4:d6137990
[Sat May  1 10:31:52 2021] [<bf427a58>] (cifs_d_revalidate [cifs]) from [<c01d881c>] (lookup_fast+0x2ac/0x2bc)
[Sat May  1 10:31:52 2021]  r5:d6137990 r4:d65bde90
[Sat May  1 10:31:52 2021] [<c01d8570>] (lookup_fast) from [<c01da9e4>] (walk_component+0x48/0x36c)
[... cut ...]
[Sat May  1 10:31:52 2021] dfa0:                   a0199adc b6f23968 ffffff9c bebf2250 000200c2 00000180
[Sat May  1 10:31:52 2021] dfc0: a0199adc b6f23968 bebf2284 00000142 00000000 0003a2f8 bebf2250 00000000
[Sat May  1 10:31:52 2021] dfe0: b6cad000 bebf0088 b6bbc31c b6c282f0
[Sat May  1 10:31:52 2021] Code: bad PC value
```
It looks to me like this is a bug in the kernel CIFS code. It's entirely possible this has been fixed upstream already, so try updating the kernel to the latest. (`sudo apt update && sudo apt upgrade`)

miles267, 2021-05-01:
Thanks @marcone.  There were 76 updates available.  Applying them all now.  Will see whether it addresses the issue.

rmyadsk, 2021-05-01:
> There were 76 updates available.

This seems like an awful lot. Would it be reasonable to consider implementing an autoupdate function into the teslausb image? Just a bot that periodically makes the filesystem r/w, does a global update/upgrade, and reboots.

marcone, 2021-05-01:
You can add a cron job for that if you want. The problem I see with auto-updating is that if something goes wrong (and something will eventually go wrong), you won't know until you see it's no longer recording, at which time you've probably got places to go and no time to fix the Pi.

rmyadsk, 2021-05-01:
> The problem I see with auto-updating is that if something goes wrong (and something will eventually go wrong)

Point taken; however maybe there is a happy medium - perhaps restricting apt to more mature (and hopefully less volatile) repositories via `/etc/apt/sources.list` or another method? Would we agree that in no situation should users find themselves where @miles267 was (i.e. a severely aged-out system layer inhibiting basic app operation)? Isn't that the opposite of too frequent updates, and just as destructive to production?

[48 more replies]

closed 2024-04-06 by marcone