marcone/teslausb #626: Can I change the shared folder without redoing the entire thing?

muffstic, 2021-04-09:
Eg. my current shared folder is setup as C://TeslaCam but I want to change it to H://TeslaCam for example. Is this possible to do without going through all the setup again?

marcone, 2021-04-09:
easiest would be to edit /etc/fstab on the Pi

closed 2021-04-09 by marcone

muffstic, 2021-04-10:
Can you explain that so a dummy can understand?

marcone, 2021-04-11:
If you're using a shared folder on a Windows machine, and that folder used to be C:/TeslaCam and is now H:/TeslaCam, I think you should just be able to stop sharing the C:/TeslaCam folder and start sharing the H:/TeslaCam folder on the Windows machine, without needing any changes on the teslausb side.
If you do need to make changes on the teslausb side, for example because you're using a different computer or shared folder name, then ssh into the Pi, and run
```
sudo -i
bin/remountfs_rw
nano teslausb_setup_variables.conf
```
to edit the configuration file. Change the share info, save it, then run `bin/setup-teslausb && reboot`
If that looks too daunting, just start over from scratch.