marcone/teslausb #632: Webook to Home Assistant Issue

jun3280net, 2021-04-24:
Hi there.  I recently updated Teslausb - and noticed the significant improvements since the last time I did a fresh install.  While everything works, I wanted to enable notifications to my phone when the files were being processed.  I noticed the ability to use Webhooks and Home Assistant, so I decided to enable it.  I configured Home Assistant according to their documentation : 

[https://www.home-assistant.io/docs/automation/trigger/#webhook-trigger](https://www.home-assistant.io/docs/automation/trigger/#webhook-trigger)

My entry in HA is as follows :
[image]

I ran the CURL command : curl -X POST https://hassio.xyz.ddns.net/api/webhook/teslausb
[image]

I confirmed I received messages on my phone for 2/3 tests (2nd one failed as I adjusted network config).

Teslausb config file is setup as follows : 
[image]

Teslausb logs (using the newly implemented web interface - which is awesome) shows the following : 
[image]

However - I receive no notifications on my phone.  Any idea what I might have done wrong? or how I can debug / troubleshoot this?

Thanks in advance for any insights.

marcone, 2021-04-24:
The webhook notification also sends along data as part of the POST: https://github.com/marcone/teslausb/blob/main-dev/run/send-push-message#L56
Maybe Home Assistant doesn't like that?
Or maybe it just wasn't able to connect. The fact that it apparently only archived 7 files in 9 minutes tells me your Pi has really poor wifi reception.

You can try just running `/root/bin/send-push-message title message` on the command line on the Pi, and see if that works or reports something useful.

jun3280net, 2021-04-24:
> The webhook notification also sends along data as part of the POST: https://github.com/marcone/teslausb/blob/main-dev/run/send-push-message#L56
> Maybe Home Assistant doesn't like that?
> Or maybe it [...]

Thank you for your prompt response @marcone!  Is there any way to check the speed of the file transfers or check the strength of the wifi connection?  That said, Teslausb from what I've observed, the files have always made it to the NAS.

On that note - I've run the command you've suggested.  I must have something configured wrong.  
[image]

That said I navigated to the bin directory, and did a "nano send-push-message" to see what might be inside : 
[image]

I searched through the thread to see how that command might be used - and saw a comment you made in another thread.  Same issue with the command.
[image]

Thanks again for your help

marcone, 2021-04-24:
Try:
```
sudo -i
bash -c bash notbash
function log { echo "$@"; }
export -f log
bin/send-push-message title message
```

jun3280net, 2021-04-24:
> Try:
> 
> ```
> sudo -i
> bash -c bash notbash
> function log { echo "$@"; }
> export -f log
> bin/send-push-message title message
> ```

Great - PROGRESS!
[image]

It is now registering with Home Assistant!  I am now getting the same message as the original CURL command.  I think I need to adjust the Home Assistant side of things to process whatever information is being sent via Webhook.

Cheers and thanks

jun3280net, 2021-04-24:
@marcone thanks for your help.  Getting it to work from the Raspberry Pi was the key.  I was able to figure out what was required in the Home Assistant - Automations.yaml file.  Teslausb Webhook passes in JSON data.

[image]

[image]

closed 2021-04-24 by jun3280net