marcone/teslausb #633: Build documents

CodeChief, 2021-04-29:
I'm trying to fix other issues but what fails me in this repository compared to normal repositiories is a lack of a "Build" documentation section. After editing some scripts to behave the way I want to test, it didn't work even with the correct REPO and BRANCH exports in the "/boot/teslausb_setup_variables.conf" configuration file.

So please tell me (and update the documentation for others) what the build environment is exactly and how we build it to the ZIP/... in order to get the update command on a test unit to load it.

marcone, 2021-04-29:
I'm not sure what you mean by "build" in this context. There's nothing to build. The setup and runtime scripts are pulled directly from the repo and branch specified in the config file. So in your case you would add:
```
export REPO=CodeChief
export BRANCH=Development
```
to the configuration file.

CodeChief, 2021-04-29:
It's okay now thanks. I was thrown off track by the SNAPSHOTS_ENABLED setting, which wss still documented in my settings sample file. The setup scripts appears to have logic still in which downloads some archive, which I don't have because I cleaned the other branches ("no-snapshots" specifically).

```
Thu 29 Apr 03:36:31 CEST 2021 : curl -s https://api.github.com/repos/CodeChief/teslausb/git/refs/heads/Development
Thu 29 Apr 03:36:32 CEST 2021 : curl -o /tmp/setup-teslausb https://raw.githubusercontent.com/CodeChief/teslausb/2315f597b82bc8a2c1d207b932da724c14339e5f/setup/pi/setup-teslausb
Thu 29 Apr 03:36:32 CEST 2021 : Downloaded /tmp/setup-teslausb ...
Thu 29 Apr 03:36:32 CEST 2021 : WARNING: using 'no-snapshots' branch because SNAPSHOTS_ENABLED is not true
Thu 29 Apr 03:36:32 CEST 2021 : preparing for full upgrade
Thu 29 Apr 03:36:43 CEST 2021 : WARNING: using 'no-snapshots' branch because SNAPSHOTS_ENABLED is not true
Thu 29 Apr 03:36:43 CEST 2021 : curl -L https://github.com/CodeChief/teslausb/archive/no-snapshots.tar.gz
Thu 29 Apr 03:36:44 CEST 2021 : 'curl -L https://github.com/CodeChief/teslausb/archive/no-snapshots.tar.gz' failed, retrying
Thu 29 Apr 03:36:47 CEST 2021 : 'curl -L https://github.com/CodeChief/teslausb/archive/no-snapshots.tar.gz' failed, retrying
Thu 29 Apr 03:36:51 CEST 2021 : 'curl -L https://github.com/CodeChief/teslausb/archive/no-snapshots.tar.gz' failed, retrying

```

So removing the depreciated SNAPSHOTS_ENABLED setting worked.

closed 2021-04-29 by CodeChief