marcone/teslausb #634: Partial archiving

Jughead135, 2021-05-02:
I've noticed a few times where I've had a large number of Sentry files (because the car was parked for an extended period, and/or in a "busy" place with lots of Sentry activations), that not all of the Sentry events get archived.

For example:  My car was parked recently for four days, evidently by a popular walkway.  I got back to the car, it reported something like 805 (!!!) Sentry events.  Some significant number of those were undoubtedly overlaps (multiple events saved in same Sentry file).  Many got deleted along the way for space reasons by the TeslaUSB monitor, as indicated in the log.

After being parked from 29 Apr - 02 May, and reporting over 800 events, the Pi SMB SentryClips share contains 56 Sentry events, with the earliest being 01 May about 4PM.  The TeslaUSB archive grabbed the 32 most recent of those, ending with the one triggered right when I got to the car on 02 May a bit after 10AM:

`There are 32 event folder(s) with 981 file(s) and 0 track mode file(s) to move. 3 short recording(s) will be skipped.`

The other (earlier) 24 are still there on the Pi, at least in the SMB share--I can see them, open them, etc.; but they are not being archived by TeslaUSB.

Is there a difference between what is saved and what is deleted in the SMB share vs the snapshots?  Is there some maximum archive number (32?) or something that I'm running into?  Having not "seen" these files, will they be deleted when space demands, or are they forever "invisible" to TeslaUSB?

marcone, 2021-05-02:
Please attach diagnostics

Jughead135, 2021-05-02:
Ugh.  You'd think I'd learned something by now....

Sorry.  Attached here.
[diagnostics.txt](https://github.com/marcone/teslausb/files/6412129/diagnostics.txt)

marcone, 2021-05-02:
I don't see anything out of the ordinary in your diagnostics. Having 56 Sentry events in the Samba share but fewer being archived is probably normal in this case since you got so many Sentry events before archiving: the car deletes older events to make room for newer events (I think), but because of snapshots, some of these still exist in the Samba share.
The latest teslausb archives from snapshots so would have archived those other 24 events too. In order to keep and archive all 805 events accumulated over 4 those days though, you would probably need a larger sd card, and dedicate most of it to CAM_SIZE.

closed 2021-05-02 by marcone

Jughead135, 2021-05-02:
> probably normal in this case since you got so many Sentry events before archiving: the car deletes older events to make room for newer events (I think), but because of snapshots, some of these still exist in the Samba share.

OK, I think you're saying that the car (not TeslaUSB) deletes older Sentry events (which seems it must, since it only "sees" a USB drive of a specific capacity)--AND that TeslaUSB archives Sentry events directly from the car's repository (on the "USB drive" that TeslaUSB presents to it) of Sentry files, and *not* from the snapshots which may include a longer history (up to the actual capacity of the cam drive, as determined by the physical size of the card and the size setting used on initial setup).

Do I have that correct?

> The latest teslausb archives from snapshots so would have archived those other 24 events too.

Are you saying that I have an outdated copy of TeslaUSB--and that a simple update will have it start copying SentryClips from the snapshots instead of the car's direct recording?  I'll update right away!

Thanks, as always!

marcone, 2021-05-02:
> OK, I think you're saying that the car (not TeslaUSB) deletes older Sentry events (which seems it must, since it only "sees" a USB drive of a specific capacity)--AND that TeslaUSB archives Sentry [...]

Yes.
 
> > The latest teslausb archives from snapshots so would have archived those other 24 events too.
> 
> Are you saying that I have an outdated copy of TeslaUSB--and that a simple update will have it start copying SentryClips from the snapshots instead of the car's direct recording?

Yes, though note that it won't archive SentryClips that were already present at the the time of the update (i.e. those 24 events that it didn't archive). To force it it to archive those, delete or edit /mutable/sentry_files_archives after the update and then reboot.

Jughead135, 2021-05-03:
Understood it won't retroactively archive--and, a non-issue in this case, since I simply copied over the 24 events in question prior to the update (man, Samba is s-l-l-l-l-o-w... but, it did its thing in the background & got the job done).

Upgrade completed--as always, thanks for such a great project, and I wish you were in charge of tech support for some of the paid software out there...!!