marcone/teslausb #635: Archiving does not really archive any file

joanquro, 2021-05-03:
Hello.

Yesterday I re-flashed my Pi. Then I upgraded to last version.

Everything was setup as usual, except that I enabled Telegram.

I used the conf file but customized for my particular case.

When I arrive home, the Pi connects to WiFi and sends IFTTT and Telegram message. The archiving begins but no file is transfered.

I attach diagnostics.txt and archiveloop.log

Thank you.
[diagnostics.txt](https://github.com/marcone/teslausb/files/6415330/diagnostics.txt)

joanquro, 2021-05-03:
I forgot to attach arhiveloop.log

[archiveloop.log](https://github.com/marcone/teslausb/files/6415332/archiveloop.log)

joanquro, 2021-05-03:
Looks like rsync is working but I see nothing in my Windows share. I has worked flawlessly for months.

[image]

joanquro, 2021-05-03:
This is weird.

I see a folder in my TeslaCam folder in my Windows computer named ".teslausbtmp"

Inside this folder I see a one temp file whose size 0 bytes but the name is constantly changing:

2021-05-03_10-48-40-right_repeater.mp4.N9BMap then changes to:
2021-05-03_10-49-40-back.mp4.KuCQNE then changes to:
2021-05-03_10-49-40-front.mp4.o57XMj then changes to:
2021-05-03_10-49-40-right_repeater.mp4.GgTBjF and so on...

What is going on?

marcone, 2021-05-03:
The ".teslausbtmp" folder is a temporary folder that rsync uses while syncing. It moves files to that folder first, then renames them to their final location. I don't know why that latter step is apparently not happening for you.
Your diagnostics show it is trying to archive 2276 files. That's going to take a while. After 57 minutes of archiving, the Pi appears to have restarted.

joanquro, 2021-05-03:
Hi marcone, thanks for the fast reply!

I think I know what's going on here. It's archiving RecentClips. I don't want it to do that. I have disabled that option and I'm running setup againg. I will test it in a few minutes. 

Regarding the Pi restarting, that's because I didn't want to wait more time. I just tried to reboot and see if it worked.

Thanks a lot, I'll keep you posted just in case I get it solved (or not).

joanquro, 2021-05-03:
By the way, is there a way to change something in the setup without running the entire the setup process again?.

For example, if I want to change the Telegram token. Where is it stored besides the conf file?

Thanks.

marcone, 2021-05-03:
> For example, if I want to change the Telegram token. Where is it stored besides the conf file?

It's stored in the config file, but you don't need to rerun setup. Just change it and reboot.

closed 2021-05-03 by marcone