marcone/teslausb #640: get_script failed, retrying

mhashemi2, 2021-05-30:
I have tried with 2.2 and 2.5, to get connected to wifi. After booting, the Tesla camera and music folders to not show up in Windows and the device does not join my network. Looking at teslausb-headless-setup.log, I see:

> Wed 22 Jan 23:46:25 GMT 2020 : Detecting whether to update wpa_supplicant.conf
> Wed 22 Jan 23:46:25 GMT 2020 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
> Wed 22 Jan 23:46:25 GMT 2020 : Rebooting...
> Wed 22 Jan 23:47:01 GMT 2020 : Detecting whether to update wpa_supplicant.conf
> Wed 22 Jan 23:47:01 GMT 2020 : Grabbing main setup file.
> Wed 22 Jan 23:47:12 GMT 2020 : get_script failed, retrying

Then the failure message over and over. I do get WIFI_ENABLED and TESLAUSB_SETUP_STARTED, but that's it. Since I cannot ssh into the device and I do not have a monitor cable that'll fit mini-HDMI, I am at a loss on what to do.

The top of my config file looks like this (all the rest is defaults from the sample file):

`# Default variables for CIFS (Windows/Mac file sharing) setup
export ARCHIVE_SYSTEM=cifs
export ARCHIVE_SERVER=192.168.1.x
export SHARE_NAME=myshare
export SHARE_USER=myuser
export SHARE_PASSWORD='myuserpw'
# export cifs_version="3.0"

# Uncomment the following line and select drive if you want to use an external drive instead of built-in MicroSD card.
# External USB drive is supported only for RPi4.
# export USB_DRIVE=/dev/sda

# Uncomment to disable snapshots. This also disables samba and RecentClips-extension
#export SNAPSHOTS_ENABLED=false

export CAM_SIZE=64G
#export MUSIC_SIZE=8G

# export musicsharename=your_music_share/music_folder

# Wifi setup information. Note that Raspberry Pi Zero W only supports 2.4 GHz wifi
export SSID='myssid'
export WIFIPASS='mywifipw'

# uncomment to enable a samba/cifs server that makes the recordings
# available as a network share
export SAMBA_ENABLED=true
# uncomment to enable guest access to the samba server
export SAMBA_GUEST=true

# uncomment to enable teslausb to act as a wifi access point with the given
# SSID, so you can access it while on the road.
export AP_SSID='TESLAUSB WIFI'
# Change this! The setup script will not accept the default 'password'
# Also note that the wifi password must be at least 8 characters.
export AP_PASS='myapwifipw'
# IP address is optional. The AP will give its clients IP addresses in the
# x.x.x.100-150 range so make sure you don't put the AP's IP address in
# that range.
#export AP_IP='192.168.66.1'

# uncomment if you want to set the time zone to something other than the default 'BST'
# Can be an actual time zone, or "auto" to attempt automatic timezone detection
export TIME_ZONE="America/Denver"`

marcone, 2021-05-30:
Most like the wifi info is incorrect. If your wifi SSID or password contains special characters, pay extra attention to the escaping rules as described here: https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md

mhashemi2, 2021-05-30:
> Most like the wifi info is incorrect. If your wifi SSID or password contains special characters, pay extra attention to the escaping rules as described here: https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md

Yeah, I saw that. I do have special characters (which do not include single quotes) but I thought wrapping the whole thing inside single quotes would mean I do not also need to escape the chars. Is that incorrect?

marcone, 2021-05-30:
I think that should work, but I haven't test it exhaustively.
Instead of entering the wifi info in the .conf file, you could try adding it to the wpa_supplicant.conf.sample instead. Then delete WIFI_ENABLED, put the card back in the Pi and boot it.

mhashemi2, 2021-05-30:
I commented out ssid and wifipass in teslausb_setup_variables.conf and added them to wpa_supplicant.conf. After deleting WIFI_ENABLED, I booted up the Pi, but the same thing happened. the log file just shows, "get_script failed, retrying". Now though, I'm not getting "WIFI_ENABLED".

marcone, 2021-05-30:
oh, right, looks like you still need to specify dummy SSID and WIFIPASS variables or the setup script won't work correctly.
So:
- in the wpa_supplicant.conf.sample file, replace TEMPSSID and TEMPPASS with your actual ssid and password.
- in teslausb_setup_variables.conf, specify dummy values for SSID and WIFIPASS.

mhashemi2, 2021-05-30:
Okay, my wpa_supplicant.conf file looks like this:

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
  ssid="mywifissid"
  psk="mywifipw"
  key_mgmt=WPA-PSK
}
```

And teslausb_setup_variables.conf file has "dummyssid" and "dummypw" for SSID and WIFIPASS.

The date in the log file is still wrong and it still says "get_script failed". The interesting thing is that I can see the device has connected to my AP, but it does not have an IP.

marcone, 2021-05-31:
Are you using some kind of login portal or device whitelist or something?
If it's connected to your AP but doesn't get an IP, that's no longer related to teslausb or the teslausb setup process, I think.

mhashemi2, 2021-05-31:
No, just the regular Wifi password and no device filtering.

I tried two different networks too, in case there was something odd with the first one (which is the network to which most of my devices are connected). Same behavior on both. The second network is 2.4GHz only (where the first one is both 2.4 and 5GHz).

mhashemi2, 2021-05-31:
Is there any way to get it to create the folder structure so I could at least use the SD card for TeslaCam, even if TeslaUSB won't upload the video?

mhashemi2, 2021-05-31:
I just realized that the TeslaUSB device I see in the AP logs is the SD card plugged into my laptop but I cannot ssh to pi@teslausb.local because the name is not resolvable.

marcone, 2021-05-31:
> I just realized that the TeslaUSB device I see in the AP logs is the SD card plugged into my laptop

That makes no sense whatsoever. If the sd card is plugged in to your laptop it's just a storage medium. It's not running any code and certainly can't connect to your wifi.

mhashemi2, 2021-05-31:
I found that I have an HDMI/mini-HDMI cable in my collection. When I boot, I see my IP is 169.254.192.216 and that OpenBSD Secure Shell server is started, then the script seems to take over and get_script fails with, "Could not resolve host: raw.githubusercontent.com". 

I'm going to pick up an adapter so I can plug my keyboard into the micro-USB port on the Pi, so I can interact with it.

rmyadsk, 2021-05-31:
That 169.* address is a fallback IP address that occurs when the machine can't obtain an actual IP address - there is a fault in your networking config.

By default the system should try obtain an address via DHCP - so unless you're overriding that somewhere (say, /etc/dhcpcd.conf) I suspect the SSID is getting scrambled

mhashemi2, 2021-05-31:
Yeah, I can see dhcpd starting. 

I have a keyboard now. How can I cancel rc.local (I assume that's what's trying to reach githubusercontent.com)? Ctrl-C isn't breaking.

mhashemi2, 2021-05-31:
Well, I don't know what's up, but I updated Etcher and re-flashed the SD card again (I used the same version of Etcher, but on a different computer yesterday). After that, the setup went smooth.

closed 2021-05-31 by mhashemi2