marcone/teslausb #642: verify-and-configure-archive: STOP: no working combination of vers and sec mount options worked

Michaelillingby, 2021-06-05:
verify-and-configure-archive: STOP: no working combination of vers and sec mount options worked

Can anyone off advice to get past this ?

Michaelillingby, 2021-06-05:
pi@teslausb:/boot $ sudo -i

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

root@teslausb:~# /etc/rc.local 
My IP address is 10.0.1.220 
Detecting whether to update wpa_supplicant.conf
Remounting / as read/write
Remounting /boot as read/write
Done. Make any changes needed, and reboot the Pi when ready.
Filesystems / and /boot will be mounted as read only after reboot.
Starting setup.
/root/bin/setup-teslausb is up to date
rc.local is up to date
Updating package index files...
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Verifying that the requested configuration is valid...
Downloaded /tmp/verify-configuration.sh ...
USB_DRIVE is not set. SD card will be used.
Verifying that there is sufficient space available on the MicroSD card...
There is sufficient space available.
Downloading additional setup scripts.
Downloaded /root/bin/remountfs_rw ...
Downloaded /root/bin/make_snapshot.sh ...
Downloaded /root/bin/mount_snapshot.sh ...
Downloaded /root/bin/release_snapshot.sh ...
Downloaded /root/bin/force_sync.sh ...
Downloaded /tmp/create-backingfiles-partition.sh ...
Downloaded /tmp/create-backingfiles.sh ...
Downloaded /tmp/make-root-fs-readonly.sh ...
Downloaded /tmp/configure.sh ...
~ ~
Fixing the modules-load parameter in /boot/cmdline.txt...
Fixed cmdline.txt.
USB_DRIVE not set. Proceeding to SD card setup
create-backingfiles-partition: using existing backingfiles and mutable partitions
umount: /mnt/cam: not mounted.
calling configure.sh
configure: Pushover not enabled.
configure: Gotify not enabled.
configure: IFTTT not enabled.
configure: Webhook not enabled.
configure: Slack not enabled.
configure: Matrix not configured.
configure: Telegram not enabled.
configure: SNS not configured.
Downloaded /root/bin/send-push-message ...
Downloaded /root/bin/send_sns.py ...
Downloaded /root/bin/send_matrix.py ...
configure: Checking archive configs: 
building file list ... done
testfile.dat

sent 105 bytes  received 43 bytes  296.00 bytes/sec
total size is 9  speedup is 0.06
configure: rsync seems to work OK
configure: done
configure: Using archive module: run/cifs_archive
configure: Installing base archive scripts into /root/bin
Downloaded /root/bin/archiveloop ...
Downloaded /root/bin/waitforidle ...
Downloaded /root/bin/remountfs_rw ...
configure: Skipping tesla_api.py install because no credentials were provided
configure: Installing archive module scripts
Downloaded /tmp/verify-and-configure-archive.sh ...
Downloaded /root/bin/archive-clips.sh ...
Downloaded /root/bin/connect-archive.sh ...
[... 89 lines cut ...]
verify-and-configure-archive: mount -t cifs '//10.0.1.10/teslausb' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=2.0,sec=ntlmv2'
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
verify-and-configure-archive: Trying mount command-line:
verify-and-configure-archive: mount -t cifs '//10.0.1.10/teslausb' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=2.0,sec=ntlm'
mount error(2): No such file or directory
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
verify-and-configure-archive: Trying mount command-line:
verify-and-configure-archive: mount -t cifs '//10.0.1.10/teslausb' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=1.0,'
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
verify-and-configure-archive: Trying mount command-line:
verify-and-configure-archive: mount -t cifs '//10.0.1.10/teslausb' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=1.0,sec=ntlmssp'
mount error(13): Permission denied
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
verify-and-configure-archive: Trying mount command-line:
verify-and-configure-archive: mount -t cifs '//10.0.1.10/teslausb' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=1.0,sec=ntlmv2'
mount error(2): No such file or directory
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
verify-and-configure-archive: Trying mount command-line:
verify-and-configure-archive: mount -t cifs '//10.0.1.10/teslausb' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=1.0,sec=ntlm'
mount error(2): No such file or directory
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
verify-and-configure-archive: STOP: no working combination of vers and sec mount options worked
root@teslausb:~#

marcone, 2021-06-05:
Does your server actually have a share named 'teslausb' ? (you can verify this by installing "smbclient" on the Pi and running `smbclient -L 10.0.1.10 -N`)

Michaelillingby, 2021-06-05:
Maybe not. Since it isn't working, I may have done this wrong. I am using Mac OS 10.15.7 Catalina for my Local Server. 
I did create shares using the credentials that match the CONF file, but maybe they are not setup correctly or just incompatible. Do you know the correct steps I need to follow to create the correct shared?

marcone, 2021-06-05:
Make sure you check the option to enable Samba sharing, otherwise it will use Apple sharing.
Not sure if it's the same in Catalina, but in Big Sur the option is called "Share files and folders using SMB"

Michaelillingby, 2021-06-05:
Still getting the same errors.

[image]
[image]

Michaelillingby, 2021-06-05:
that ONE connected user Is my other computer that I connected to this one with to test the share. This is another MAC, so no surprise that it works, but can't get teslacam to work :(

Michaelillingby, 2021-06-05:
okay, I tried something else, and got this. I had to drop the -N and instead use the -U teslausb and then provide the password. this is the result.

Is this helpful in any way?

[pi@teslausb:~ $ smbclient -L 10.0.1.10 -U teslausb
Unable to initialize messaging context
Enter WORKGROUP\teslausb's password: 

	Sharename       Type      Comment
	---------       ----      -------
	8TB             Disk      
	IPC$            IPC       
	Michael Illingby’s Public Folder Disk      
	teslamusic      Disk      
	teslausb        Disk      
	teslausb        Disk      
Reconnecting with SMB1 for workgroup listing.
do_connect: Connection to 10.0.1.10 failed (Error NT_STATUS_CONNECTION_REFUSED)
Failed to connect with SMB1 -- no workgroup available

marcone, 2021-06-05:
That appears to be showing that you have two shares both named teslausb. Maybe that's the problem?

rmyadsk, 2021-06-06:
I also run MacOS and have found Apple's implementation of the SMB protocol to be deficient to the point of regular failure. I would strongly advise MacOS users to look into the rsync method as an alternative, as it leverages the built-in underpinnings on MacOS (rsync and SSH) which itself has its roots in BSD. I have found rsync to be bulletproof for teslausb transfers (though unfortunately my vehicles still tend to fall asleep early and often).

The rsync method is well documented, but I can expand on a couple things if anyone has questions.

Michaelillingby, 2021-06-06:
yes please explain how to do the rsync, or provide links please? I am not in love with any transfer method, I just wan it to work. 😃

rmyadsk, 2021-06-06:
[as requested](https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRSync.md) - this should get you started. You're going to get a bit in the weeds creating keys, but this is to bypass password authentication so your pi can operate autonomously.

If you're not comfortable in a Linux shell, go as far as you can and tell me where you're hung up. We'll get you there.

rmyadsk, 2021-06-06:
First gremlin not explicitly called out in the documentation: for MacOS, you'll want to enable remote login - this starts the SSH service. Go to Apple icon -> System Preferences -> Sharing and check "Remote Login".

Michaelillingby, 2021-06-06:
I'll check into this tomorrow, hope that it works. I am not a newbie to linux, but have never used rsync before. "Remote login" ON. hehehe thanks Ryan

marcone, 2021-06-10:
Closing this since it's more of a "getting linux to work with MacOS file shares" issue, and not really a teslausb issue

closed 2021-06-10 by marcone