marcone/teslausb #643: TeslaUSB - Sync Music Only

jun3280net, 2021-06-11:
Hi @marcone, I was wondering if there's a way to sync just the music and NOT the video.  I'm noticing the videos never sync with my storage because the Tesla sleeps within minutes by the time I get home.  Or  it could be that my wifi connection speed is too slow - is there a way to check transfer speed?  It is however fast enough for synching the music I find.

That said, the web interface you've designed is fantastic for viewing files, and I think I'm satisfied with just leaving the videos on the sdcard now.  Adding additional functionality to the web interface to drop snapshots to recover disk space would be perfect?  To be honest, I'm not entirely sure how the sd card space is managed.  I have a 256GB card, 128GB for TeslaCam, 10GB for Music, is the remainder of the space used for Snapshots?

Thanks for all the great work!

marcone, 2021-06-11:
Perhaps I misunderstood what you meant, but what you describe can't happen. Teslausb syncs music after it finishes archiving, so if the car goes to sleep during archiving, it will not sync music.
The remaining space is indeed used for snapshots, and snapshots are managed automatically by teslausb so it doesn't seem super useful to manage them manually through the web interface.

jun3280net, 2021-06-13:
> Perhaps I misunderstood what you meant, but what you describe can't happen. Teslausb syncs music after it finishes archiving, so if the car goes to sleep during archiving, it will not sync music. [...]

Hi @marcone, thanks for the response.  Synching music is only a few hundred MBs, so it should finish before Teslausb sleeps.  However, the video files are a few GBs which is why archiving never finishes.   As the web interface you developed works so well, I'm not sure there's any value in synching the video to the NAS for me.  I'd rather Teslausb be used to synch my music updates instead now going forward if that's possible.  

Or is the snapshot management part of the archiving process?  On that note, is there a way to have Teslausb send an MQTT command perhaps (archive start and archive end), where I could leverage Home Assistant/NodeRED to keep Tesla awake until archive is completed?

Thanks again for your great work!

marcone, 2021-06-16:
> is there a way to have Teslausb send an MQTT command perhaps

Maybe the IFTTT or generic webhook messaging would work for that?

jun3280net, 2021-06-16:
> > is there a way to have Teslausb send an MQTT command perhaps
> 
> Maybe the IFTTT or generic webhook messaging would work for that?

@marcone good suggestion.  I will look into seeing if I can get HA/NODERED to see if it can leverage the webhooks to trigger actions.  

On that note - do you know what would cause TESLAUSB not to mount in the Tesla?  I even tried unplugging and re-plugging in the device while driving.  However, when I returned home, I could confirm I could connect to Teslausb via your web interface - so it appears it is working.

Screenshot noted below : 
[image]

Diagnostics : 
`Jun 15 22:04:11 teslausb daemon.info ntpd[456]: Soliciting pool server 205.206.70.7
Jun 15 22:04:11 teslausb daemon.info ntpd[456]: Soliciting pool server 85.195.227.162
Jun 15 22:04:12 teslausb daemon.info ntpd[456]: Soliciting pool server 157.230.86.26
Jun 15 22:04:12 teslausb daemon.info ntpd[456]: Soliciting pool server 138.197.135.239
Jun 15 22:04:12 teslausb daemon.info ntpd[456]: Soliciting pool server 193.30.120.245
Jun 15 22:04:12 teslausb daemon.info ntpd[456]: Soliciting pool server 129.250.35.250
Jun 15 22:04:13 teslausb daemon.info ntpd[456]: Soliciting pool server 62.197.224.14
Jun 15 22:04:13 teslausb daemon.info ntpd[456]: Soliciting pool server 96.235.18.130
Jun 15 22:04:13 teslausb daemon.info ntpd[456]: Soliciting pool server 212.51.144.44
Jun 15 22:04:13 teslausb daemon.info ntpd[456]: Soliciting pool server 162.159.200.1
Jun 15 22:04:14 teslausb daemon.info ntpd[456]: Soliciting pool server 80.74.64.2
Jun 15 22:04:14 teslausb daemon.info ntpd[456]: Soliciting pool server 108.59.2.24
Jun 15 22:04:14 teslausb daemon.info ntpd[456]: Soliciting pool server 45.79.51.42
Jun 15 22:04:15 teslausb daemon.info ntpd[456]: Soliciting pool server 159.203.8.72
Jun 15 22:04:15 teslausb daemon.info ntpd[456]: Soliciting pool server 69.164.213.136
Jun 15 22:07:10 teslausb authpriv.notice sudo: www-data : TTY=unknown ; PWD=/var/www/html/cgi-bin ; USER=root ; COMMAND=/root/bin/setup-teslausb diagnose
Jun 15 22:07:10 teslausb authpriv.info sudo: pam_unix(sudo:session): session opened for user root by (uid=0)
====== dmesg ======
[Tue Jun 15 22:02:34 2021] Booting Linux on physical CPU 0x0
[Tue Jun 15 22:02:34 2021] Linux version 5.4.79+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1373 Mon Nov 23 13:18:15 GMT 2020
[Tue Jun 15 22:02:34 2021] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[Tue Jun 15 22:02:34 2021] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[Tue Jun 15 22:02:34 2021] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[Tue Jun 15 22:02:34 2021] Memory policy: Data cache writeback
[Tue Jun 15 22:02:34 2021] Reserved memory: created CMA memory pool at 0x17c00000, size 64 MiB
[Tue Jun 15 22:02:34 2021] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[Tue Jun 15 22:02:34 2021] On node 0 totalpages: 114688
[Tue Jun 15 22:02:34 2021]   Normal zone: 1008 pages used for memmap
[Tue Jun 15 22:02:34 2021]   Normal zone: 0 pages reserved
[Tue Jun 15 22:02:34 2021]   Normal zone: 114688 pages, LIFO batch:31
[Tue Jun 15 22:02:34 2021] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[Tue Jun 15 22:02:34 2021] pcpu-alloc: [0] 0 
[Tue Jun 15 22:02:34 2021] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[Tue Jun 15 22:02:34 2021] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:7D:93:52 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=3f98dc8b-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2 quiet fastboot noswap ro
[Tue Jun 15 22:02:34 2021] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[Tue Jun 15 22:02:34 2021] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[Tue Jun 15 22:02:34 2021] mem auto-init: stack:off, heap alloc:off, heap free:off
[Tue Jun 15 22:02:34 2021] Memory: 376184K/458752K available (7814K kernel code, 712K rwdata, 2468K rodata, 512K init, 804K bss, 17032K reserved, 65536K cma-reserved)
[Tue Jun 15 22:02:34 2021] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[Tue Jun 15 22:02:34 2021] ftrace: allocating 27740 entries in 55 pages
[Tue Jun 15 22:02:34 2021] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[Tue Jun 15 22:02:34 2021] random: get_random_bytes called from start_kernel+0x2a8/0x4d8 with crng_init=0
[Tue Jun 15 22:02:34 2021] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[Tue Jun 15 22:02:34 2021] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[Tue Jun 15 22:02:34 2021] bcm2835: system timer (irq = 27)
[Tue Jun 15 22:02:34 2021] Console: colour dummy device 80x30
[Tue Jun 15 22:02:34 2021] printk: console [tty1] enabled
[Tue Jun 15 22:02:34 2021] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[Tue Jun 15 22:02:34 2021] pid_max: default: 32768 minimum: 301
[Tue Jun 15 22:02:34 2021] LSM: Security Framework initializing
[Tue Jun 15 22:02:34 2021] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[Tue Jun 15 22:02:34 2021] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... cut ...]
root      2935   105  4 22:07 ?        00:00:00 /lib/systemd/systemd-udevd
root      2942   105  2 22:07 ?        00:00:00 /lib/systemd/systemd-udevd
root      2943   105  3 22:07 ?        00:00:00 /lib/systemd/systemd-udevd
root      2944   105  3 22:07 ?        00:00:00 /lib/systemd/systemd-udevd
root      2958   236  0 22:07 ?        00:00:00 sleep 1
root      2966  2897  0 22:07 ?        00:00:00 ps -eaf
teslausb has been up 4 minutes. System time is Tue 15 Jun 22:07:14 MDT 2021
====== end of diagnostics ======`

marcone, 2021-06-16:
I think you only copied part of the diagnostics, since a large part is missing. You might try plugging the Pi in to a PC and see if the CAM drive shows up there.

jun3280net, 2021-06-16:
> I think you only copied part of the diagnostics, since a large part is missing. You might try plugging the Pi in to a PC and see if the CAM drive shows up there.

Thanks for the quick reply @marcone.  I've uploaded the full diagnostics.  

On that note, plugging it into the computer will mount and fixes the problem temporarily, until it longer mounts in the Tesla again.  Is it possible this is a result of the archiving process getting interrupted and not completing?

marcone, 2021-06-16:
That shouldn't be an issue. The only odd thing i see in your diagnostics is that there were two instances (on June 9 and 14) where the "autofs" system service apparently didn't start up. That would prevent teslausb from working correctly.

jun3280net, 2021-06-16:
> That shouldn't be an issue. The only odd thing i see in your diagnostics is that there were two instances (on June 9 and 14) where the "autofs" system service apparently didn't start up. That would prevent teslausb from working correctly.

@marcone thanks for taking the time to review the logs.  Those would be the two dates that recently had problems.  Unplugging the device for 30 sec, then plugging it back into the Tesla will not fix the issue.  Plugging into the computer however has fixed the problem though.  Should I reinstall the device perhaps - although this was just reinstalled at the beginning of the month.

jun3280net, 2021-06-19:
@marcone  hooking the Teslausb back to the computer works - but plugging it back into the car didn't work.  I ended up reinstalling, and that fixed it - it loaded up in 30 seconds.  Can I ask if you use CIFS or RSYNC?  From experience Teslausb seems to work well for a couple weeks, then something causes it to not to mount when connected to the Tesla (although it works fine connected to the computer).  The only thing I noticed when I SSH into the device, were 3-4 folders in the MUTABLE folder that did not adhere to the SNAPSHOT naming convention.  Those looked like errors.  Could that be causing the problem?

marcone, 2021-06-19:
The Pi just acts as a USB storage device. It has no real knowledge of the host, so it shouldn't make a difference whether it's connected to a PC or to the car. It publishes the same drive in both cases.
I'm not sure what you mean about folders in the mutable partition not adhering to the snapshot naming convention. The snapshots are in the backingfiles partition, there are no snapshots in the mutable partition.

jun3280net, 2021-06-20:
@marcone, sorry I misspoke.  I was referring to the Recent Drives folder.  The folders are usually a date stamp (eg 2021-06-18), but what I noticed some weird folder names that did not adhere to that standard.  I was curious that might cause the problem.  I had captured these weird folders in a previous post.
[image]
  
Tesla must be sensitive to something running in the PI that the Windows PC is not.  I've even tried rebooting the MCU in the Tesla to confirm if its the car.  When things work in the Tesla - it's usually up within 30 seconds.  Plugging into the computer, it is always up in 30 seconds. 

Since I've reinstalled from the 20201211 image - things seem to be working ok for now.  I'll hold up on updating for now to see if this mounting problem shows up again.  
[image]

Thanks for your insights

jun3280net, 2021-06-24:
Reinstall seems to have fixed most of the issues I was experiencing.  I suppose configuring Teslausb at setup no to synch any of the videos would have achieved my original request - but I never tried.

The new online viewer eliminates the need to synch though.

On that note - @marcone is there a way to access the snapshots directly plugging the PI into a computer?  For instance if I notice something from the Sentry capture, but I want to see observe the details around the time of the event - I would need to check the snapshot right?

closed 2021-06-24 by jun3280net

marcone, 2021-06-24:
> is there a way to access the snapshots directly plugging the PI into a computer?

no, there isn't. The recordings in the snapshots are accessible via Samba or the web interface.