marcone/teslausb #644: Viewer bugs / questions

marcone, 2021-06-13:
This is the main place for bugs and issues with the the web interface viewer. **All other reported viewer issues will be closed**. Feel free to comment here. Pull requests for improvements are welcome.

Know issues:
- play/pause button doesn't toggle state in Firefox. This is because Firefox [doesn't allow changing SVG paths in CSS](https://bugzilla.mozilla.org/show_bug.cgi?id=1571119), and I didn't feel like working around browser bugs too much, especially when the browser only has a few percent market share.
- the Sentry location map sometimes doesn't scale right in Firefox. This appears to be another Firefox bug. Use Chrome.
- videos are not always synchronized. 

Anticipated questions:
- Why Bing maps and not Google or Openstreetmap? Google doesn't allow embedding maps unless you sign up as a "cloud developer", and restricting the API key doesn't seem feasible for the way teslausb needs to use it. Openstreetmap doesn't have satellite view.
- Why no location for non-Sentry videos? Tesla doesn't put location metadata in the recording files, only in the "event.json" file, and that file is only created for Sentry events.

maxhac03, 2021-06-14:
I like the new player!
Suggestion: Allow the choice to invert or not the back and side cameras.

rmyadsk, 2021-06-14:
Very cool addition. I wonder if navigation arrows (or perhaps just a back arrow) could be added to video the windows containing video clips, for ease of returning to the directory structure? Once a clip is viewed, to get back to the previous page I need to right-click over the grey border space and select "back". (Client is Safari on MacOS)

marcone, 2021-06-15:
> Very cool addition. I wonder if navigation arrows (or perhaps just a back arrow) could be added to video the windows containing video clips, for ease of returning to the directory structure? Once a [...]

It sounds like you're talking about the "recordings" tab, not the "viewer" tab that this issue is about.

rmyadsk, 2021-06-16:
Ah, my mistake. Everyone please be sure to run a "/root/bin/teslausb-setup upgrade" from the shell if you don't see a "Viewer" tab to the right of "Recordings" in the web console; this appears to be a new (and excellent) feature.

miles267, 2021-06-16:
Can we also please get the ability to keep the teslaUSB awake with the new 2FA? At this point, the pi can no longer finish before the unit sleeps. The cycle repeats. The only way to get a complete successful sync is by removing the pi from the car and plugging it into the wall.

marcone, 2021-06-16:
> Can we also please get the ability to keep the teslaUSB awake with the new 2FA? At this point, the pi can no longer finish before the unit sleeps. The cycle repeats. The only way to get a complete successful sync is by removing the pi from the car and plugging it into the wall.

That's got nothing to do with the viewer, and there's already an [open issue](https://github.com/marcone/teslausb/issues/384) for this. If you don't want to take the Pi out of the car, you can use the Tesla app to turn on Sentry or climate control to keep the USB ports powered.

joanquro, 2021-06-16:
> Ah, my mistake. Everyone please be sure to run a "/root/bin/teslausb-setup upgrade" from the shell if you don't see a "Viewer" tab to the right of "Recordings" in the web console; this appears to be a new (and excellent) feature.

Hello. I've done an upgrade and I don't see the new tab. Should I do something else?.

Thanks.

jun3280net, 2021-06-20:
> Hello. I've done an upgrade and I don't see the new tab. Should I do something else?.
>
> Thanks.

@joanquro I had tried tesla-setup upgrade to try and address other issues I was experiencing, and did not notice this new Viewer. 

I did a fresh install of the 20201211 image, and the Viewer is now there.  Pretty slick stuff.  I mentioned this in a different thread, but the new web interface negates the need to sync the videos back to server now.

czyz, 2021-11-10:
Two notes -- 

Note 1 -- suggested addition to the documentataion: How to change settings of an existing setup:

1. sudo -i
2. bin/remountfs_rw
3. edit /root/teslausb_setup_variables.conf
4. /root/bin/setup-teslausb

At least I think that's the process I ended up doing. If that's the official method it might be useful to add it to the documentation or FAQ.

Note 2 -- can I just take the SD card out of my pi zero w and put it in a pi zero w 2 and expect it to work? I'm asking, of course, after I already tried this and it failed, but I don't know why. How to transition ones' setup from on piece of pi hardware to a replacement would be a nice piece of info for the FAQ.

Thanks.

marcone, 2021-11-11:
> PS. Oops, sorry this is in the wrong thread. Not sure where it should be moved to.

Opening a new issue is the usual process

marcone, 2022-09-26:
Closing this since there haven't been many reported issues with the viewer. Please file any bugs as separate issues from now on.