marcone/teslausb #645: Password for the web interface

maxhac03, 2021-06-16:
I'm not knowledgeable enough to do it myself and not breaking anything so i'm asking here.
Could it be possible to configure a password on the web interface?

In my case, i had my home wifi and my work's guest wifi network configured in my pi so i could downloads saved recording while at work.

Now with the web interface open to anyone on the network i had to remove the wifi network from my pi.

Thanks.

rmyadsk, 2021-06-16:
I don't believe wifi credentials appear in the diagnostic logs or are otherwise reachable via the web interface, even if you do use the "hotspot" switch to serve a local wifi network.

That being said, it should be possible to password-protect the web interface using .htaccess and a modification of nginx config.

marcone, 2021-06-16:
I have no interest in implementing this myself, but will gladly accept a PR.

maxhac03, 2021-06-16:
> I don't believe wifi credentials appear in the diagnostic logs or are otherwise reachable via the web interface, even if you do use the "hotspot" switch to serve a local wifi network.
>
> That being [...]

It was more to not allow anybody in the same network to see the recordings.

I do have some basic nginx knowledge and maybe i could figure something if i take enough time.

rmyadsk, 2021-06-16:
> It was more to not allow anybody in the same network to see the recordings.
>
> I do have some basic nginx knowledge and maybe i could figure something if i take enough time.

This should do it for you:

http://www.digitalocean.com/community/tutorials/how-to-set-up-password-authentication-with-nginx-on-ubuntu-14-04

maxhac03, 2021-06-17:
> This should do it for you:
> 
> http://www.digitalocean.com/community/tutorials/how-to-set-up-password-authentication-with-nginx-on-ubuntu-14-04

It's working! Thanks!

When is was trying earlier, i was using the wrong nginx config file (using another guide).

If somebody else want to do it too, here is what the nginx config should be:
/etc/nginx/sites-enabled/default
`server {
    listen 80 default_server;
    listen [::]:80 default_server;

    root /var/www/html;

    index index.html

    server_name _;

    location / {
        try_files $uri $uri/ =404;
        auth_basic "Restricted Content";
        auth_basic_user_file /etc/nginx/.htpasswd;
    }

    location /cgi-bin/ {
         gzip off;
         root /var/www/html;
         fastcgi_pass  unix:/var/run/fcgiwrap.socket;
         include /etc/nginx/fastcgi_params;
         fastcgi_param SCRIPT_FILENAME  $document_root$fastcgi_script_name;
    }
    location /TeslaCam/ {
         root /var/www/html;
         fancyindex on;
    }
}`

boba51, 2021-08-04:
Wow, not sure how I missed the Web Interface up to now, how great :)

Raiders95, 2021-08-22:
Sorry for the dumb question, but how does one access the web interface, and is there any special configuration that needs to be setup first?

rmyadsk, 2021-09-06:
> Sorry for the dumb question, but how does one access the web interface, and is there any special configuration that needs to be setup first?

@Raiders95 sorry, I somehow missed your question here. Just visit your pi's IP address in your Web browser. If you have a recent version of teslausb installed, no additional configuration is necessary (unless, for example, you want to password-protect the directory as above).

closed 2022-12-19 by marcone