marcone/teslausb #650: Does teslausb store videos over the hour time frame

zikronix, 2021-06-30:
I couldnt find the answer to this. Does this allow the videos that arnt saved to persist past and hour, and does the cam eventually overwrite those files. If not is there a way to make it do these things, like a traditional dash cam.

marcone, 2021-06-30:
It does this already. Get a large sd card (I recommend at least 128 GB) and leave CAM_SIZE set to the default (30).

closed 2021-06-30 by marcone

zikronix, 2021-06-30:
Cool thanks! I couldn’t find information that said that.

So what does bat 30 option do how long will it keep the clips or does it use a first in first out method for overwriting them?

I also couldn’t figure out how to not off load the files. I’d rather they just stay on the sd card and get over written. I tired by not configuring the shares but it didn’t seem to work.

Last but not least how do I configure the web browser. When the WiFi comes up and I try to hit the  ip 192.168.66.1 no web page shows up.

Any help would be great thanks!

Get Outlook<https://aka.ms/qtex0l> for iOS

marcone, 2021-06-30:
> So what does bat 30 option do

That should be "30G" (I've corrected my original reply). It basically tells teslausb to make  a 30 GB drive available to the car.

> how long will it keep the clips or does it use a first in first out method for overwriting them?

Teslausb will create regular copies of the 30GB drive that the car sees, which allows keeping recordings older than an hour. These copies are managed by teslausb and are first in first out. With a 128 GB card and 30 GB CAM_SIZE, I'd expect about 10-12 hours of actual recording time. Note that when Sentry is on, the car is always recording.

> I also couldn’t figure out how to not off load the files. 

You can use `export ARCHIVE_SYSTEM=none` in the config

> Last but not least how do I configure the web browser. When the WiFi comes up and I try to hit the ip 192.168.66.1 no web page shows up.

192.168.66.1 is the Pi's IP when you are connected to the Pi's AP. You probably want to connect to "teslausb" or "teslausb.local" instead.

zikronix, 2021-06-30:
Thanks!

Get Outlook<https://aka.ms/qtex0l> for iOS