marcone/teslausb #651: Unstable connection problem with iPhone hotspot

missflash, 2021-07-01:
Thank you for your awesome program.
It's really great!

I finished TeslaUSB setup by following your one step setup instructions.
(In my parking lot, there is no internet connection. So I registered my iPhone's hotspot.)

After then, I installed rclone using my iPhone's hotspot.

However, it was pretty hard to connect. I had to try several times. At first, it looked like well connected, but disconnected after then.

I never have any problems to use my hotspot. (Laptop, WiFi version iPad, and so on) It's so weird.

Do you have any ideas or solutions?

rmyadsk, 2021-07-01:
I think you may be misunderstanding the concept of the hotspot. In the context of teslausb setup, invoking a hotspot means the Raspberry pi acts as a hotspot (as a Wifi access point), but your narrative seems to point to using your iPhone's hotspot for connectivity, using the pi as a client.

If this is true, you would want to make sure your iPhone's hotspot functionality is enabled, then configure the pi to connect to it as a client using the SSID and WIFIPASS variables.

missflash, 2021-07-01:
If this is true, you would want to make sure your iPhone's hotspot functionality is enabled, then configure the pi to connect to it as a client using the SSID and WIFIPASS variables.

-> Correct. I already turned on my iPhone's hotspot. Also I added correct SSID and WIFIPASS variables at pi's conf file.

However, the connection was not stable enough. Sometimes no problem, but other times couldn't. So I couldn't look into raspberry via ssh.

BTW, I want to check pi's status because my rclone didn't works well.

marcone, 2021-07-01:
The fact that you can't ssh in to the Pi when it is connected to the phone's hotspot could mean several things. For example, the Pi could not actually be connected to the phone, or the phone could be isolating different clients from each other, so that for example you can't connect from your laptop to the Pi via the phone.

I would try setting up your Pi with your (hopefully more stable) home wifi first, make sure everything works correctly, then add your phone's wifi AP info to /etc/wpa_supplicant/wpa_supplicant.conf. That way the Pi will be able to connect to both, and it'll make debugging a lot easier.

In any case, this isn't really a teslausb issue, but more of a generic Raspberry Pi and wifi issue, so I'm closing this.

closed 2021-07-01 by marcone

missflash, 2021-07-01:
Thank you! I'll do that!