marcone/teslausb #652: How to use custom port for cifs?

missflash, 2021-07-02:
I want to use custom port (1234) for cifs.
The following script was already tested and works fine.
`mount -t cifs -o **port=1234**,user='USERID',password='PASSWORD' //MYHOME.duckdns.org/TeslaUSB /mnt/archive`

I modified teslausb_setup_variables.conf file, but couldn't find port option.
```
export ARCHIVE_SYSTEM=cifs
export ARCHIVE_SERVER=MYHOME.duckdns.org
export SHARE_NAME=TeslaUSB
export SHARE_USER=USERID
export SHARE_PASSWORD=PASSWORD
```

Because of that, pi can't access to my NAS as shown in the following,

Fri  2 Jul 22:02:59 KST 2021: Starting archiveloop at 23.87 seconds uptime...
Fri  2 Jul 22:03:00 KST 2021: taking snapshot of cam disk in /backingfiles/snapshots/snap-000008
Fri  2 Jul 22:03:05 KST 2021: waiting for autofs to be active
Fri  2 Jul 22:03:08 KST 2021: took snapshot
Fri  2 Jul 22:03:13 KST 2021: comparing new snapshot with /backingfiles/snapshots/snap-000007/snap.bin
Fri  2 Jul 22:03:13 KST 2021: new snapshot is identical to previous one, discarding
Fri  2 Jul 22:03:17 KST 2021: Running fsck on /backingfiles/cam_disk.bin...
Fri  2 Jul 22:03:18 KST 2021: | fsck from util-linux 2.33.1
Fri  2 Jul 22:03:19 KST 2021: | fsck.fat 4.1 (2017-01-24)
Fri  2 Jul 22:03:19 KST 2021: | /dev/loop0p1: 215 files, 186962/1536097 clusters
Fri  2 Jul 22:03:19 KST 2021: Finished fsck on /backingfiles/cam_disk.bin.
Fri  2 Jul 22:03:20 KST 2021: Connecting usb to host...
Fri  2 Jul 22:03:20 KST 2021: Connected usb to host.
**Fri  2 Jul 22:03:20 KST 2021: Waiting for archive to be reachable...**

marcone, 2021-07-02:
teslausb assumes cifs is running on the standard port. If you're doing something custom you'll have to modify the scripts. In this case it's "Waiting for archive to be reachable" forever because the `archive-is-reachable.sh` script probes port 445.

closed 2021-07-02 by marcone

missflash, 2021-07-02:
Thank you! That's what I want to find! 😆