marcone/teslausb #654: Teslausb stops mounting in Tesla after a few weeks

jun3280net, 2021-07-07:
Teslausb installed on Pi-Zero works great after a fresh install.  After a couple of weeks, the device begins to experience problems with mounting in the Tesla.  It's not clear to me whether this is a result of Teslausb being turned off (Tesla Sleeps) during "archiveloop" process when connected at home.  When the car wakes from sleep to drive, Teslausb starts back up, but often does not mount in the Tesla.  The lights flash - as normal, but no mounting in the car after a few minutes.  Sometimes, if I disconnect Teslausb, and reconnect, it will mount (but this is rare).  I've tried rebooting the Tesla MCU - but this makes no difference.

However, while connected to the Tesla (not mounted), I can confirm I can connect to the AP, and even access the new webpage/viewer.  I can even reboot from the webpage, (I don't think archiveloop can be forced unless it's connected to the home SSID).  Is there perhaps a way to force mount the device, as I can SSH into the device via the AP.  If there is a command to do this, can it be built into the Viewer?

It should be noted while Teslausb does not mount in the car, I can confirm it mounts when connected to the computer.  It takes about 1 min before the CAM and MUSIC drives show up.

Update : TeslaUSB is plugged into the car via a HUB.  The device was has not been mounting as described.  Ten minutes into the drive, I plugged in a spare USB key into the HUB to ensure dashcam was recording.  Surprisingly, this triggered the mounting of TeslaUSB.  I confirmed via the viewer some video recorded today, and the music was available to the car.   I have not tried replicating this since - but I'll try again tomorrow assuming Teslausb does not mount again.

Any ideas or thoughts on how to troubleshoot would be greatly appreciated.  Thanks

[archiveloop.log](https://github.com/marcone/teslausb/files/6774318/archiveloop.log)
[diagnostics.txt](https://github.com/marcone/teslausb/files/6774319/diagnostics.txt)
[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/6774320/teslausb-headless-setup.log)

marcone, 2021-07-11:
You didn't say when you took the diagnostics, but they show that the attached host mounted the drives less than 2 minutes before.
There were a few times on previous days when the "autofs" system service failed to start, blocking teslausb, but I don't know what would cause that.

jun3280net, 2021-07-13:
> You didn't say when you took the diagnostics, but they show that the attached host mounted the drives less than 2 minutes before.
> There were a few times on previous days when the "autofs" system service failed to start, blocking teslausb, but I don't know what would cause that.

Thanks for looking into this @marcone.  I pulled those files from the Teslausb Viewer on July 6, 2021 at 4:28PM.  Is there a way to identify whether the mounting was on the computer versus on the Tesla?  I guess if it mounted to the Tesla, it would show files being written/processed in the TeslaCam folders.  

It's definitely bizarre that the device works fine for a couple weeks before this mounting issue begins to plague the device.

mane-wt, 2021-07-26:
I got similiar problems after firmware update 2021.12.25.6, worked perfectly before.
It worked once after I made a software update via ssh but then it does not show up in the car anymore.

mane-wt, 2021-08-01:
I have updated the car to 2021.12.25.7 but it is still the same behavior, i.e. after a connection to a computer both Music and TeslaCAM is mounted every time. I the do a apt update; apt upgrade and the put the Rasperry PI Zero back in the car. FIrst startup works fine but next time it lokks that the PI is not installed again. Just to be sure I tested with an ordinaru USB-stick and tha works (no surprise). I enclose my diagnostic.txt that was readout 29th of july 2021 at 8 a.m.
[diagnostics.txt](https://github.com/marcone/teslausb/files/6913001/diagnostics.txt)

jun3280net, 2021-08-01:
> I have updated the car to 2021.12.25.7 but it is still the same behavior, i.e. after a connection to a computer both Music and TeslaCAM is mounted every time. I the do a apt update; apt upgrade and [...]

@mane-wt  Great that you are replicating the steps I did to debug as well.  @marcone is there a way to increase logging specifically to the mounting process to see why it works with a computer versus the Tesla?  The next test I am going to do is run Teslasub only for TeslaCAM (No Music) - maybe there's something weird going on with handling two drives?

I've been testing running just a normal microsd card in the Tesla without issues for a couples weeks now, and synching my music collection from my NAS to my phone using free Android synching software.

Poncherelly, 2021-08-01:
Before removing music, can you check to see if USB is mounted in the car even if the cam isn't working?  Mine is doing this and a do the double wheel reboot and the cam starts right up.

Reply-to: ***@***.***
Cc: ***@***.***

marcone, 2021-08-01:
@mane-wt your diagnostics show the Pi is making the drive available to the car, but the car chose not to access it:
```
USB state (350.4 seconds ago): mass storage ready, but not connected to host (check cable)
```

@jun3280net The Pi just emulates a USB drive. It has no visibility into whether it's connected to a PC or a car.

boba51, 2021-08-01:
Just to add my experience and offer any assistance with logs etc.:

My installation has been working fine for some time but has stopped after my M3 LR was recently upgraded to 2021.12.25.6.
I have:
Rebooted my M3 both in and out of WiFi range - no change.
Tried the Tesla supplied USB drive which works straight away, but re-inserting the TeslaUSB still - no change.
Further Info:
USB drives mount and work fine when connected to PC.
I'm also syncing Music.

Please let me know if I can provide any more information, unfortunately technically I will need guidance, but happy to assist where I can.

Bob

mane-wt, 2021-08-02:
After some more testing:
It's enough to put the Pi to the cmputer and let it mount the Music- and TeslaCam drives. Just to be sure the filesystem wasn't corruptet I logged in to the Pi via ssh and issued a "sudo halt" command. After that both music and TeslaCam was woring in the car, but only first time as before.
I always get both music and TeslaCam or neither of them.
@marcone Strange that the drive was ready but the car didn't use it. Could the reason be that the filesystem is OK the first time after the Pi has been connected to a computer and then the second time in the car the filesystem is corrupted and won't be fixed the same way compared to when it's connected to a computer?

jun3280net, 2021-08-03:
@mane-wt - regarding your "sudo halt" command.  Assuming you can SSH into your Teslausb - when it is not mounting in the car, does running the "sudo halt" command, and rebooting the Teslausb get it to mount in the car?

On a separate note, I tried to do a reinstall of the Teslausb on the Pi Zero W - as I am wondering if the Music drive might be part of the problem.  It seems all of us posting here - have a music drive.  That said - I'm wondering if my issues are deeper, as I am no longer able to reinstall any of the Teslausb images.  I chased down a few threads related to the "get_script failed, retrying" and updating the "wpa_supplicant.conf" info as explained.  No improvement.  

In fact, I do not even see wlan0 when I run "ifconfig".  Is there any way to confirm whether the wireless hardware is actually dead?  Thanks

[image]

boba51, 2021-08-03:
Update:
Car updated to 2021.12.25.7 this morning, rebooted car and connected
Raspberry Pi - No change.

Regarding jun3280net's comment re the music drive, I was also wondering
about that and if Tesla had introduced some checks for more than one
connected device to prevent the use of hubs etc with the dashcam, it would
be good if someone without the Music partition could confirm?

Bob

Jamez3rd, 2021-08-03:
Hello i've been following from the sidelines.  I have a Raspberry Pi Zero.  I'm currently on 2021.4.18.10 with a 2021 Model 3 and i'm having the same issue as everyone else.  I'm only using Pi zero for Dashcam in the glovebox.  I have an external HD for just music and boombox in the center console.  So i can confirm no music partition here.

marcone, 2021-08-03:
I use a music drive, got 2021.12.25.7 yesterday, and everything still works with my Pi0.

boba51, 2021-08-04:
@marcone Hi, anything you can suggest for those of us with the issue to try or any further info we can provide, or do you feel it's a case of waiting for a future Tesla update?

mane-wt, 2021-08-04:
After more testing I am almost convinced that the issue is a result from a corrupt file system. I have tried to connect the Pi to a computer and after the Music drive and the TeslaCam drive have been mounted I just unplugged the Pi in a similar way when the car cuts its power. Like this the Pi won't work in the car. If I instead log in to the Pi with ssh and issues a "sudo halt" command to safely take down the Pi, it works once when connecting it to the car. It is known that the file system on the Pi will be corrupted when the car suddenly cuts power to the Pi. Before when everything worked the Pi fixed the filesystem upon startup but not anymore. Unfortunately it will not work to issue a "sudo reboot" command when the Pi sits in the car.

It would be very interesting if it is possible to issue a command to fix the filesystem via ssh when the Pi sits in the car, @marcone do you know such a command? One other theory is that if the Pi, sitting in the car, eventually fixed the filesystem it wont mount the Music drive and the TeslaCam drive even if it has been fixed. Is it possible to repeat the command to try another time to mount the 2 drives, and if so how to do that?

marcone, 2021-08-04:
> After more testing I am almost convinced that the issue is a result from a corrupt file system.

Which one? Teslausb uses multiple partitions, plus disk images, each with their own filesystem.

mane-wt, 2021-08-05:
Both partitions. I always get both Music and TeslaCAM or neither of them.

marcone, 2021-08-05:
CAM and MUSIC are only two of the 6 or more filesystems that could be in use at any given time, hence my question. In any case, those two are checked for errors on every boot, before being presented to the car, so the problem shouldn't be caused by corruption of those two. The fact that you only ever get neither or both also points at that, since it suggests that the Pi got "stuck" on something before publishing the drives to the car, like in the case where autofs wasn't starting up in an earlier comment above.

mane-wt, 2021-08-05:
For me the problem stared with version 25.6 but I also think Tesla did something unfortunately. Despite if it is due to Tesla or not maybe we can do something about it any way? Would it be possible to check all filesystems upon every boot and/or try to mount more than one time after some delay?

boba51, 2021-08-06:
Yesterday, I checked everything was working fine connected to my PC, ejected the Drives and Halted the Pi, placed it back in the GloveBox, I could see the TeslaUSB online, SSH to it, mount and display the folders and sync Music, also (now I know about it) use the web interface.
This morning the TeslaUSB was not visible on the network, I went into the Tesla App and once connected confirmed Sentry Mode was active, then when I checked again I could see the TeslaUSB on the network and all the above worked again (I didn't try syncing anything).
Just checked in the car, no overnight Sentry Mode events, Dashcam functioning OK, could see/delete events from yesterday/this morning, usual active alert re No USB in GloveBox.
Again anything in the way of logs I can provide, just let me know.

marcone, 2021-08-06:
The Pi not being on the network and then reappearing on the network kind of sounds like the USB port powered off at some point, and then opening the app caused it to power up again. The archiveloop log would tell if it lost power and restarted.

boba51, 2021-08-06:
### Not exactly sure what I'm looking for but for the 5th/6th there were these entries regarding car
Thu	5	Aug	08:20:49	BST	2021:00:00	not keeping car awake.
Thu	5	Aug	10:54:34	BST	2021:00:00	not keeping car awake.
Thu	5	Aug	19:26:57	BST	2021:00:00	not keeping car awake.
Fri	6	Aug	08:29:59	BST	2021:00:00	not keeping car awake.
Fri	6	Aug	13:08:56	BST	2021:00:00	not keeping car awake.

### There was also this weird sequence where the time appeared to go out by just over an hour:
Thu	5	Aug	08:05:19	BST	2021:00:00	low space, deleting /backingfiles/snapshots/snap-000216
Thu	5	Aug	08:05:26	BST	2021:00:00	low space, deleting /backingfiles/snapshots/snap-000217
Thu	5	Aug	**08:05:31**	BST	2021:00:00	low space, deleting /backingfiles/snapshots/snap-000218
Thu	5	Aug	**07:17:02**	BST	2021:00:00	Starting archiveloop at 24.11 seconds uptime...
Thu	5	Aug	07:17:03	BST	2021:00:00	taking snapshot of cam disk in /backingfiles/snapshots/snap-000229
Thu	5	Aug	07:17:09	BST	2021:00:00	waiting for autofs to be active
Thu	5	Aug	07:17:10	BST	2021:00:00	waiting for autofs to be active
Thu	5	Aug	07:17:11	BST	2021:00:00	waiting for autofs to be active
Thu	5	Aug	07:17:13	BST	2021:00:00	waiting for autofs to be active
Thu	5	Aug	07:17:14	BST	2021:00:00	waiting for autofs to be active
Thu	5	Aug	07:17:15	BST	2021:00:00	waiting for autofs to be active
Thu	5	Aug	07:17:16	BST	2021:00:00	waiting for autofs to be active
Thu	5	Aug	07:17:17	BST	2021:00:00	waiting for autofs to be active
Thu	5	Aug	07:17:18	BST	2021:00:00	waiting for autofs to be active
Thu	5	Aug	07:17:20	BST	2021:00:00	took snapshot
Thu	5	Aug	07:17:25	BST	2021:00:00	comparing new snapshot with /backingfiles/snapshots/snap-000228/snap.bin
Thu	5	Aug	07:17:25	BST	2021:00:00	making links for /tmp/snapshots/snap-000229, retargeted to /backingfiles/snapshots/snap-000229/mnt
Thu	5	Aug	07:17:29	BST	2021:00:00	made all links for /tmp/snapshots/snap-000229
Thu	5	Aug	07:17:29	BST	2021:00:00	Running fsck on /backingfiles/cam_disk.bin...
Thu	5	Aug	07:17:30	BST	2021:00:00	| fsck from util-linux 2.33.1
Thu	5	Aug	07:17:33	BST	2021:00:00	| fsck.fat 4.1 (2017-01-24)
Thu	5	Aug	07:17:33	BST	2021:00:00	| 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Thu	5	Aug	07:17:33	BST	2021:00:00	|  Automatically removing dirty bit.
Thu	5	Aug	07:17:33	BST	2021:00:00	| Performing changes.
Thu	5	Aug	07:17:33	BST	2021:00:00	| /dev/loop1p1: 264 files, 221882/2888512 clusters
Thu	5	Aug	07:17:33	BST	2021:00:00	Finished fsck on /backingfiles/cam_disk.bin.
Thu	5	Aug	07:17:33	BST	2021:00:00	Running fsck on /backingfiles/music_disk.bin...
Thu	5	Aug	07:17:33	BST	2021:00:00	| fsck from util-linux 2.33.1
Thu	5	Aug	07:17:35	BST	2021:00:00	| fsck.fat 4.1 (2017-01-24)
Thu	5	Aug	07:17:35	BST	2021:00:00	| /dev/loop1p1: 685 files, 197463/1443854 clusters
Thu	5	Aug	07:17:35	BST	2021:00:00	Finished fsck on /backingfiles/music_disk.bin.
Thu	5	Aug	**07:17:35**	BST	2021:00:00	Trying to set time...
Thu	5	Aug	**08:20:49**	BST	2021:00:00	Time adjusted by 3793.354031 seconds after 0.190000 seconds
Thu	5	Aug	08:20:49	BST	2021:00:00	not keeping car awake.
Thu	5	Aug	08:20:49	BST	2021:00:00	Archiving... 

### There are quite a lot of time adjustments, many of quite large amounts throughout my log:

Sat	10	Jul	15:20:48	BST	2021:00:00	Time adjusted by 0.041322 seconds after 0.460000 seconds
Sat	10	Jul	16:32:05	BST	2021:00:00	Time adjusted by 2.765470 seconds after 2.390000 seconds
Sat	10	Jul	18:32:12	BST	2021:00:00	Time adjusted by 873.085900 seconds after 0.220000 seconds
Sun	11	Jul	18:41:11	BST	2021:00:00	Time adjusted by 1417.996986 seconds after 0.180000 seconds
Mon	12	Jul	18:50:13	BST	2021:00:00	Time adjusted by 1964.287524 seconds after 0.190000 seconds
Mon	12	Jul	20:52:46	BST	2021:00:00	Time adjusted by 0.140406 seconds after 0.190000 seconds
Tue	13	Jul	18:59:22	BST	2021:00:00	Time adjusted by 2510.485778 seconds after 0.190000 seconds
Tue	13	Jul	21:25:52	BST	2021:00:00	Time adjusted by 5.292144 seconds after 5.340000 seconds
Wed	14	Jul	22:13:56	BST	2021:00:00	Time adjusted by 41570.694784 seconds after 0.210000 seconds
Thu	15	Jul	15:19:53	BST	2021:00:00	Time adjusted by 3742.877806 seconds after 0.180000 seconds
Thu	15	Jul	19:05:37	BST	2021:00:00	Time adjusted by 0.178487 seconds after 0.250000 seconds
Fri	16	Jul	00:45:24	BST	2021:00:00	Time adjusted by 1673.035861 seconds after 0.180000 seconds
Fri	16	Jul	01:26:09	BST	2021:00:00	Time adjusted by 31.308971 seconds after 0.270000 seconds
Tue	20	Jul	12:30:14	BST	2021:00:00	Time adjusted by 52761.069820 seconds after 5.340000 seconds
Tue	20	Jul	16:46:43	BST	2021:00:00	Time adjusted by 3988.889017 seconds after 5.230000 seconds
Tue	20	Jul	21:02:45	BST	2021:00:00	Time adjusted by 6315.919796 seconds after 0.180000 seconds
Wed	21	Jul	00:23:48	BST	2021:00:00	Time adjusted by 11175.889354 seconds after 1.250000 seconds
Wed	21	Jul	01:33:07	BST	2021:00:00	Time adjusted by 15338.300939 seconds after 0.150000 seconds
Thu	22	Jul	03:12:13	BST	2021:00:00	Time adjusted by 3282.388175 seconds after 0.180000 seconds
Thu	22	Jul	21:32:26	BST	2021:00:00	Time adjusted by -0.033865 seconds after 0.270000 seconds
Fri	23	Jul	03:21:36	BST	2021:00:00	Time adjusted by 3828.054385 seconds after 0.180000 seconds
Fri	23	Jul	15:44:55	BST	2021:00:00	Time adjusted by 0.037149 seconds after 0.300000 seconds
Fri	23	Jul	17:56:00	BST	2021:00:00	Time adjusted by 0.203233 seconds after 0.250000 seconds
[... cut ...]
Wed	4	Aug	23:28:13	BST	2021:00:00	Time adjusted by 0.713779 seconds after 0.710000 seconds
Thu	5	Aug	08:20:49	BST	2021:00:00	Time adjusted by 3793.354031 seconds after 0.190000 seconds
Thu	5	Aug	10:54:33	BST	2021:00:00	Time adjusted by 0.383021 seconds after 0.420000 seconds
Thu	5	Aug	19:26:57	BST	2021:00:00	Time adjusted by 0.121718 seconds after 0.240000 seconds
Fri	6	Aug	08:29:59	BST	2021:00:00	Time adjusted by 737.210348 seconds after 0.190000 seconds
Fri	6	Aug	13:08:56	BST	2021:00:00	Time adjusted by 0.164750 seconds after 0.260000 seconds

marcone, 2021-08-06:
Look for something like:
```
==============================================
Mon  2 Aug 18:44:32 PDT 2021: Starting archiveloop at 23.11 seconds uptime...
```
That indicates the Pi just booted up.

The time adjustments happen every time the Pi does an archive operation. In your log-excerpt the larger values likely indicate the Pi was powered off for a while. The small values are just corrections for clock-drift.

boba51, 2021-08-06:
Quite a few events, but I can't be 100% sure which ones may be where I've been connecting/disconnecting whilst testing on the PC etc., but definitely on the 5th/6th I didn't, so we are thinking potentially a loss of power, I have Sentry Mode enabled and the logs show every hour so doesn't look like a prolonged loss of power.

Hopefully, these extracts help provide some useful information for everyone trying to piece this together and happy to try anything and provide more information, just let me know.

I would add since the Pi has been back in the car, it has been available every time I got in the car and stored a few Sentry Mode events so seems to be behaving itself ATM.

Mon	26	Jul	07:17:03	BST	2021:00:00	Starting archiveloop at 23.36 seconds uptime...
Mon	26	Jul	21:17:03	BST	2021:00:00	Starting archiveloop at 23.09 seconds uptime...
Tue	27	Jul	07:17:02	BST	2021:00:00	Starting archiveloop at 23.24 seconds uptime...
Tue	27	Jul	22:17:04	BST	2021:00:00	Starting archiveloop at 24.73 seconds uptime...
Tue	27	Jul	23:17:02	BST	2021:00:00	Starting archiveloop at 23.20 seconds uptime...
Tue	27	Jul	23:17:02	BST	2021:00:00	Starting archiveloop at 22.87 seconds uptime...
Wed	28	Jul	20:17:03	BST	2021:00:00	Starting archiveloop at 22.91 seconds uptime...
Thu	29	Jul	09:17:03	BST	2021:00:00	Starting archiveloop at 22.96 seconds uptime...
Thu	29	Jul	09:17:03	BST	2021:00:00	Starting archiveloop at 23.21 seconds uptime...
Sat	31	Jul	23:17:03	BST	2021:00:00	Starting archiveloop at 23.24 seconds uptime...
Sat	31	Jul	23:17:02	BST	2021:00:00	Starting archiveloop at 23.37 seconds uptime...
Sat	31	Jul	23:17:03	BST	2021:00:00	Starting archiveloop at 23.14 seconds uptime...
Tue	3	Aug	12:17:03	BST	2021:00:00	Starting archiveloop at 23.24 seconds uptime...
Tue	3	Aug	16:17:03	BST	2021:00:00	Starting archiveloop at 22.92 seconds uptime...
Tue	3	Aug	16:17:03	BST	2021:00:00	Starting archiveloop at 23.37 seconds uptime...
Wed	4	Aug	20:32:49	BST	2021:00:00	Starting archiveloop at 23.01 seconds uptime...
Wed	4	Aug	21:00:47	BST	2021:00:00	Starting archiveloop at 23.31 seconds uptime...
Wed	4	Aug	21:17:03	BST	2021:00:00	Starting archiveloop at 23.02 seconds uptime...
Wed	4	Aug	23:09:32	BST	2021:00:00	Starting archiveloop at 23.07 seconds uptime...
**Thu	5	Aug	07:17:02	BST	2021:00:00	Starting archiveloop at 24.11 seconds uptime...
Fri	6	Aug	08:17:03	BST	2021:00:00	Starting archiveloop at 23.70 seconds uptime...**

boba51, 2021-08-06:
@marcone I was just thinking about what you said about the large adjustments due to being off for a while but did you notice that the time went back nearly an hour and then adjusted to 15 mins or so of the original time it was logging events at, which indicates to me it wasn't off that long, but I may not be fully understanding?

Thu 5 Aug 08:05:31** BST 2021:00:00 low space, deleting /backingfiles/snapshots/snap-000218
Thu 5 Aug 07:17:02 BST 2021:00:00 Starting archiveloop at 24.11 seconds uptime...

Thu 5 Aug 07:17:35 BST 2021:00:00 Trying to set time...
Thu 5 Aug 08:20:49 BST 2021:00:00 Time adjusted by 3793.354031 seconds after 0.190000 seconds

marcone, 2021-08-06:
It's really hard to draw any conclusions from what you've posted so far, since it's just been snippets of the log and not the entire log.
Normally every `Starting archiveloop` line indicates the Pi was powered on after having turned off. I would expect the first "Time adjusted" after power-on to be large, and any subsequent ones to be small.

boba51, 2021-08-06:
@marcone Sorry, I was trying to focus on things I thought looked odd, such as this time anomaly to try and make it easier, but happy to supply the full log, shall I just attach it here?

marcone, 2021-08-06:
sure, just attach it here (diagnostics log would be even better)

boba51, 2021-08-07:
@marcone 
I've attached both, looking at them briefly it looks like it rebooted again this morning.

[archiveloop (1).log](https://github.com/marcone/teslausb/files/6948888/archiveloop.1.log)
[diagnostics.txt](https://github.com/marcone/teslausb/files/6948889/diagnostics.txt)

marcone, 2021-08-07:
Yes, the diagnostics show that the Pi restarted some time this morning, probably due to a very brief power loss.
When the Pi started back up the car didn't mount the drives (though the Pi made them available). So while Sentry was probably still on, the car wouldn't actually have recorded anything.
I would try plugging the Pi in to a different port in the car and use a different cable, to see if that makes a difference. You might also try just plugging the Pi into a power adapter for a few days and see if it has a habit of spontaneously rebooting.

boba51, 2021-08-07:
> Yes, the diagnostics show that the Pi restarted some time this morning, probably due to a very brief power loss.
> When the Pi started back up the car didn't mount the drives (though the Pi made them [...]

@marcone
Thanks I'll give that a go and feedback on the results.

In case it provides any additional info this is what happened today and a copy of my latest Diagnostic log:

From here onwards away from home no network

You were right the USB/dashcam, wasn't available in the car, unfortunately, I didn't realise until halfway through my journey when I tried to save some footage of the M25 gridlock.

14:08 - I arrived at my destination, with the glove box open to monitor the Pi confirmed that it was merrily flashing away, I rebooted the cars using the 2 scroll wheel method, I saw the Pi powered off, it then came back up and went through its boot sequence settling down to single flash, but the dashcam was still unavailable.

14:35 - Unplugged & plugged the Pi back in, it went through its boot sequence settling down to single flash, but the dashcam was still unavailable.

18:58 - Returned to car, Pi still running with the normal single flash, but still no dashcam - Unplugged Pi

19:12 - Plugged the Pi back in, it went through its boot sequence settling down to single flash, but the dashcam was still unavailable.

From here onwards home connected to network

22:35 - Just went to car to take the Pi out for testing and the Dashcam is active again, I've done nothing except access the Pi via SSH and the WebGUI, left Pi connected and took attached diagnostics.

[diagnostics sat evening 7-8-21.txt](https://github.com/marcone/teslausb/files/6949847/diagnostics.sat.evening.7-8-21.txt)

marcone, 2021-08-07:
Your diagnostics agree with your timeline. 

In the archiveloop section we see the Pi booting up with its clock set to 17:17, but later when it connects to the network to get the real time it adjusts the clock by 6931 seconds, so adjusting 17:17 by that we see that it really booted up at 19:12.
It then waits for the archive server to be reachable, but apparently that doesn't happen until 34 minutes later:
```
Sat  7 Aug 17:17:54 BST 2021: Connected usb to host.
Sat  7 Aug 17:17:54 BST 2021: Waiting for archive to be reachable...
Sat  7 Aug 17:51:59 BST 2021: Archive is reachable.
Sat  7 Aug 17:52:00 BST 2021: Trying to set time...
Sat  7 Aug 19:47:31 BST 2021: Time adjusted by 6931.856671 seconds after 0.160000 seconds
```
It then performs an archive cycle, which temporarily removes the drives from the car, and when that is done at 19:48 it again publishes the drives to the car.

Similarly the kernel log shows the Pi booted up at 19:12, shows it making the drives available shortly after, and then again at 19:48, however the car only actually accessed the drives at 19:48. The car ignored the drives when they were first published after the Pi booted up.

roger-richards, 2021-08-07:
FWIW, my M3 2021 has the same symptoms. TeslaUSB configured for both CAM and MUSIC partitions.  TeslaUSB was very solid until a couple weeks ago.

boba51, 2021-08-07:
> Your diagnostics agree with your timeline.
>
> In the archiveloop section we see the Pi booting up with its clock set to 17:17, but later when it connects to the network to get the real time it adjusts [...]

@marcone Thanks, does this help shed any light on whether the issue is Pi or Tesla related or it is still unknown?
It's a shame that's become unreliable as it's a great tool to have and I (and sure all other users) appreciate your efforts in making it available and helping to try and troubleshoot it, I only wish I was able to help more but I'm a little out of my depth with most of the programmatic things discussed, so thanks again.

boba51, 2021-08-07:
> FWIW, my M3 2021 has the same symptoms. TeslaUSB configured for both CAM and MUSIC partitions. TeslaUSB was very solid until a couple weeks ago.

Yes, definitely appears related to a recent Tesla update.

marcone, 2021-08-07:
> does this help shed any light on whether the issue is Pi or Tesla related or it is still unknown?

Teslausb made the drives available to the car but the car chose not to use them. That seems like a car issue to me.

BTW though, you said earlier:

> 19:12 - Plugged the Pi back in, it went through its boot sequence settling down to single flash, but the dashcam was still unavailable.
>
> From here onwards home connected to network

but note that the Pi couldn't actually reach the archive server until 34 minutes after bootup, so you may have some network issues somewhere.

boba51, 2021-08-08:
> BTW though, you said earlier:
>
>
> but note that the Pi couldn't actually reach the archive server until 34 minutes after bootup, so you may have some network issues somewhere.

No, the Pi was connected at 19:12 whilst away from home, it then connected once I got home, I see lots of entries for failed NTP when it boots off the network, does this have a negative effect on the boot-up, not particularly related to the car accepting the USB but generally?

Again forgive me if a silly idea or difficult to implement, but could the Pi check for network connection on boot to decide whether to try and connect to the Archive/NTP Servers?

boba51, 2021-08-08:
An observation:
58 of the 62 entries in my Archiveloop log start @ 17 mins past the hour?
The original log also attached
[archiveloop Sunday 8-8-21.log](https://github.com/marcone/teslausb/files/6950551/archiveloop.Sunday.8-8-21.log)

NB Multiple entries for some?  **_bolded & italicized_**

Day	Date	Month	Time	Year	Event
Mon	12	Jul	18:17:03	2021:00:00	Starting archiveloop at 22.95 seconds uptime...
Tue	13	Jul	18:17:02	2021:00:00	Starting archiveloop at 22.92 seconds uptime...
Wed	14	Jul	10:17:02	2021:00:00	Starting archiveloop at 22.98 seconds uptime...
Thu	15	Jul	14:17:02	2021:00:00	Starting archiveloop at 22.57 seconds uptime...
Fri	16	Jul	00:17:03	2021:00:00	Starting archiveloop at 22.92 seconds uptime...
Fri	16	Jul	10:17:02	2021:00:00	Starting archiveloop at 23.29 seconds uptime...
Fri	16	Jul	19:17:03	2021:00:00	Starting archiveloop at 24.05 seconds uptime...

**_Fri	16	Jul	20:17:03	2021:00:00	Starting archiveloop at 23.26 seconds uptime...
Fri	16	Jul	20:17:03	2021:00:00	Starting archiveloop at 23.22 seconds uptime...
Fri	16	Jul	20:17:03	2021:00:00	Starting archiveloop at 23.21 seconds uptime...
Fri	16	Jul	20:17:02	2021:00:00	Starting archiveloop at 23.41 seconds uptime..._**

**_Sat	17	Jul	06:17:03	2021:00:00	Starting archiveloop at 22.98 seconds uptime...
Sat	17	Jul	06:17:02	2021:00:00	Starting archiveloop at 23.18 seconds uptime..._**

**_Sat	17	Jul	11:17:06	2021:00:00	Starting archiveloop at 26.36 seconds uptime...
Sat	17	Jul	11:17:03	2021:00:00	Starting archiveloop at 23.02 seconds uptime...
Sat	17	Jul	11:17:03	2021:00:00	Starting archiveloop at 23.05 seconds uptime..._**

**_Sat	17	Jul	12:17:03	2021:00:00	Starting archiveloop at 23.47 seconds uptime...
Sat	17	Jul	12:17:04	2021:00:00	Starting archiveloop at 23.34 seconds uptime...
Sat	17	Jul	12:17:04	2021:00:00	Starting archiveloop at 22.99 seconds uptime...
Sat	17	Jul	12:17:04	2021:00:00	Starting archiveloop at 23.20 seconds uptime...
Sat	17	Jul	12:17:07	2021:00:00	Starting archiveloop at 25.87 seconds uptime...
Sat	17	Jul	12:17:04	2021:00:00	Starting archiveloop at 23.07 seconds uptime...
Sat	17	Jul	12:17:04	2021:00:00	Starting archiveloop at 23.01 seconds uptime...
Sat	17	Jul	12:17:03	2021:00:00	Starting archiveloop at 23.47 seconds uptime..._**

**_Sat	17	Jul	17:17:08	2021:00:00	Starting archiveloop at 27.84 seconds uptime...
Sat	17	Jul	17:17:08	2021:00:00	Starting archiveloop at 27.70 seconds uptime..._**

Sun	18	Jul	19:17:03	2021:00:00	Starting archiveloop at 23.15 seconds uptime...
Tue	20	Jul	13:17:03	2021:00:00	Starting archiveloop at 23.04 seconds uptime...
Tue	20	Jul	19:17:03	2021:00:00	Starting archiveloop at 23.46 seconds uptime...

**_Tue	20	Jul	21:17:03	2021:00:00	Starting archiveloop at 23.13 seconds uptime...
Tue	20	Jul	21:17:02	2021:00:00	Starting archiveloop at 24.39 seconds uptime..._**

Thu	22	Jul	02:17:03	2021:00:00	Starting archiveloop at 23.15 seconds uptime...
Fri	23	Jul	02:17:03	2021:00:00	Starting archiveloop at 23.64 seconds uptime...
Sat	24	Jul	05:17:03	2021:00:00	Starting archiveloop at 23.21 seconds uptime...
Sun	25	Jul	05:17:03	2021:00:00	Starting archiveloop at 24.87 seconds uptime...
Mon	26	Jul	07:17:03	2021:00:00	Starting archiveloop at 23.36 seconds uptime...
Mon	26	Jul	21:17:03	2021:00:00	Starting archiveloop at 23.09 seconds uptime...
Tue	27	Jul	07:17:02	2021:00:00	Starting archiveloop at 23.24 seconds uptime...
Tue	27	Jul	22:17:04	2021:00:00	Starting archiveloop at 24.73 seconds uptime...

**_Tue	27	Jul	23:17:02	2021:00:00	Starting archiveloop at 23.20 seconds uptime...
Tue	27	Jul	23:17:02	2021:00:00	Starting archiveloop at 22.87 seconds uptime..._**

Wed	28	Jul	20:17:03	2021:00:00	Starting archiveloop at 22.91 seconds uptime...

**_Thu	29	Jul	09:17:03	2021:00:00	Starting archiveloop at 22.96 seconds uptime...
Thu	29	Jul	09:17:03	2021:00:00	Starting archiveloop at 23.21 seconds uptime..._**

**_Sat	31	Jul	23:17:03	2021:00:00	Starting archiveloop at 23.24 seconds uptime...
Sat	31	Jul	23:17:02	2021:00:00	Starting archiveloop at 23.37 seconds uptime...
Sat	31	Jul	23:17:03	2021:00:00	Starting archiveloop at 23.14 seconds uptime..._**

Tue	3	Aug	12:17:03	2021:00:00	Starting archiveloop at 23.24 seconds uptime...

**_Tue	3	Aug	16:17:03	2021:00:00	Starting archiveloop at 22.92 seconds uptime...
Tue	3	Aug	16:17:03	2021:00:00	Starting archiveloop at 23.37 seconds uptime..._**

Wed	4	Aug	21:17:03	2021:00:00	Starting archiveloop at 23.02 seconds uptime...
Thu	5	Aug	07:17:02	2021:00:00	Starting archiveloop at 24.11 seconds uptime...
Fri	6	Aug	08:17:03	2021:00:00	Starting archiveloop at 23.70 seconds uptime...
Sat	7	Aug	08:17:03	2021:00:00	Starting archiveloop at 23.55 seconds uptime...

**_Sat	7	Aug	13:17:03	2021:00:00	Starting archiveloop at 23.55 seconds uptime...
Sat	7	Aug	13:17:03	2021:00:00	Starting archiveloop at 22.92 seconds uptime..._**

Sat	7	Aug	17:17:03	2021:00:00	Starting archiveloop at 23.06 seconds uptime...
Sun	8	Aug	08:17:04	2021:00:00	Starting archiveloop at 23.66 seconds uptime...

marcone, 2021-08-08:
> does this have a negative effect on the boot-up

No, it doesn't matter.

The clock time at bootup is completely irrelevant, since the Pi doesn't have a clock. Duplicate entries simply means that it loaded the same cached value on those occasions. The fact that it often starts at 17 minutes after the hour is interesting, but doesn't matter either. The clock is only needed when archiving, and will be set from the network at that point.

jun3280net, 2021-08-08:
@boba51 has presented a great deal of interesting information.  I've always wondered whether teslausb is stuck in archiveloop state preventing the device from mounting in the car.  For instance, if the device was to eventually re-mount, it would be at home after it had reconnected to the network.

On that note though, because my pi-zero w is dead, I've been using a USB stick in the car without issue for weeks now.  The USB stick does not mount and unmount from the car.  However, the USB stick has no issues when the power is cut to the drive.  

A few dumb questions : 
-is there a way execute the archiveloop process without mount/unmount (I can only assume the mounting of the FAT32 drive, prevents Raspbian from creating the archive data in the remaining space of the SD card).  I assume as much, because the archive video can only be accessed via SMB.  
-Related to the above, can the archiveloop process be executed on the TeslaCam FAT32 drive?  Set TeslaCam drive to max size (eg if card is 256GB, set TeslaCam to 250GB - but you would set Archive Folder to 150GB, allowing Tesla to overwrite 100GB for Recent, Sentry, Saved folders).  Within the TeslaCam drive, create a folder Archive, that archiveloop  could write to?    
-If Raspbian Desktop was installed - could you connect Teslausb to a monitor via HDMI, and view the archive video from within Raspbian directly?
-For the purposes of troubleshooting, is there a way to disable the archiveloop process (I know that defeats the purpose of teslausb).  I'm just curious as to whether the problems are caused by the archiveloop mount/unmount process or some sort of OS file corruption caused by the power being cut to Teslausb.  By forcing it to never archive, it might give us better insight where else to investigate.

Thanks for everyones insight into this issue

trinitron79, 2021-08-09:
just ran across this thread today, I'm hitting the same problem with my X (12.25.7).  I thought it was my pi Zero dying but it looks like it might not be the case.  If there's any I can gather please let me know.

Using a 128gb SD card with no music part.

boba51, 2021-08-09:
> I would try plugging the Pi in to a different port in the car and use a different cable, to see if that makes a difference. You might also try just plugging the Pi into a power adapter for a few days and see if it has a habit of spontaneously rebooting.

So the Pi has been up solid for over 24 hours, whilst connected to a power supply, connected to the network no drives mounted:

- teslausb has been up 1 day, 50 minutes. System time is Mon  9 Aug 21:52:24 BST 2021

Tomorrow I'm going to connect to my PC and see if this has any effect, then after another 24 hours try with no drives and no network, I am going to simulate losing the network by blocking it's MAC Address on my Router (unless anyone had any other suggestions?) as I don't want to alter any other settings on the Pi itself, although in this state I would be able to connect to monitor so will just leave 24 hours and reconnect to the network.

brettjenkins, 2021-08-10:
Hitting the same issue in my 2021 model 3 after updating the car, worked okay before (albeit with warnings), now not mounting at all.

boba51, 2021-08-10:
> Hitting the same issue in my 2021 model 3 after updating the car, worked okay before (albeit with warnings), now not mounting at all.

Be interesting to see what Tesla has done once it all comes out, most frustrating!

boba51, 2021-08-10:
> So the Pi has been up solid for over 24 hours, whilst connected to a power supply, connected to the network no drives mounted:
>
> * teslausb has been up 1 day, 50 minutes. System time is Mon  9 Aug [...]

so after a solid : teslausb has been up 1 day, 16 hours, 49 minutes. System time is Tue 10 Aug 13:51:40 BST 2021

I'm going to connect to my PC, so mounted drives and network.

boba51, 2021-08-10:
> so after a solid : teslausb has been up 1 day, 16 hours, 49 minutes. System time is Tue 10 Aug 13:51:40 BST 2021
>
> I'm going to connect to my PC, so mounted drives and network.

So far so good, connected to PC with the same cable used in Tesla, Drives available and don't believe they've been offline as the same  Explorer windows open for both drives, certainly pointing towards a Tesla issue.

"teslausb has been up 7 hours, 19 minutes. System time is Tue 10 Aug 21:59:18 BST 2021"
USB state (26500.8 seconds ago): connected to host and host mounted drive(s)

hansendc, 2021-08-10:
I've been having the same problems as some of the other posters.  Mine was working quite well since September of last year and then suddenly stopped working in July.  I went on a bit of a systemd ramage, looking at "systemd-analyze critical-chain", hoping that the problems were due to boot times.  I disabled nginx and set the dhcpcd timeout to 3 seconds.  I also removed all the snapshotting from archiveloop.  I put the pi back in the Tesla and it started working again, for an hour or two, then it went back to being broken again.

During all this time, the pi continued to mount without issues in my PC.

I have a few "Long filename fragment "2021-08-09_10" found outside a LFN sequence." errors from fsck.vfat, but otherwise no error messages on the pi.

I really suspect this is something that changed on the car side.

brettjenkins, 2021-08-11:
I wonder if someone could setup a pi that does nothing but emulate a USB drive (without all the archive loop stuff), and see if that works, just to rule it out

boba51, 2021-08-11:
> So far so good, connected to PC with the same cable used in Tesla, Drives available and don't believe they've been offline as the same Explorer windows open for both drives, certainly pointing [...]

So now been up for "teslausb has been up 1 day, 3 hours, 23 minutes. System time is Wed 11 Aug 18:03:09 BST 2021" and explorer windows still open on PC so believe been available 100% of the time.

I'm going to block the device from my router to see if the lack of network connection has an effect, once I've done it attached to the PC will try the Pi powered standalone with no network, but from the looks of the other conversations here and on discord it looks like everyone coming to the conclusion that the Tesla is the culprit.

AXEBro, 2021-08-12:
strangely I got this after about a week or two of using it. 

Yesterday, I got a brand new microSD card, and installed fresh (though I used the same config settings) - and the car still doesn't recognize the device. I haven't received any Tesla updates in a while, so totally confused here.

boba51, 2021-08-12:
> > > > > > I would try plugging the Pi in to a different port in the car and use a different cable, to see if that makes a difference. You might also try just plugging the Pi into a power adapter for a few days and see if it has a habit of spontaneously rebooting.
> > > > > 
> > > > > 
> > > > > So the Pi has been up solid for over 24 hours, whilst connected to a power supply, connected to the network no drives mounted:
> > > > > 
> > > > > * teslausb has been up 1 day, 50 minutes. System time is Mon  9 Aug 21:52:24 BST 2021
> > > > > 
> > > > > Tomorrow I'm going to connect to my PC and see if this has any effect, then after another 24 hours try with no drives and no network, I am going to simulate losing the network by blocking it's MAC Address on my Router (unless anyone had any other suggestions?) as I don't want to alter any other settings on the Pi itself, although in this state I would be able to connect to monitor so will just leave 24 hours and reconnect to the network.
> > > > 
> > > > 
> > > > so after a solid : teslausb has been up 1 day, 16 hours, 49 minutes. System time is Tue 10 Aug 13:51:40 BST 2021
> > > > I'm going to connect to my PC, so mounted drives and network.
> > > 
> > > 
> > > So far so good, connected to PC with the same cable used in Tesla, Drives available and don't believe they've been offline as the same Explorer windows open for both drives, certainly pointing towards a Tesla issue.
> > > "teslausb has been up 7 hours, 19 minutes. System time is Tue 10 Aug 21:59:18 BST 2021"
> > > USB state (26500.8 seconds ago): connected to host and host mounted drive(s)
> > 
> > 
> > So now been up for "teslausb has been up 1 day, 3 hours, 23 minutes. System time is Wed 11 Aug 18:03:09 BST 2021" and explorer windows still open on PC so believe been available 100% of the time.
> > I'm going to block the device from my router to see if the lack of network connection has an effect, once I've done it attached to the PC will try the Pi powered standalone with no network, but from the looks of the other conversations here and on discord it looks like everyone coming to the conclusion that the Tesla is the culprit.
> > Anyway to access the Pi's console whilst connected to the PC via USB or is it just a case of checking the logs after?
> 
> As per 00:10 blocked Pi MAC Address on my Router, single flash of the Pi LED, whilst I was messing with my router and it caused wireless clients to disconnect briefly, I lost the drives on my PC twice but they came back by themselves so probably discounts my theory that they hadn't disconnected as I would know if I hadn't seen them this time.
> 
> Anyway leaving on PC with no network overnight, obviously cannot SSH or access web interface now so will have to wait until back on the network to look at the log events.

Not being connected did seem to have much effect on uptime:
teslausb has been up 2 days, 3 hours, 6 minutes. System time is Thu 12 Aug 17:46:10 BST 2021

boba51, 2021-08-13:
> > > > > > > I would try plugging the Pi in to a different port in the car and use a different cable, to see if that makes a difference. You might also try just plugging the Pi into a power adapter for a few days and see if it has a habit of spontaneously rebooting.
> > > > > > 
> > > > > > 
> > > > > > So the Pi has been up solid for over 24 hours, whilst connected to a power supply, connected to the network no drives mounted:
> > > > > > 
> > > > > > * teslausb has been up 1 day, 50 minutes. System time is Mon  9 Aug 21:52:24 BST 2021
> > > > > > 
> > > > > > Tomorrow I'm going to connect to my PC and see if this has any effect, then after another 24 hours try with no drives and no network, I am going to simulate losing the network by blocking it's MAC Address on my Router (unless anyone had any other suggestions?) as I don't want to alter any other settings on the Pi itself, although in this state I would be able to connect to monitor so will just leave 24 hours and reconnect to the network.
> > > > > 
> > > > > 
> > > > > so after a solid : teslausb has been up 1 day, 16 hours, 49 minutes. System time is Tue 10 Aug 13:51:40 BST 2021
> > > > > I'm going to connect to my PC, so mounted drives and network.
> > > > 
> > > > 
> > > > So far so good, connected to PC with the same cable used in Tesla, Drives available and don't believe they've been offline as the same Explorer windows open for both drives, certainly pointing towards a Tesla issue.
> > > > "teslausb has been up 7 hours, 19 minutes. System time is Tue 10 Aug 21:59:18 BST 2021"
> > > > USB state (26500.8 seconds ago): connected to host and host mounted drive(s)
> > > 
> > > 
> > > So now been up for "teslausb has been up 1 day, 3 hours, 23 minutes. System time is Wed 11 Aug 18:03:09 BST 2021" and explorer windows still open on PC so believe been available 100% of the time.
> > > I'm going to block the device from my router to see if the lack of network connection has an effect, once I've done it attached to the PC will try the Pi powered standalone with no network, but from the looks of the other conversations here and on discord it looks like everyone coming to the conclusion that the Tesla is the culprit.
> > > Anyway to access the Pi's console whilst connected to the PC via USB or is it just a case of checking the logs after?
> > 
> > 
> > As per 00:10 blocked Pi MAC Address on my Router, single flash of the Pi LED, whilst I was messing with my router and it caused wireless clients to disconnect briefly, I lost the drives on my PC twice but they came back by themselves so probably discounts my theory that they hadn't disconnected as I would know if I hadn't seen them this time.
> > Anyway leaving on PC with no network overnight, obviously cannot SSH or access web interface now so will have to wait until back on the network to look at the log events.
> 
> Not being connected to network didn't seem to have much effect on uptime:
> 
> teslausb has been up 2 days, 3 hours, 6 minutes. System time is Thu 12 Aug 17:46:10 BST 2021
> 
> [diagnostics 12-8-21.txt](https://github.com/marcone/teslausb/files/6977139/diagnostics.12-8-21.txt)
> [archiveloop 12-8-21.log](https://github.com/marcone/teslausb/files/6977141/archiveloop.12-8-21.log)

Pi had been up without issue for over 24hrs attached to a battery power supply, so no Drives but networked, the last test will be attached to the battery but I'm power cycling whilst blocked on the network, so simulating a reset when away from home, not expecting much difference, but just for completeness.

boba51, 2021-08-14:
> Pi had been up without issue for over 24hrs attached to a battery power supply, so no Drives but networked, the last test will be attached to the battery but I'm power cycling whilst blocked on the [...]

So finished final test, again the Pi seems to have been stable throughout the time, booted with no network:
teslausb has been up 1 day, 3 hours, 48 minutes. System time is Sat 14 Aug 22:22:15 BST 2021

I hope the testing has been useful, although I feel it's only demonstrated what we believe something has changed on the Cars since a recent update, causing instability with the TeslaUSB :(

[archiveloop - 14-08-21.log](https://github.com/marcone/teslausb/files/6987209/archiveloop.-.14-08-21.log)
[diagnostics 14-08-21.txt](https://github.com/marcone/teslausb/files/6987211/diagnostics.14-08-21.txt)

jun3280net, 2021-08-14:
@boba51 thanks for the thorough testing.  Did you ever test in the car with no archiveloop - so basically it acts just like a normal microsd/usb drive regardless of whether it connects/disconnects to the network?

boba51, 2021-08-14:
> @boba51 thanks for the thorough testing. Did you ever test in the car with no archiveloop - so basically it acts just like a normal microsd/usb drive regardless of whether it connects/disconnects to the network?

@jun3280net no, sorry I would need a little guidance to config this as although IT literate, I don't have a lot of Linux experience.

jun3280net, 2021-08-14:
> @jun3280net no, sorry I would need a little guidance to config this as although IT literate, I don't have a lot of Linux experience.

It's literally just with the original setup in the config file you update when you setup a new installation.  You add this line somewhere in the file " use export ARCHIVE_SYSTEM=none" .  

That said - I think I might be able to test this since I can probably set this up while connected to internet via ethernet - now that I figured that out.

For context my PI Zero W - the Wifi on the device died - but the device still works.  I was able to confirm and update Raspbian by  installing using the internet via USB from another computer.  Then I found a USB ethernet connector - and confirmed that works on the device.  I'll try re-installing with the USB ethernet connector attached.  It would essentially never connect to network once I plug it into the car.

rmyadsk, 2021-08-14:
Actually a new installation isn't necessary - an upgrade should suffice. First run:

`/root/bin/remountfs_rw`

Edit /root/teslausb_setup_variables.conf to reflect ARCHIVE_SYSTEM=none. Then run:

`/root/bin/setup-teslausb upgrade`

Upon reboot (triggered by the upgrade process), archiving should be disabled.

jun3280net, 2021-08-14:
> Actually a new installation isn't necessary - an upgrade should suffice. First run:
>
> `/root/bin/remountfs_rw`
>
> Edit /root/teslausb_setup_variables.conf to reflect ARCHIVE_SYSTEM=none. Then run: [...]

@rmyadsk yes that is the way to update.  However, I would caveat that once Teslausb stops mounting in the car, updates don't ever seem to fix the problem.  

I'll try a fresh install and disable archive and see how long that lasts.

rmyadsk, 2021-08-14:
Yes, but isn't the intent of disabling archiving to see if the archive process might somehow be interfering with the mount process?

boba51, 2021-08-14:
> Actually a new installation isn't necessary - an upgrade should suffice. First run:
>
> `/root/bin/remountfs_rw`
>
> Edit /root/teslausb_setup_variables.conf to reflect ARCHIVE_SYSTEM=none. Then run: [...]

Done, I'll put it back in the car later and feedback on the results.

Do I just remove "ARCHIVE_SYSTEM=none" and run update again to remove it or something else?

[195 more replies]