marcone/teslausb #655: SSH to music folder to delete files

Poncherelly, 2021-07-22:
Since the music folder doesn't sync, is it possible to ssh to the folder to delete files then new files could copy over the next time it connects?  I'd just prefer not having to pull the device out to make changes with my desktop.

mrd0n, 2021-07-23:
Yes, you can ssh into a terminal session.  Try connecting via ‘ssh pi@teslausb.local’ or ‘ssh pi@<ip addr>‘ ensuring that the device is powered on. 

I’ve changed the default pi account password long ago and don’t recall the default, but I’m sure a quick look through the docs will reveal if you don’t know it.

Poncherelly, 2021-07-23:
> Reply-to: ***@***.***
>
> Yes, you can ssh into a terminal session. Try connecting via ‘ssh ***@***.******@***.***>’ or ‘ssh pi@‘ ensuring that the device is powered on.

What would the path be?

Richard

Purchasing a TESLA?  Please use My Referral Code<https://ts.la/richard48613> for free supercharging

marcone, 2021-07-23:
(please don't advertise your referral code here)

Not sure why said "Since the music folder doesn't sync", but you can always ssh in to the Pi and mount it manually using `mount /mnt/music`

closed 2021-07-23 by marcone

Poncherelly, 2021-07-23:
Sorry, it was attached as my signature.

The music syncs one way but if I delete a file on the server, it doesn't update the pi. I figured I could ssh to the pi to manually delete files I no longer want.

marcone, 2021-07-23:
> The music syncs one way but if I delete a file on the server, it doesn't update the pi.

Music sync has deleted files on the Pi that no longer exist on the server since 2019. Have you not updated teslausb since then?

Poncherelly, 2021-07-23:
I'll try updating again but I'm certain I've updated the pi at least once this year.

If deleting a file in the music folder on the server will delete the file on the pi at next sync, then I'm a happy camper.

Thank you.