marcone/teslausb #657: Enable USB storage only

mluggy, 2021-07-31:
What would be the minimal requirements to get Tesla to only recognize my Pi's SD card as USB storage?

marcone, 2021-07-31:
You mean you don't want the archiving functionality? Use `export ARCHIVE_SYSTEM=none` in the config.

closed 2021-07-31 by marcone

CodeChief, 2021-08-01:
> What would be the minimal steps to get Tesla to only recognize my Pi's SD card as USB storage?

The core is of course the original basic configuration of the OS to enable the USB storage emulation, as acknowledged at the bottom of this repository start document, implemented in this particular design by cimryan then extended by marcone:

[image]

Given basic emulation ([link from above](https://www.reddit.com/r/teslamotors/comments/9m9gyk/build_a_smart_usb_drive_for_your_tesla_dash_cam/)) you could in theory do whatever you want on any distribution of Linux. But what you have here already works if you avoid a couple of documented pitfalls and known "limitations" ('21 refresh USB warning).

But what you will lose and have to re-implement, even if you don't want this design for getting files off the device, is at least:
1. No file history from "recent" clips.
2. Have to install some kind of your own server to get access to the files.
3. Have to write your own web interface or find some kind of generic file browsing interface.
4. If you wanted to sync with some of the more smart methods, find an alternative to the cloud solutions.
5. Find a solution for accessing the files from Tesla AND the OS at the same time (limitation of current driver).

The main problem is #5 in my opinion. The principle use case is that it's a smart/network capable USB stick. So until the car and device can access the data at the same time (read and write) it *has to be* way more complicated than necessary. Having written a SCSI device driver in the past, I don't see the *requirement* as a show stopper (it can be done). The difficulty, I guess, is somebody has to invest a significant amount time and effort to write a new Linux USB emulation driver. I'm interested but don't have time at the moment. Maybe later if Tesla still don't fix the USB warnings.

Until then, I wonder if there is an existing configuration of Linux which can create a volume with USB SCSI disk emulation. If that were acceptable to the car and the OS maybe no new driver is necessary. Or is there a Linux disk virtualization system which could act as a shim to both sides, the key being arbitration between the USB and OS IO requests/locks. In enterprise servers/clusters this is known as a DLM (Distributed Lock Manager).

If we can improve the USB<->disk emulation, everything else is easy.

mluggy, 2021-08-01:
I want to write a new cloud syncer that doesn't change/move files. what are the minimal steps I need from this basic emulation to get Tesla to start writing files?

CodeChief, 2021-08-01:
> I want to write a new cloud syncer that doesn't change/move files. what are the minimal steps I need from this basic emulation to get Tesla to start writing files?

Follow the link above, then the instructions.

marcone, 2021-08-01:
> I want to write a new cloud syncer that doesn't change/move files. what are the minimal steps I need from this basic emulation to get Tesla to start writing files?

It's not entirely clear to me what exactly you want to do, but keeping in mind your "minimal steps" requirement:
- If you just want to change the "move" behavior (teslausb doesn't change the files), then modify archiveloop to do so.
- If you want to add support for a new cloud storage service (i.e. something other than the dozens of cloud storage systems already supported by rclone), then add the appropriate scripts to [teslausb/run/\<name>_archive/](https://github.com/marcone/teslausb/tree/main-dev/run) and it should pretty much work if you set ARCHIVE_SYSTEM to the right value. Alternatively you could contribute the new backend support to rclone, but that's probably more work.

If you want to create something completely independent of teslausb, then Google "g_mass_storage" and take it from there, but that no longer qualifies as "minimal steps" IMO.

mluggy, 2021-08-02:
I think one thing I couldn't get from the code is what makes an external thumb drive plugged into the pi to show up on the Tesla. I tried following the USB_DRIVE setup but couldn't get it recognized (usb is mounted and shows up on the pi)

marcone, 2021-08-02:
Using a USB drive only works with Pi4. It's easiest to just make that USB drive the boot drive also (i.e. instead of using an sd card, use the USB drive exclusively). You don't need to specify the USB_DRIVE variable in that case.

mluggy, 2021-08-02:
Thanks! I have Pi4. So I should write the prebuilt image on that USB and just boot without an SD card? my intention is to have the USB easily detachable to be read in a computer if I want to.

marcone, 2021-08-02:
> Thanks! I have Pi4. So I should write the prebuilt image on that USB and just boot without an SD card? my intention is to have the USB easily detachable to be read in a computer if I want to.

Yes, though note that you need a recent-ish firmware on the Pi to let it boot from USB. If your Pi is relatively new, it should be good to go.

mluggy, 2021-08-03:
Setup worked really well (after a few reboots and minor diagnostic). The pi shows TeslaCam when plugged in entirely but if you connect solely the thumb drive it will only show boot, at least on windows.

marcone, 2021-08-03:
> Setup worked really well (after a few reboots and minor diagnostic). The pi shows TeslaCam when plugged in entirely but if you connect solely the thumb drive it will only show boot, at least on windows.

That's expected. There is only one partition that Windows can read, the other 3 partitions are readable by linux. Additionally the emulated drive(s) that the car sees are backed by files in one of the linux partitions, not by an actual partition.

> Also, is there a way to read/access the mp4 files while pi is connected to the car? I couldn't find / -name *.mp4 anything (snapshots are disabled, archive set to none and cam_size is 100%).

Sort of. You can mount the disk image used for drive emulation on the Pi, however if it is also connected to the car at the same time, then the Pi would see a file system that keeps changing out from under it.

mluggy, 2021-08-03:
Hmm..

> Sort of. You can mount the disk image used for drive emulation on the Pi, however if it is also connected to the car at the same time, then the Pi would see a file system that keeps changing out [...]

Explain? I just want read access to the files.. will turning on snapshots or one of the archives would do it for me?

marcone, 2021-08-03:
> Explain?

I already did. I'm not sure how else I can explain it. I suggest looking at the scripts to see how teslausb actually works.