marcone/teslausb #659: PI Zero W Single Blink no wireless

trinitron79, 2021-08-11:
I've tried two different SD Cards at this point and a new USB cable.   The unit will boot and configure itself just fine and act normal, even the LED will do the 2 blinks that everything is normal.   However after some time it will change to a steady single blink and lose network connectivity.

I've hooked the unit up to a monitor but during the boot process it will disable to video, this seems to be normal as it does the same thing during the build process then at some point during the build it will disable the video.

The Pi Zero seems to be working fine as I've used a 3rd SD card and installed the normal PI software which had zero problems for a over 24 hrs, never dropped a ping nor did the unit reboot or NOT display video.

Any log files I could grab to help understand this problem?  I get about 2 mins or so that I'm able to SSH into the unit before it drops that connection.

trinitron79, 2021-08-11:
was able to get the diag file created and copied before unit dropped from wifi.  Was doing the double blink now back to single blink and no wireless connection

marcone, 2021-08-21:
IIRC from discussion in the Discord, you narrowed this down to a problem with a specific access point?
Closing this for now, but if there's something to be done in the Pi/teslausb side please let me know.

closed 2021-08-21 by marcone