marcone/teslausb #662: Request for new prebuilt image

rmyadsk, 2021-08-14:
I’d like to make a case that it’s time for a new prebuilt image. The current image is from December 2020 and, when utilizing a 64GB SD card with default partitioning on a Raspberry Pi Zero, results in an error when attempting to complete Raspian OS updates via apt-get owing to a disk space error. Also, the current teslausb_setup_variables.conf is 233 lines and includes a considerable bit of refactoring since the included 178-line version included in the December image.

I am a fan of the "setup-teslausb upgrade" bootstrapping for application scripts, but I think perhaps biannual releases are appropriate to keep the critical mass of OS updates under control:

Here is disk allocation shortly after firstboot (“watch df -h” is useful to see these values update):

> pi@teslausb-blue:~ $ df -h
> Filesystem      Size  Used Avail Use% Mounted on
> /dev/root       1.8G  1.2G  458M  73% /
> devtmpfs        184M     0  184M   0% /dev
> tmpfs           216M     0  216M   0% /dev/shm
> tmpfs           216M  5.8M  211M   3% /run
> tmpfs           5.0M     0  5.0M   0% /run/lock
> tmpfs           216M     0  216M   0% /sys/fs/cgroup
> /dev/mmcblk0p1  253M   55M  198M  22% /boot
> none            216M  512K  216M   1% /tmp/sourcedir
> tmpfs            44M     0   44M   0% /run/user/1000

Here we are after initial setup, make-root-fs-readonly, and reboot:

> Filesystem      Size  Used Avail Use% Mounted on
> /dev/root       1.8G  1.2G  456M  73% /
> devtmpfs        184M     0  184M   0% /dev
> tmpfs           216M     0  216M   0% /dev/shm
> tmpfs           216M  5.8M  211M   3% /run
> tmpfs           5.0M     0  5.0M   0% /run/lock
> tmpfs           216M     0  216M   0% /sys/fs/cgroup
> tmpfs           216M     0  216M   0% /var/log/nginx
> tmpfs           216M     0  216M   0% /var/tmp
> tmpfs           216M     0  216M   0% /var/lib/nginx
> tmpfs           216M     0  216M   0% /tmp
> tmpfs           216M     0  216M   0% /var/spool
> /dev/mmcblk0p4   96M  1.6M   87M   2% /mutable
> /dev/mmcblk0p1  253M   55M  198M  22% /boot
> /dev/mmcblk0p3   56G  448M   56G   1% /backingfiles
> /dev/loop0       30G   48K   30G   1% /tmp/snapshots/snap-000000
> tmpfs            44M     0   44M   0% /run/user/1000

So far, so good. Now, attempt the following:

`apt-get update; apt-get upgrade`

> The following packages will be upgraded:
>   apt apt-utils avahi-daemon base-files bind9-host ca-certificates curl device-tree-compiler file firmware-atheros firmware-brcm80211
>   firmware-libertas firmware-misc-nonfree firmware-realtek groff-base iproute2 iputils-ping isc-dhcp-client isc-dhcp-common klibc-utils
>   libapt-inst2.0 libapt-pkg5.0 libavahi-common-data libavahi-common3 libavahi-core7 libbind9-161 libbsd0 libcurl4 libdns-export1104
>   libdns1104 libgcrypt20 libgnutls30 libgssapi-krb5-2 libhogweed4 libisc-export1100 libisc1100 libisccc161 libisccfg163 libk5crypto3
>   libklibc libkrb5-3 libkrb5support0 libldap-2.4-2 libldap-common liblwres161 liblz4-1 libmagic-mgc libmagic1 libnettle6 libp11-kit0
>   libpam-systemd libpolkit-agent-1-0 libpolkit-backend-1-0 libpolkit-gobject-1-0 libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc
>   libraspberrypi0 libssl1.1 libsystemd0 libudev1 libx11-6 libx11-data libxml2 libzstd1 linux-libc-dev openssh-client openssh-server
>   openssh-sftp-server openssl policykit-1 python-apt-common python-rpi.gpio python3-apt raspberrypi-bootloader raspberrypi-kernel
>   raspberrypi-sys-mods rpi-eeprom rpi-update rpi.gpio-common ssh sudo systemd systemd-sysv tzdata udev unzip wpasupplicant
> 88 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> Need to get 162 MB of archives.
> After this operation, 8,746 kB of additional disk space will be used.
> Do you want to continue? [Y/n] 

Doesn't sound so bad, however during the upgrade process, bad things happen:

> Adding 'diversion of /boot/overlays/wm8960-soundcard.dtbo to /usr/share/rpikernelhack/overlays/wm8960-soundcard.dtbo by rpikernelhack'
> Unpacking raspberrypi-kernel (1:1.20210805-1) over (1.20201201-1) ...
> dpkg: error processing archive /tmp/apt-dpkg-install-i3qblr/42-raspberrypi-kernel_1%3a1.20210805-1_armhf.deb (--unpack):
>  cannot copy extracted data for './lib/modules/5.10.52-v8+/kernel/net/sctp/sctp.ko' to '/lib/modules/5.10.52-v8+/kernel/net/sctp/sctp.ko.dpkg-new': failed to write (No space left on device)
> dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
> run-parts: executing /etc/kernel/postrm.d/initramfs-tools 5.10.52+ /boot/kernel.img
> run-parts: executing /etc/kernel/postrm.d/initramfs-tools 5.10.52-v7+ /boot/

Trying again, apt-get alerts us it needs the kernel, which is kind of a big deal:

> The following packages will be upgraded:
>   raspberrypi-kernel
> 1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> 73 not fully installed or removed.
> Need to get 0 B/79.6 MB of archives.
> After this operation, 3,592 kB disk space will be freed.
> Do you want to continue? [Y/n] 

Here we are right before the kernel update breaks things:

> Filesystem      Size  Used Avail Use% Mounted on
> /dev/root       1.8G  1.7G  2.4M 100% /
> devtmpfs        184M     0  184M   0% /dev
> tmpfs           216M     0  216M   0% /dev/shm
> tmpfs           216M  8.6M  208M   4% /run
> tmpfs           5.0M     0  5.0M   0% /run/lock
> tmpfs           216M     0  216M   0% /sys/fs/cgroup
> tmpfs           216M     0  216M   0% /var/log/nginx
> tmpfs           216M     0  216M   0% /var/spool
> tmpfs           216M   36K  216M   1% /tmp
> tmpfs           216M     0  216M   0% /var/tmp
> tmpfs           216M     0  216M   0% /var/lib/nginx
> /dev/mmcblk0p4   96M  1.6M   87M   2% /mutable
> /dev/mmcblk0p1  253M  1.3M  251M   1% /boot
> /dev/mmcblk0p3   56G  448M   56G   1% /backingfiles
> tmpfs            44M     0   44M   0% /run/user/1000

Error is repeatable:

> dpkg: error processing archive /var/cache/apt/archives/raspberrypi-kernel_1%3a1.20210805-1_armhf.deb (--unpack):
>  cannot copy extracted data for './lib/modules/5.10.52-v8+/kernel/drivers/usb/gadget/function/u_serial.ko' to '/lib/modules/5.10.52-v8+/kernel/drivers/usb/gadget/function/u_serial.ko.dpkg-new': failed to write (No space left on device)
> dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)

Here is apt-get’s coda:

> Errors were encountered while processing:
>  /var/cache/apt/archives/raspberrypi-kernel_1%3a1.20210805-1_armhf.deb
> E: Sub-process /usr/bin/dpkg returned an error code (1)

marcone, 2021-08-14:
> The current image is from December 2020

Is there any particular reason you want to update it? Any specific bugfixes or other improvements you're hoping to gain?
I consider teslausb an appliance. If it works, there's no real need to update the underlying operating system.

> Also, the current teslausb_setup_variables.conf is 233 lines and includes a considerable bit of refactoring since the included 178-line version included in the December image.

That's why there's a link to the current file in the one included in the image. I definitely don't want to make a new image every time the default config file is updated.

To get extra space, you can specify the `INCREASE_ROOT_SIZE` variable in the config file for initial setup. If you didn't do that for whatever reason, you can also free up quite a bit of space by deleting kernel modules you don't need before doing an update: `sudo rm -rf /lib/modules/!($(uname -r))`

All that being said, I've been meaning to make a new image that includes more packages by default, so that setup can be faster, but I don't plan to make new images on a regular basis just for the sake of being new.

CodeChief, 2021-08-14:
Is there any specific reason you cannot update it? Heard about DevOps? TDD? Why they make sense? Even if there were 1000 reasons to do nothing, the security holes, delay and effort requred by end users to get the thing running in a reasonable stand should already make you want to keep it at least binaurally up to date. Sorry to complain but really this is the kind of answer people make who are not innovating. Then the next fork is necessary.

rmyadsk, 2021-08-14:
> Is there any particular reason you want to update it? 

For one, security updates, as a matter of best practice. Any system that is not air-gapped has a threat surface, and beyond recording video, my VIN and Tesla credentials are visible in plaintext on the system drive.

> I consider teslausb an appliance. If it works, there's no real need to update the underlying operating system.

All the more reason to stamp a new image built on the most recent GA Raspberry Pi OS release. That OS (nee Raspbian) has had point releases on January 11, March 3, and May 7.

CodeChief, 2021-08-14:
Just automate tests and build pipeline. Then sit back and only have to deal with occasional package/dependency conflicts/warnings. Just looking at the long list of errors reported during boot raises massive cause for concern. Who knows perhaps a lot of the recent troubles are caused by the need to adjust a few parameters to the new format, i.e. " this parameter is no longer valid" being one of the main errors reported with the current TeslaUsb configuration.

marcone, 2021-08-15:
Sounds like new images will be regularly available at https://github.com/CodeChief/TeslaUsb from now on. Thanks!

closed 2021-08-20 by marcone