marcone/teslausb #664: Wrong fs type

bahree, 2021-08-18:
I have had Tesla_dashcam running without any issues on a pi w for a long time (I want to say 18+ months but I can't remember exactly). Every now and then I did do an update but mostly it was just working.

But now it stopped. The car started complaining that the USB drive was off (I can't remember the exact error) and I noticed, it wasn't copying saved events to the NAS.

Connecting it to a Windows computer, the drive mounts and I can see the following content, and that tells me this hasn't copied things off the SD card for a while now.
[image]
[image]

When I SSH in and run an upgrade I get the error wrong fs type. Any ideas on what is wrong? Is this a bug or is the SD card starting to go bad or something else? Below is the output when I run the upgrade.

root@teslausb-Model3:~# bin/setup-teslausb upgrade
preparing for full upgrade
/tmp/setup-teslausb is up to date
rc.local is up to date
Updating package index files...
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Fetched 15.0 kB in 3s (4,864 B/s)
Reading package lists... Done
Verifying that the requested configuration is valid...
Downloaded /tmp/verify-configuration.sh ...
USB_DRIVE is not set. SD card will be used.
Verifying that there is sufficient space available on the MicroSD card...
There is sufficient space available.
Downloading additional setup scripts.
Downloaded /root/bin/remountfs_rw ...
Downloaded /root/bin/make_snapshot.sh ...
Downloaded /root/bin/mount_snapshot.sh ...
Downloaded /root/bin/release_snapshot.sh ...
Downloaded /root/bin/force_sync.sh ...
Downloaded /tmp/create-backingfiles-partition.sh ...
Downloaded /tmp/create-backingfiles.sh ...
Downloaded /tmp/make-root-fs-readonly.sh ...
Downloaded /tmp/configure.sh ...
~ ~
Fixing the modules-load parameter in /boot/cmdline.txt...
Fixed cmdline.txt.
USB_DRIVE not set. Proceeding to SD card setup
create-backingfiles-partition: using existing backingfiles and mutable partitions
umount: /mnt/cam: not mounted.
mount: /mnt/cam: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.
STOP: Couldn't check image

Thanks!

marcone, 2021-08-18:
Did you format the drive in the car or in Windows?
What does `fdisk -l /backingfiles/cam_disk.bin` say?

bahree, 2021-08-18:
Appreciate the quick response. No, haven't formatted it in the car or in Windows. Here is what I get:

[image]

And the text in case folks need to search:

> root@teslausb-Model3:~# fdisk -l /backingfiles/cam_disk.bin
> Disk /backingfiles/cam_disk.bin: 49.4 GiB, 53033979904 bytes, 103581992 sectors
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 512 bytes
> I/O size (minimum/optimal): 512 bytes / 512 bytes
> Disklabel type: dos
> Disk identifier: 0x8940c71b
> 
> Device                      Boot Start       End   Sectors  Size Id Type
> /backingfiles/cam_disk.bin1       2048 103581991 103579944 49.4G  c W95 FAT32 (LBA)
> root@teslausb-Model3:~#
> 

Any ideas on what might be wrong? I am happy to reformat and install the image again if that might be easier.

Thanks.

marcone, 2021-08-18:
I don't see anything obviously wrong there. What happens if you run `sudo mount /mnt/cam` manually?

bahree, 2021-08-18:
When I do that, I get the original error:

```
root@teslausb-Model3:~# sudo mount /mnt/cam
mount: /mnt/cam: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.
root@teslausb-Model3:~#

```

Maybe I can flash it again and see if that fixes it.

marcone, 2021-08-18:
Flashing again will almost certainly fix the problem, but it would be good to find out what's actually wrong.
My guess is either something is wrong in /etc/fstab, or the disk image is corrupted in such a way that Windows will still accept it as a valid drive, while linux does not.

bahree, 2021-08-18:
I agree and I suspect it might be the latter - given I haven't touch this for months and it was working fine, until it wasnt. Possibly something didn't finish writing whenever the car went to sleep and powered off, corrupting something. I will flash again and report back here.

closed 2021-08-21 by marcone