marcone/teslausb #665: Latest version one step setup / after unable to SSH. Invalid credentials

jtslade777, 2021-08-20:
After the one step setup Tesla USB - Marcone is working perfectly but no matter what combination of default credentials found in documentation am unable to SSH to the Pi. Has credentials or SSH been disabled in the latest version.

marcone, 2021-08-20:
ssh is still enabled and the default credentials are the same (user: pi, password: raspberry)
Can you reach the web interface?

jtslade777, 2021-08-20:
Yes I am able to reach the Web interface

marcone, 2021-08-21:
So what happens when you `ssh pi@teslausb` (where "teslausb" should be replaced with whatever name or IP address you use to access the web interface)

jtslade777, 2021-08-21:
When I try and connect with putty:

Using username "pi".
pi@192.168.1.121's password:
Access denied
pi@192.168.1.121's password:

marcone, 2021-08-21:
> When I try and connect with putty: (using rasberry as password)

Just in case that wasn't a typo: it's "raspberry", not "rasberry"

jtslade777, 2021-08-22:
That fixed it, I had been cut /pasting the password and it had a mistake. TY!

closed 2021-08-22 by marcone