marcone/teslausb #667: [RFE] Verbose output during file transfers

jolleyutah, 2021-08-21:
It would be really useful to have a verbose output tab on the Web Interface (or in archiveloop.log) to be able to monitor the status of music files being copied from music archive, or for video files being sent to the archive.

githubsean, 2022-05-22:
you can try looking at /tmp/rsynclog and/or /tmp/archive-rsync-cmd.log
For example, for me, I can see in /mutable/archiveloop.log
`Archiving 22 event folder(s) with 653 file(s) starting at Sun 22 May 18:33:10 AEST 2022`

And then in /tmp/archive-sync-cmd.log, I can see each file - and it has done 36 files - so a lot remaining. When this happens I manually put the car in Sentry Mode (normally it is off when in my garage), so that everything stays powered up. I'll check back in a few hours once it has finished the transfer.

mulka, 2024-10-25:
I'd also like to see this for the rclone archiving.