marcone/teslausb #669: Updating to the latest version

tux43, 2021-08-23:
Hi,
I am trying to run self-update and it is failing with a repository error. Suggestions on what I am doing wrong?

`root@teslafunnel:~/bin# /root/bin/setup-teslausb 
WARNING: main-dev contains a different version of /root/bin/setup-teslausb. It is recommended to run '/root/bin/setup-teslausb selfupdate' to update to that version
rc.local is up to date
Updating package index files...
Hit:1 http://archive.raspberrypi.org/debian buster InRelease                   
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]      
^Cading package lists... 47%   
root@teslafunnel:~/bin# /root/bin/setup-teslausb selfupdate
/root/bin/setup-teslausb updated
other scripts may need to be updated by running /root/bin/setup-teslausb
root@teslafunnel:~/bin# /root/bin/setup-teslausb 
/root/bin/setup-teslausb is up to date
rc.local is up to date
Updating package index files...
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Reading package lists... Done                              
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
`

JohPePe, 2021-08-23:
I ran “sudo apt update --allow-releaseinfo-change”
Solved the problem for me! 😀

marcone, 2021-08-23:
I see the same thing when running `apt update`, but on a recently configured Pi it only happened the first time, and it wasn't an error:
```
root@teslausb2:~# apt update
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 http://archive.raspberrypi.org/debian buster/main armhf Packages [378 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]  
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]                                                                                              
Fetched 13.4 MB in 1min 12s (186 kB/s)                                                                                                                                           
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
N: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
root@teslausb2:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
root@teslausb2:~# 
```
On a Pi that was last updated a few months ago it's an error like @tux43 described.
I've added the `--allow-releaseinfo-change` option to the setup script.

commit 7eba6dcd (2021-08-23): Allow release info change when updating package index

Fixes #669
closed 2021-08-23 by marcone