marcone/teslausb #672: Unable to upgrade

sirHulda, 2021-09-05:
Hello,
Trying to upgrade but getting this error:

preparing for full upgrade
/tmp/setup-teslausb is up to date
rc.local is up to date
Updating package index files...
E: Command line option --allow-releaseinfo-change is not understood in combination with the other options
Failed, retrying
E: Command line option --allow-releaseinfo-change is not understood in combination with the other options
Failed, retrying
E: Command line option --allow-releaseinfo-change is not understood in combination with the other options
Failed, retrying
E: Command line option --allow-releaseinfo-change is not understood in combination with the other options
Failed, retrying
E: Command line option --allow-releaseinfo-change is not understood in combination with the other options
Failed, retrying
E: Command line option --allow-releaseinfo-change is not understood in combination with the other options
Failed, retrying

"--allow-releaseinfo-change" only works with apt update and not apt upgrade.
https://manpages.debian.org/unstable/apt/apt-get.8.en.html

marcone, 2021-09-05:
`setup-teslausb` only calls `apt-get update --allow-releaseinfo-change`, so not sure what "other options" it is complaining about there.
Which version of `apt` do you have installed? (just run `apt` without arguments and it will tell you)

sirHulda, 2021-09-05:
Just updated it because it didnt work. But getting same error even after upgrade.
apt 1.4.11 (armhf)

Running apt-get update --allow-releaseinfo-change manually throwing the same error as well.

marcone, 2021-09-05:
Are you still running the "Stretch" release of Raspbian?

sirHulda, 2021-09-05:
Maybe, i have another setup that works fine. Not sure if that's Buster or Stretch..
I'll reinstall and try again.

commit c0313f37 (2021-09-07): Make setup work on Stretch

Try apt update with and without allow-releaseinfo-change option,
since that option doesn't work with the version of apt used in
Stretch.

Fixes #672
closed 2021-09-07 by marcone