marcone/teslausb #674: chmod: changing permissions of 'TeslaCam': Function not implemented

ktsenvolt, 2021-09-06:
Hardware:Raspberrypi Zero W
Remote PC:
I wish I could delete TeslaCam data under var/www/html/TeslaCam. to save space .(Lastest  teslausb-20210815)
Here is what I have ran into a problem:
root@teslausb:/var/www/html# sudo -i
Run 'bin/setup-teslausb upgrade' to update to the latest version of teslausb,
or run 'bin/remountfs_rw' to allow writing to the root partition.

root@teslausb:~# bin/remountfs_rw
Remounting / as read/write
Remounting /boot as read/write
Done. Make any changes needed, and reboot the Pi when ready.
Filesystems / and /boot will be mounted as read only after reboot.
root@teslausb:~# pwd
/root
root@teslausb:~# cd /var/www/html
root@teslausb:/var/www/html# ls -al
total 53
drwxrwxrwx 4 root root  4096 Sep  6 13:15 .
drwxrwxrwx 3 root root  4096 Sep  6 13:15 ..
lrwxrwxrwx 1 root root    24 Sep  6 13:15 archiveloop.log -> /mutable/archiveloop.log
drwxr-xr-x 2 root root  4096 Sep  6 13:15 cgi-bin
-rw-r--r-- 1 root root   900 Sep  6 13:15 diagnostics.html
lrwxrwxrwx 1 root root    20 Sep  6 13:15 diagnostics.txt -> /tmp/diagnostics.txt
-rw-r--r-- 1 root root   318 Sep  6 13:15 favicon.ico
-rw-r--r-- 1 root root 31361 Sep  6 13:15 index.html
drwxr-xr-x 5 root root  1024 Aug 29 23:32 TeslaCam
lrwxrwxrwx 1 root root    33 Sep  6 13:15 teslausb-headless-setup.log -> /boot/teslausb-headless-setup.log
root@teslausb:/var/www/html# sudo chmod 0777 TeslaCam
chmod: changing permissions of 'TeslaCam': Function not implemented

I have log in as root. Can not figure out why I can't change permission?

Thanks for the help.

rmyadsk, 2021-09-06:
`var/www/html/TeslaCam` is the source folder for the Web interface, and the disk utilization there should be trivial. To check, go to that directory and execute a `du -h`.

The largest drive usage should come from video storage itself, and from snapshots, but the system should automatically manage itself you're never in danger of running out of available space.

marcone, 2021-09-06:
/var/www/html/TeslaCam does not take up any space, as it's simply a mountpoint for a FUSE filesystem.
What are you actually trying to achieve?

ktsenvolt, 2021-09-06:
Thanks for clarifying.

ktsenvolt, 2021-09-06:
Here is the result after df -h
445M    ./RecentClips/2021-09-04
24G     ./RecentClips/2021-08-31
15G     ./RecentClips/2021-09-02
33K     ./RecentClips/20719C~1.M
33K     ./RecentClips/event.json
942M    ./RecentClips/2021-09-05
488M    ./RecentClips/2021-08-29
144M    ./RecentClips/2021-09-06
3.0G    ./RecentClips/2021-08-30
33K     ./RecentClips/thumb.png
43G     ./RecentClips
823M    ./SentryClips/2021-09-02_11-30-04
1.3G    ./SentryClips/2021-08-31_08-46-20
1.5G    ./SentryClips/2021-08-31_10-05-23
171M    ./SentryClips/2021-08-31_11-00-51
1.5G    ./SentryClips/2021-09-02_11-08-33
34M     ./SentryClips/2021-08-31_10-32-11
1.1G    ./SentryClips/2021-09-02_10-21-49
878M    ./SentryClips/2021-09-02_10-43-31
1.5G    ./SentryClips/2021-08-31_10-59-35
830M    ./SentryClips/2021-08-31_10-39-32
1.6G    ./SentryClips/2021-09-02_10-12-27
731M    ./SentryClips/2021-09-02_10-48-45
575M    ./SentryClips/2021-09-02_11-22-09
1.1G    ./SentryClips/2021-09-02_11-16-37
1.2G    ./SentryClips/2021-08-31_10-31-54
15G     ./SentryClips
1.2G    ./SavedClips/2021-09-02_10-37-17
142M    ./SavedClips/2021-08-29_23-47-33
965M    ./SavedClips/2021-08-31_09-53-44
151M    ./SavedClips/2021-09-04_12-15-26
1.6G    ./SavedClips/2021-08-31_08-36-09
1.5G    ./SavedClips/2021-08-31_10-23-29
1.6G    ./SavedClips/2021-08-31_11-21-45
28M     ./SavedClips/2021-08-29_23-47-47
6.9G    ./SavedClips
64G     .

marcone, 2021-09-07:
Again, that's not actually space used in /var/www, or even in the root partition. What are you actually trying to achieve?

closed 2021-09-12 by marcone