marcone/teslausb #675: Setup Variables File Changes

GoodyGizmos, 2021-09-10:
What is the easiest way to make a quick change to the setup variables file (e.g. new IFTTT config) for an installation already in use? I know to make the file system RW per the instructions, but not sure where to edit the file currently in use with 'sudo nano..."

rmyadsk, 2021-09-10:
You'll first want to make the filesystem r/w, then edit `/root/teslausb_setup_variables.conf` to fit your desired changes, then run `/root/bin/setup-teslausb upgrade` to make those changes take effect. The end of that last step will set the filesystem back to r/o and trigger a reboot with your changes intact.

marcone, 2021-09-10:
What @rmyadsk said. Note that not all options can be changed after initial setup, but changing the IFTTT config shouldn't be a problem.

closed 2021-09-10 by marcone