marcone/teslausb #678: Large number of files, slow to archive

TJCOS, 2021-09-14:
The latest pre-built image 20210815 seems to be creating ~10x the number of files as 20201211. Is this a setting in the config file?

marcone, 2021-09-15:
Not sure what you mean. The car creates 4 files per minute. Which prebuilt image you use doesn't change that.

TJCOS, 2021-09-15:
For example, yesterday it stated "Archiving 6 event folders with 2787 files." I see 2737 files in a TeslaCam\RecentClips folder on my server, but 45 files in each of the 6 SentryClips folders for a total of 270. Is this because I have export ARCHIVE_RECENTCLIPS=true ?

marcone, 2021-09-15:
Yes, if you added `ARCHIVE_RECENTCLIPS=true`, it'll archive *everything* it recorded, not just Sentry events, and the car is always recording when it's on.
So this is working as expected.

closed 2021-09-15 by marcone

TJCOS, 2021-09-15:
Thank you. Makes sense now that I looked into it. My apologies and love the product that you have created and shared.