marcone/teslausb #679: exFat Trimming free space in /mnt/cam failed

TommyUSB, 2021-09-16:
I use an RPI zero W and have set up the installation twice. With the latest teslausb-20210815 image and teslausb_setup_variables.conf with export USE_EXFAT = true. The CAM directory is then exFat. But every time I get the message in the archive loop log: Trimming free space in / mnt / cam failed. See archive loop log. Is the function not supported with exFat instalation or what could be the reason?

[archiveloop.log](https://github.com/marcone/teslausb/files/7180174/archiveloop.log)

marcone, 2021-09-16:
Exfat doesn't support the trim/discard operation. I recommend against using it at this point.

closed 2021-09-16 by marcone

TommyUSB, 2021-09-16:
Thank you for your prompt reply. Can I somehow bypass the error message in the archiveloop log or is it not relevant for the function?

marcone, 2021-09-16:
I'm not sure what you mean by "bypass the error message". It's just a warning, and doesn't affect functionality other than that storage will be used suboptimally.