marcone/teslausb #682: teslausb-20210815 won't install/setup on RPi0w

alanhoyle, 2021-09-28:
I decided to upgrade a working setup to the lastest release by resetting everything from scratch.  

I downloaded the teslausb-20210815.zip, burned it to the same microSD with balenaEtcher.  I edited `teslausb_setup_variables.conf.sample` with my settings and saved it into `teslausb_setup_variables.conf` in the boot folder.  (my settings included the CIFS setup and the two WiFi lines.  

When I put the card back into the Pi and plug it into power, the WiFi seemingly never comes up.  

After letting it sit for a while, I unplug the Pi and move the card over to my reader.  In the `boot` volume, I see `TESLA_USB_SETUP_STARTED`, `WIFI_ENABLED` files, but the `teslausb_setup_variables.conf` has disappeared.  `teslausb-headless-setup.log` is there and looks like the following:

```
Mon 16 Aug 02:47:45 BST 2021 : Detecting whether to update wpa_supplicant.conf
Mon 16 Aug 02:47:45 BST 2021 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Mon 16 Aug 02:47:46 BST 2021 : Rebooting...
Mon 16 Aug 02:48:28 BST 2021 : Detecting whether to update wpa_supplicant.conf
Mon 16 Aug 02:48:28 BST 2021 : Grabbing main setup file.
Mon 16 Aug 02:48:40 BST 2021 : get_script failed, retrying
Mon 16 Aug 02:48:53 BST 2021 : get_script failed, retrying
Mon 16 Aug 02:49:06 BST 2021 : get_script failed, retrying
```
The `get_script failed, retrying` line is repeated 149 times in the log on my latest try.  

Any special characters in my setup are quoted.  

Any hints?

marcone, 2021-09-28:
The 20210815 image works on Pi0w, I set one up from scratch just a few days ago.
You said that WiFi seemingly never comes up, but did you conclude that because you never see the device show up on your network/router, or because setup didn't complete?
Judging from the log snippet you posted, the clock hadn't been updated. That could be caused by networking not working at all, or by NTP being blocked somehow on your network. If would recommend plugging the Pi into a PC for the initial setup, since that should provide a network connection over USB, which might be helpful for troubleshooting.

alanhoyle, 2021-09-29:
Sigh.  I put the wrong password in.  sorry.

closed 2021-09-29 by alanhoyle