marcone/teslausb #683: No changes detected in Snapshot during archive, but /backingfiles usage is at 93%

syspimp, 2021-09-29:
When archiving, nothing is sent over and nothing is removed. What is the process to clean off the files in /backingfiles...Clips?  I can rsync them off, but the filesystem is read-only and I would like to have the supported cleaning operation.

```shell
[me@xxxxx ~]$ ssh pi@teslausb
Warning: Permanently added 'teslausb,10.x.x.x' (ECDSA) to the list of known hosts.
Linux teslausb 4.19.93+ #1290 Fri Jan 10 16:34:37 GMT 2020 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Run 'sudo -i' if you need to make changes.

pi@teslausb:~ $ df -h | grep mmc
/dev/mmcblk0p4   93M   25M   62M  29% /mutable
/dev/mmcblk0p1  253M  118M  135M  47% /boot
/dev/mmcblk0p3  237G  220G   17G  93% /backingfiles
pi@teslausb:~ $ sudo /root/bin/force_sync.sh
Forcing archiveloop to sync by pretending to take archive host offline and back online.
Setting archive unreachable..
Waiting for archiveloop to see canary file..
Done!  archiveloop process should now start its sync process automagically as soon as it sees the archive.
pi@teslausb:~ $ tail /mutable/archiveloop.log -f
Tue 28 Sep 20:50:31 EDT 2021: Simulating archive being unreachable.
Tue 28 Sep 20:50:32 EDT 2021: Waiting for archive to be reachable...
Tue 28 Sep 20:50:32 EDT 2021: Archive is reachable.
Tue 28 Sep 20:50:32 EDT 2021: Trying to set time...
Tue 28 Sep 20:50:32 EDT 2021: Time adjusted by 0.278407 seconds after 0.280000 seconds
Tue 28 Sep 20:50:32 EDT 2021: not keeping car awake.
Tue 28 Sep 20:50:53 EDT 2021: taking snapshot of cam disk in /backingfiles/snapshots/snap-001545
Tue 28 Sep 20:51:48 EDT 2021: took snapshot
Tue 28 Sep 20:52:03 EDT 2021: comparing new snapshot with /backingfiles/snapshots/snap-001544/snap.bin
Tue 28 Sep 20:52:03 EDT 2021: new snapshot is identical to previous one, discarding
Tue 28 Sep 20:52:46 EDT 2021: Disconnecting usb from host...
Tue 28 Sep 20:52:47 EDT 2021: Disconnected usb from host.
Tue 28 Sep 20:52:47 EDT 2021: Running fsck on /backingfiles/cam_disk.bin...
Tue 28 Sep 20:52:47 EDT 2021: | fsck from util-linux 2.33.1
Tue 28 Sep 20:53:06 EDT 2021: | fsck.fat 4.1 (2017-01-24)
Tue 28 Sep 20:53:06 EDT 2021: | Long filename fragment "2020-12-10_18" found outside a LFN sequence.
Tue 28 Sep 20:53:06 EDT 2021: |   (Maybe the start bit is missing on the last fragment)
Tue 28 Sep 20:53:06 EDT 2021: |   Not auto-correcting this.
Tue 28 Sep 20:53:06 EDT 2021: | Long filename fragment "2020-12-04_15" found outside a LFN sequence.
Tue 28 Sep 20:53:06 EDT 2021: |   (Maybe the start bit is missing on the last fragment)
Tue 28 Sep 20:53:06 EDT 2021: |   Not auto-correcting this.
Tue 28 Sep 20:53:06 EDT 2021: | Long filename fragment "2020-12-01_10" found outside a LFN sequence.
Tue 28 Sep 20:53:06 EDT 2021: |   (Maybe the start bit is missing on the last fragment)
Tue 28 Sep 20:53:06 EDT 2021: |   Not auto-correcting this.
Tue 28 Sep 20:53:06 EDT 2021: | Long filename fragment "2020-11-30_11" found outside a LFN sequence.
Tue 28 Sep 20:53:06 EDT 2021: |   (Maybe the start bit is missing on the last fragment)
Tue 28 Sep 20:53:06 EDT 2021: |   Not auto-correcting this.
Tue 28 Sep 20:53:06 EDT 2021: | /dev/loop1p1: 8996 files, 6134076/6744448 clusters
Tue 28 Sep 20:53:06 EDT 2021: Finished fsck on /backingfiles/cam_disk.bin.
Tue 28 Sep 20:53:06 EDT 2021: Archiving...
Tue 28 Sep 20:53:06 EDT 2021: Ensuring cam archive is mounted...
Tue 28 Sep 20:53:06 EDT 2021: Mounting /mnt/archive...
Tue 28 Sep 20:53:06 EDT 2021: Mounted /mnt/archive.
Tue 28 Sep 20:53:06 EDT 2021: Ensured cam archive is mounted.
Tue 28 Sep 20:53:06 EDT 2021: Checking saved folder count...
Tue 28 Sep 20:53:06 EDT 2021: Ensuring cam file is mounted...
Tue 28 Sep 20:53:06 EDT 2021: Mounting /mnt/cam...
Tue 28 Sep 20:53:06 EDT 2021: Mounted /mnt/cam.
Tue 28 Sep 20:53:06 EDT 2021: Ensured cam file is mounted.
Tue 28 Sep 20:53:10 EDT 2021: There are 0 event folder(s) with 0 file(s) and 0 track mode file(s) to move.  0 short recording(s) will be skipped.
Tue 28 Sep 20:53:10 EDT 2021: cleaning cam mount
Tue 28 Sep 20:53:26 EDT 2021: done cleaning cam mount
Tue 28 Sep 20:53:29 EDT 2021: Trimming free space in /mnt/cam, which has 107717 extents
Tue 28 Sep 20:53:32 EDT 2021: Trim complete, image now has 107717 extents
Tue 28 Sep 20:53:32 EDT 2021: Unmounting /mnt/cam...
Tue 28 Sep 20:53:32 EDT 2021: Unmounted /mnt/cam.
Tue 28 Sep 20:53:32 EDT 2021: Finished archiving.
Tue 28 Sep 20:53:32 EDT 2021: Music archive not configured or unreachable
Tue 28 Sep 20:53:32 EDT 2021: unmounting /mnt/archive
Tue 28 Sep 20:53:33 EDT 2021: Truncating log...
Tue 28 Sep 20:53:33 EDT 2021: Connecting usb to host...
Tue 28 Sep 20:53:33 EDT 2021: Connected usb to host.
Tue 28 Sep 20:53:33 EDT 2021: Waiting for archive to be unreachable...

pi@teslausb:~ $ df -h | grep mmc
/dev/mmcblk0p4   93M   25M   62M  29% /mutable
/dev/mmcblk0p1  253M  118M  135M  47% /boot
/dev/mmcblk0p3  237G  220G   17G  93% /backingfiles
pi@teslausb:~ $ ls /mutable/TeslaCam/SentryClips/|tail -n 5
2021-08-30_09-13-55
2021-08-30_11-57-21
2021-08-30_17-06-37
2021-08-30_17-13-30
2021-08-30_19-36-19
pi@teslausb:~ $ 
```

marcone, 2021-09-29:
> When archiving, nothing is sent over and nothing is removed. What is the process to clean off the files in /backingfiles...Clips? I can rsync them off, but the filesystem is read-only and I would like to have the supported cleaning operation.

Nothing is sent over because there is nothing to send (in other words, there are no Sentry files to archive that haven't already been archived).
Nothing is removed because TeslaUsb doesn't need the space yet.
This is all working as intended.

closed 2021-09-29 by marcone

syspimp, 2021-09-29:
Ah, I forgot to mention that NOTHING has been archived or recorded on the teslausb since 8/30 when it reached the max limit.

Previously I had the CAM_SIZE set to 90% on a 256 sd card, but once it hit the 93% limit it stopped archiving. I read the instructions which said to use a smaller size, so I changed it to 30%, but the usage has not gone back down.

I think the simplest step would be to reimage the card and start over. Agree?

marcone, 2021-09-29:
Yes, reimaging is usually the easiest. Note that the value of CAM_SIZE is independent of how much space is used in /backingfiles. Even if you set CAM_SIZE to 30% (which is probably still a lot more than you need, I would use the default '30G'  instead), /backingfiles will eventually fill up to 90-ish percent. This is by design.

miles267, 2021-09-29:
> Yes, reimaging is usually the easiest. Note that the value of CAM_SIZE is independent of how much space is used in /backingfiles. Even if you set CAM_SIZE to 30% (which is probably still a lot more [...]

If setup appropriately by using CAM_SIZE=30G, will the contents of /backingfiles continue to be managed automatically by teslausb?  Or do you inevitably reach a point where /backingfiles fills up and the sdcard must simply be re-imaged?

marcone, 2021-09-29:
> Or do you inevitably reach a point where /backingfiles fills up and the sdcard must simply be re-imaged?

No, teslausb will manage the available space.

syspimp, 2021-09-29:
Thanks again. Great project, btw. Your work is appreciated.