marcone/teslausb #685: Raspberry Pi 4 coil whine

edwardrossi, 2021-10-03:
Hi all,

This topic was surfaced in the discord but I wanted to get a larger sample size.

Is anyone else observing fairly loud coil whine on the Pi 4 and if so, what did your troubleshooting find? 

I'm wondering whether it a QA issue with my Pi and a new one will remedy this?
I have a 2021 M3 MIC SR+, Pi 4 (4gb ram) in an armour passive heatsink case plugged into the glove box USB port.
I'm currently booting from the 128gb Tesla USB, but the whine persisted when booting from SD card as well.

roflmao, 2021-10-08:
I can confirm this, 2021 M3MIC SR+, Pi4 4gbram in the glovebox.

roflmao, 2021-10-11:
@edwardrossi i put in in the center console through a usb hub, no whine.

edwardrossi, 2021-10-30:
After reading through #564 again, It appears the voltage coming out the glovebox USB is not 5V but 5.2V. I wonder if some usb hubs (like the one you're using) are regulating the input voltage to the USB standard of 5V?

@roflmao would you be able to test the USB hub in the Glovebox using a USB 3.0/3.1 Type-A to Type-C adapter?

marcone, 2021-10-30:
T

> After reading through #564 again, It appears the voltage coming out the glovebox USB is not 5V but 5.2V. I wonder if some usb hubs (like the one you're using) are regulating the input voltage to the USB standard of 5V?

USB allows up to 5.25V, so 5.2V is still within spec.

CodeChief, 2021-11-01:
> > After reading through #564 again, It appears the voltage coming out the glovebox USB is not 5V but 5.2V. I wonder if some usb hubs (like the one you're using) are regulating the input voltage to the USB standard of 5V?
> 
> USB allows up to 5.25V, so 5.2V is still within spec.

[image]

Actually I measured 5.48v with a RasPi 4.

roflmao, 2021-11-02:
@edwardrossi I dont have one at the time being, but will try to get a hold of one.

CodeChief, 2021-11-03:
According to my son who's an electronics master this is most likely caused by impedance mismatch. This is difficult to correct on the "load side" as you have to know the impedance of the PCB design.

You can't just whack resistors on there as the device already has resistors according to USB specification for identification and speed select.

From what I understand, I think we have the same general problem here as with the other USB adapter/glovebox issues. Although it sounds like a Tesla design problem (for example if their PCB traces/resistors don't amount to the impedance of USB specifications or are living on the edge of them), it might still be a RasPi configuration/hardware problem. 

How does the RasPi identify itself correctly when switching behaviour from a USB device to USB gadget? Does it really appear as a USB storage device with the correct impedance for USB 3.x or if this just a leftover Linux Kernel hack and we are lucky it works at all as it's really out of spec, not the car?

The Pi is really a special device with this capability, like a Smartphone, so we should not assume it's Tesla's fault and first look at any issues (and potential hackable fixes) on the Pi. I am not a hardware engineer so all I can do is hope that points some others in the right direction.

I guess a sensible test to rule out configuration would be to connect a RasPi without the gadget driver. Just a completely plain image, perhaps also load the EEPROM recovery image first to make sure it's factory default. Then repeat the same test with the USB gadget mode enabled, but only that as [originally documented](https://www.reddit.com/r/teslamotors/comments/9m9gyk/build_a_smart_usb_drive_for_your_tesla_dash_cam/) , on its own, not the whole TeslaUSB image.

marcone, 2021-11-23:
Closing this since it's not a teslausb issue

closed 2021-11-23 by marcone