marcone/teslausb #686: issue upon first use.

jamesp132, 2021-10-10:
Hello, 

I have just rebuilt my SD card as I was facing some issues with the usb not backing up my data to google drive

but now when the build completes I get this error:

> Welcome to nginx!

If you see this page, the nginx web server is successfully installed and working. Further configuration is required.

For online documentation and support please refer to nginx.org.
Commercial support is available at nginx.com.

Thank you for using nginx.
> 

I am unable to work out how I can get past this webpage to show the normal webpage again, I can SSH into the raspberry pi no problem, but unsure what "further configuration is required"

I have attempted to restore the SD card from 2 different computers and still the same issue. 

image I'm using for the build: 2021-08-15-teslausb.img

I have searched the issue pages but couldn't see anything similar to this, so if anyone knows what to do ill greatly appreciate the support.
[image]

marcone, 2021-10-10:
Can you describe what exactly you did, starting with flashing 2021-08-15-teslausb.img onto the sd card?

jamesp132, 2021-10-10:
sure! thanks for your response, 

so I formatted the drive as fat32 on my MacBook to remove all the data before. 

I then downloaded the latest image from https://github.com/marcone/teslausb/releases/tag/v3.0, and used the .img file in Etcher and set the wiped SD card as the one to put the image onto it then completed

then I changed the file teslausb_setup_variables to include my wifi - change the storage of the drive for the videos

moved the SD card over to the raspberry pi zero WH and proceeded to boot it, I've let it sit for 15 mins or more before touching it, when I SSH into the device it works fine I can run commands and connect it but if I try visit the "TeslaUSB.local" website I get the above error.

this happens every time even after a reinstall of the image, I have tried two computers both have the exact same error

if I am doing a step wrong please let me know

marcone, 2021-10-11:
Logs show that you initially didn't provide a configuration file. Once you did, setup completed (and then you reran setup 2 more times after that)
My guess is that when nginx was showing the standard welcome page you showed above, setup simply hadn't completed yet.

closed 2021-10-11 by marcone