marcone/teslausb #688: What are the working options that keeps the car awake to transfer recordings now that Tesal API no longer work?

chilman408, 2021-10-13:
Just wondering what is everybody doing to keep the car awake while the Pi is transferring the recordings to the nas.

marcone, 2021-10-20:
> It is not going to beat getting a huge SSD and manually copying over, say, every 60 days...

Doing something for a few minutes every 45 days doesn't seem like a huge difference compared to doing something for a few minutes every 60 days. But sure, technically it's less work and if all you care about is sentry recordings then an SSD will work just fine. Chances are that after 60 days you won't even care about those recordings anymore, and reformatting the drive is even less work than copying everything off of it.

That being said, if someone wants to automate the authentication process described above, or add an option in the web interface to add/refresh the token, that'd be great. Given Tesla's track record it's probably better to rely on third party authentication software and just have a way of adding a token to teslausb. (I tried integrating TeslaPy into teslausb, but that only worked for about a week before Tesla started requiring solving captchas)

chilman408, 2021-10-20:
Thanks, is there another method of keeping the Pi awake in-lieu of connecting with Tesla?

Poncherelly, 2021-10-20:
I just turn on my climate for a bit. Sometimes I forget to turn it back off though and have received a notice from the Tesla app.

marcone, 2022-01-15:
https://github.com/marcone/teslausb/commit/d91a63a054267057383612689aab443925fea40f makes the Tesla API functional again, though you'll have to use an external app to get a token (I've only tested the Android app linked from the sample configuration file)

closed 2022-01-15 by marcone

miles267, 2022-01-15:
> d91a63a

Cool! I’ve updated my config to include the refresh token. Can’t tell
whether it’s working yet. Is there any way to tell in the log?

marcone, 2022-01-15:
> Is there any way to tell in the log?

Yes, the log will show a message saying whether or not it's trying to keep the car awake, and which method it is using.

miles267, 2022-01-16:
Which log file? What should it say when working correctly?

miles267, 2022-01-16:
After updating teslausb and configuring a line for export tesla_refresh_token= (pasted working token here), I'm not seeing any reference to it in the diagnostics via the http://teslausb page.  Unless I should be looking elsewhere.  Should I be looking elsewhere?

miles267, 2022-01-16:
@marcone after updating teslausb and configuring a line for export tesla_refresh_token= (pasted working token here), I'm seeing this in achiveloop log via the http://teslausb page.

Sun 16 Jan 09:25:00 CST 2022: Time adjusted by 14861.398504 seconds after 0.150000 seconds
Sun 16 Jan 09:25:00 CST 2022: not keeping car awake.

Should the refresh token (not access token) be between in quotation marks in the config file? Also with this update, should we also be using export tesla_wake_mode=stream? I currently have that enabled.

marcone, 2022-01-16:
Did you use `export tesla_refresh_token=...` or `export TESLA_REFRESH_TOKEN=...`? You must use the latter. Did you run setup-teslausb afterwards?

miles267, 2022-01-16:
> Did you use `export tesla_refresh_token=...` or `export TESLA_REFRESH_TOKEN=...`? You must use the latter. Did you run setup-teslausb afterwards?

1.) Confirming I've used export TESLA_REFRESH_TOKEN=(with my refresh token here; not contained within single or double quotes)
2.) Have also tried with export TESLA_WAKE_MODE=stream either enabled or disabled (commented out) without success.
3.) confirming I ran the setup-teslausb after each attempted change

when I then check the log it is still saying "not keeping car awake".  Am sure the refresh token is good.  Created it same way as I did for teslamate which is working fine.

miles267, 2022-01-16:
Has anyone else gotten this to work for them that can offer any guidance please?  Thank you in advance.

marcone, 2022-01-16:
Look at the setup log, search for "tesla_api"

miles267, 2022-01-16:
> Look at the setup log, search for "tesla_api"

Thanks.  I do see in the headless setup log it saying:
Sun 26 Dec 18:20:08 CST 2021 : configure: tesla_api.py setup failed

Though I'm unable to locate info on why this might be?

nelsongraca, 2022-01-16:
from the log seems to fail to mount /mutable could it be the issue? tesla_api requires /mutable to store config, if it can't store it, it may fail. 
can you check if it's properly mounted

miles267, 2022-01-16:
Thanks @nelsongraca yes, confirming it's mounted.  I can navigate to and browse contents of /mutable.

marcone, 2022-01-16:
> > Look at the setup log, search for "tesla_api"
> 
> Thanks. I do see in the headless setup log it saying: Sun 26 Dec 18:20:08 CST 2021 : configure: tesla_api.py setup failed
> 
> Though I'm unable to locate info on why this might be?

It means that setup wasn't able to use the provided refresh token. Try ssh'ing into the Pi, then
```
sudo -i
bash -c bash notbash
/root/bin/tesla_api.py --debug list_vehicles
```

marcone, 2022-01-16:
> When I look inside /mutable/tesla_api.json file it has my email, id 0, vehicle_id 0 but my access_token is blank. No value between quotes. If that helps.

Delete that file and try again

miles267, 2022-01-16:
> > When I look inside /mutable/tesla_api.json file it has my email, id 0, vehicle_id 0 but my access_token is blank. No value between quotes. If that helps.
> 
> Delete that file and try again

While deleting tesla_api.json and rerunning setup did not work, this appeared to work:

sudo -i
bash -c bash notbash
/root/bin/tesla_api.py --debug list_vehicles

When I then went back and viewed the /mutable/tesla_api.json file, the contents were updated/populated.  Is this a bug with the teslausb-setup or the new tesla_api.py?

nelsongraca, 2022-01-16:
thanks for the info, I will try to replicate it

rmyadsk, 2022-01-16:
I have two vehicles under my account; I've updated the .conf file on both of my teslausb installations. The "bash -c bash notbash" procedure worked for me on the first vehicle I executed this on.

On the second vehicle, however, I encounter this error:

> root@teslausb-blue:~# /root/bin/tesla_api.py --debug list_vehicles
> Sun 16 Jan 15:25:28 EDT 2022: Mutable data didn't exist, writing defaults...
> Sun 16 Jan 15:25:28 EDT 2022: Writing /mutable/tesla_api.json...
> Sun 16 Jan 15:25:28 EDT 2022: Writing /mutable/tesla_api.json...
> Sun 16 Jan 15:25:28 EDT 2022: Force setting a refresh token
> Traceback (most recent call last):
>   File "/root/bin/tesla_api.py", line 587, in <module>
>     main()
>   File "/root/bin/tesla_api.py", line 563, in main
>     _get_id()
>   File "/root/bin/tesla_api.py", line 176, in _get_id
>     result = list_vehicles()
>   File "/root/bin/tesla_api.py", line 260, in list_vehicles
>     return _execute_request(base_url, None, None, False)
>   File "/root/bin/tesla_api.py", line 72, in _execute_request
>     json_response = _rest_request(url, method, data)
>   File "/root/bin/tesla_api.py", line 99, in _rest_request
>     'Authorization': 'Bearer {}'.format(_get_api_token()),
>   File "/root/bin/tesla_api.py", line 154, in _get_api_token
>     tesla.refresh_token()
> TypeError: refresh_token() missing 1 required positional argument: 'token_url'

refresh_token is written to tesla_api.json, however a null access_token is written. id and vehicle_id are both 0. Attempting to back-in the access_token value from the first vehicle into tesla_api.json value instead results in a different error:

> root@teslausb-blue:/mutable# /root/bin/tesla_api.py --debug list_vehicles
> Sun 16 Jan 15:46:22 EDT 2022: Loading mutable data from disk...
> Traceback (most recent call last):
>   File "/root/bin/tesla_api.py", line 587, in <module>
>     main()
>   File "/root/bin/tesla_api.py", line 563, in main
>     _get_id()
>   File "/root/bin/tesla_api.py", line 176, in _get_id
>     result = list_vehicles()
>   File "/root/bin/tesla_api.py", line 260, in list_vehicles
>     return _execute_request(base_url, None, None, False)
>   File "/root/bin/tesla_api.py", line 72, in _execute_request
>     json_response = _rest_request(url, method, data)
>   File "/root/bin/tesla_api.py", line 99, in _rest_request
>     'Authorization': 'Bearer {}'.format(_get_api_token()),
>   File "/root/bin/tesla_api.py", line 140, in _get_api_token
>     if SETTINGS['REFRESH_TOKEN'] or 0 < tesla.expires_at < time.time():
>   File "/usr/local/lib/python3.7/dist-packages/teslapy/__init__.py", line 97, in expires_at
>     return self.token['expires_at']
> KeyError: 'expires_at'

I wonder, has the key expired after one use?

rmyadsk, 2022-01-17:
Hey, can we open this issue?

rmyadsk, 2022-01-17:
Okay, gals and guys, I petition we reopen this issue until https://github.com/marcone/teslausb/issues/688#issuecomment-1013949934 is resolved.

miles267, 2022-01-17:
@rmyadsk it appears they're working on it under #714

nelsongraca, 2022-01-17:
@rmyadsk I will leave that decision to @marcone I could not look into it yet I expect to in the next days.

@miles267 that PR is merged.

miles267, 2022-01-17:
> @rmyadsk I will leave that decision to @marcone I could not look into it yet I expect to in the next days.
> 
> @miles267 that PR is merged.

Thanks @nelsongraca.  Does that mean if I run the teslausb-setup again now, I'll get a bugfix for this already in main-dev branch?

marcone, 2022-01-17:
>> KeyError: 'expires_at'
>
> I wonder, has the key expired after one use?

No, I think this is because you manually copied the access token from one Pi to another, but didn't create/update the "cache.json" file that TeslaPy uses.
Also note that using the refresh token to get a new token invalidates any old tokens created with that refresh token, so if you have two cars you should generate a separate refresh token for each.

rmyadsk, 2022-01-17:
Thanks everyone - I'm happy to be a testbed for any proposed changes - standing by.

JustinGuiao, 2022-01-24:
I know this issue is closed but I don't think this question deserves a new issue. I am running the newest TeslaUSB with the refresh token update and new teslausb_setup_variables.conf off a clean install and it works fine. My issue is, after about a day the tesla api auth breaks and I get "failed to contact car, retrying" repeating in my logs. As soon as I ssh into the device and delete tesla_api.json, it works again immediately and for another day before the issue returns. Any idea why this is happening, or if I could script the periodic deletion of tesla_api.json as a workaround? I'm not new to programming but I'm not sure how I would modify that myself considering the system is mounted read only be default.

marcone, 2022-01-25:
@JustinGuiao are you sure you're not using the same refresh token for multiple cars, or perhaps for one car plus an app or something?
Using a refresh token to obtain a new token will invalidate previous tokens issues by that refresh token, so the behavior you describe could be caused by something doing that.

rmyadsk, 2022-01-25:
I don't know, I'd offer this issue deserves to be re-opened or a new issue started. I have two vehicles under one account and have asked (as above) for a solution to make them both work under the new coding. As mentioned a week ago, I am an a ready, willing, and eager testbed.

rmyadsk, 2022-01-26:
ping +24 hours for owners of multiple vehicles under one account.

marcone, 2022-01-26:
Multiple vehicles under one account should not be a problem, you just can't use the same refresh token for multiple cars. Generate a separate refresh token for each car/Pi.

rmyadsk, 2022-01-27:
Thanks @marcone - I loaded a new refresh_token from the Andorid app into the .conf file, then ran a `./setup-teslausb upgrade` then rebooted, then ran the below; it appears something is still unhappy:

```
pi@teslausb-blue:~ $ sudo -i
Run 'bin/setup-teslausb upgrade' to update to the latest version of teslausb,
or run 'bin/remountfs_rw' to allow writing to the root partition.

root@teslausb-blue:~# bash -c bash notbash
/root/bin/setup-teslausb was sourced, setting setup variables
Run 'bin/setup-teslausb upgrade' to update to the latest version of teslausb,
or run 'bin/remountfs_rw' to allow writing to the root partition.

root@teslausb-blue:~# /root/bin/tesla_api.py --debug list_vehicles
Wed 26 Jan 22:00:26 EDT 2022: Loading mutable data from disk...
Traceback (most recent call last):
  File "/root/bin/tesla_api.py", line 587, in <module>
    main()
  File "/root/bin/tesla_api.py", line 563, in main
    _get_id()
  File "/root/bin/tesla_api.py", line 176, in _get_id
    result = list_vehicles()
  File "/root/bin/tesla_api.py", line 260, in list_vehicles
    return _execute_request(base_url, None, None, False)
  File "/root/bin/tesla_api.py", line 72, in _execute_request
    json_response = _rest_request(url, method, data)
  File "/root/bin/tesla_api.py", line 99, in _rest_request
    'Authorization': 'Bearer {}'.format(_get_api_token()),
  File "/root/bin/tesla_api.py", line 140, in _get_api_token
    if SETTINGS['REFRESH_TOKEN'] or 0 < tesla.expires_at < time.time():
  File "/usr/local/lib/python3.7/dist-packages/teslapy/__init__.py", line 97, in expires_at
    return self.token['expires_at']
KeyError: 'expires_at'
root@teslausb-blue:~# 
```

marcone, 2022-01-27:
Can you run `pip3 list` on the Pi and see what version the TeslaPy package is?

rmyadsk, 2022-01-27:
```
pi@teslausb-blue:~ $ pip3 list
Package           Version  
----------------- ---------
asn1crypto        0.24.0   
certifi           2018.8.24
chardet           3.0.4    
colorzero         1.1      
cryptography      2.6.1    
entrypoints       0.3      
gpiozero          1.6.2    
idna              2.6      
keyring           17.1.1   
keyrings.alt      3.1.1    
oauthlib          3.1.1    
pip               18.1     
pycrypto          2.6.1    
PyGObject         3.30.4   
python-apt        1.8.4.3  
pyxdg             0.25     
requests          2.21.0   
requests-oauthlib 1.3.0    
RPi.GPIO          0.7.0    
SecretStorage     2.3.1    
setuptools        40.8.0   
six               1.12.0   
spidev            3.5      
ssh-import-id     5.7      
TeslaPy           2.1.0    
urllib3           1.24.1   
websocket-client  1.2.1    
wheel             0.32.3   
pi@teslausb-blue:~ $ 
```

marcone, 2022-01-27:
That's an old version than what I have. Can you try `pip3 install teslapy` ?

rmyadsk, 2022-01-27:
Yeah, it does not want to write. Is it time for me to just re-image?

```
# pip3 list
Package           Version  
----------------- ---------
asn1crypto        0.24.0   
certifi           2018.8.24
chardet           3.0.4    
colorzero         1.1      
cryptography      2.6.1    
entrypoints       0.3      
gpiozero          1.6.2    
idna              2.6      
keyring           17.1.1   
keyrings.alt      3.1.1    
oauthlib          3.1.1    
pip               18.1     
pycrypto          2.6.1    
PyGObject         3.30.4   
python-apt        1.8.4.3  
pyxdg             0.25     
requests          2.21.0   
requests-oauthlib 1.3.0    
RPi.GPIO          0.7.0    
SecretStorage     2.3.1    
setuptools        40.8.0   
six               1.12.0   
spidev            3.5      
ssh-import-id     5.7      
TeslaPy           2.1.0    
urllib3           1.24.1   
websocket-client  1.2.1    
wheel             0.32.3   
# sudo -i
Run 'bin/setup-teslausb upgrade' to update to the latest version of teslausb,
or run 'bin/remountfs_rw' to allow writing to the root partition.

root@teslausb-blue:~# pip3 install teslapy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: teslapy in /usr/local/lib/python3.7/dist-packages (2.1.0)
Requirement already satisfied: requests>=2.4.2 in /usr/lib/python3/dist-packages (from teslapy) (2.21.0)
Requirement already satisfied: websocket-client>=0.59.0 in /usr/local/lib/python3.7/dist-packages (from teslapy) (1.2.1)
Requirement already satisfied: requests-oauthlib in /usr/local/lib/python3.7/dist-packages (from teslapy) (1.3.0)
Requirement already satisfied: oauthlib>=3.0.0 in /usr/local/lib/python3.7/dist-packages (from requests-oauthlib->teslapy) (3.1.1)
root@teslausb-blue:~# pip3 list
Package           Version  
----------------- ---------
asn1crypto        0.24.0   
certifi           2018.8.24
chardet           3.0.4    
colorzero         1.1      
cryptography      2.6.1    
entrypoints       0.3      
gpiozero          1.6.2    
idna              2.6      
keyring           17.1.1   
keyrings.alt      3.1.1    
oauthlib          3.1.1    
pip               18.1     
pycrypto          2.6.1    
PyGObject         3.30.4   
python-apt        1.8.4.3  
pyxdg             0.25     
requests          2.21.0   
requests-oauthlib 1.3.0    
RPi.GPIO          0.7.0    
SecretStorage     2.3.1    
setuptools        40.8.0   
six               1.12.0   
spidev            3.5      
ssh-import-id     5.7      
TeslaPy           2.1.0    
urllib3           1.24.1   
websocket-client  1.2.1    
wheel             0.32.3   
root@teslausb-blue:~# 
```

marcone, 2022-01-27:
I'm not familiar enough with the python package manager to say whether that would fix the problem. My TeslaPy version is 2.3.0. Maybe `pip3 install -U teslapy` would force an upgrade?

rmyadsk, 2022-01-27:
That seemed to work, for a second:

```
pi@teslausb-blue:~ $ pip3 install -U teslapy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting teslapy
  Downloading https://files.pythonhosted.org/packages/21/23/d5bcf6b45850376f8603fba663fa716ed5866e71e3c160f1d6d6403fcf91/TeslaPy-2.3.0-py2.py3-none-any.whl
Requirement already satisfied, skipping upgrade: requests>=2.4.2 in /usr/lib/python3/dist-packages (from teslapy) (2.21.0)
Requirement already satisfied, skipping upgrade: websocket-client>=0.59.0 in /usr/local/lib/python3.7/dist-packages (from teslapy) (1.2.1)
Requirement already satisfied, skipping upgrade: requests-oauthlib in /usr/local/lib/python3.7/dist-packages (from teslapy) (1.3.0)
Requirement already satisfied, skipping upgrade: oauthlib>=3.0.0 in /usr/local/lib/python3.7/dist-packages (from requests-oauthlib->teslapy) (3.1.1)
Installing collected packages: teslapy
Successfully installed teslapy-2.3.0
```

Until I tested it and it failed again:

```
pi@teslausb-blue:~ $ sudo -i
Run 'bin/setup-teslausb upgrade' to update to the latest version of teslausb,
or run 'bin/remountfs_rw' to allow writing to the root partition.

root@teslausb-blue:~# bash -c bash notbash
/root/bin/setup-teslausb was sourced, setting setup variables
Run 'bin/setup-teslausb upgrade' to update to the latest version of teslausb,
or run 'bin/remountfs_rw' to allow writing to the root partition.

root@teslausb-blue:~# /root/bin/tesla_api.py --debug list_vehicles
Wed 26 Jan 23:08:43 EDT 2022: Loading mutable data from disk...
Traceback (most recent call last):
  File "/root/bin/tesla_api.py", line 587, in <module>
    main()
  File "/root/bin/tesla_api.py", line 563, in main
    _get_id()
  File "/root/bin/tesla_api.py", line 176, in _get_id
    result = list_vehicles()
  File "/root/bin/tesla_api.py", line 260, in list_vehicles
    return _execute_request(base_url, None, None, False)
  File "/root/bin/tesla_api.py", line 72, in _execute_request
    json_response = _rest_request(url, method, data)
  File "/root/bin/tesla_api.py", line 99, in _rest_request
    'Authorization': 'Bearer {}'.format(_get_api_token()),
  File "/root/bin/tesla_api.py", line 140, in _get_api_token
    if SETTINGS['REFRESH_TOKEN'] or 0 < tesla.expires_at < time.time():
  File "/usr/local/lib/python3.7/dist-packages/teslapy/__init__.py", line 97, in expires_at
    return self.token['expires_at']
KeyError: 'expires_at'
root@teslausb-blue:~# 

```

And the coup de grâce following this:

```
root@teslausb-blue:~# pip3 list
Package           Version  
----------------- ---------
asn1crypto        0.24.0   
certifi           2018.8.24
chardet           3.0.4    
colorzero         1.1      
cryptography      2.6.1    
entrypoints       0.3      
gpiozero          1.6.2    
idna              2.6      
keyring           17.1.1   
keyrings.alt      3.1.1    
oauthlib          3.1.1    
pip               18.1     
pycrypto          2.6.1    
PyGObject         3.30.4   
python-apt        1.8.4.3  
pyxdg             0.25     
requests          2.21.0   
requests-oauthlib 1.3.0    
RPi.GPIO          0.7.0    
SecretStorage     2.3.1    
setuptools        40.8.0   
six               1.12.0   
spidev            3.5      
ssh-import-id     5.7      
TeslaPy           2.1.0    
urllib3           1.24.1   
websocket-client  1.2.1    
wheel             0.32.3   
root@teslausb-blue:~# 
```

rmyadsk, 2022-01-27:
As it turns out, a very explicit `pip3 uninstall TeslaPy` followed by a reboot then an `pip3 install TeslaPy` works for my install - will test for usability in the coming days.

marcone, 2022-01-27:
My guess is that since you ran the original `pip3 install -U teslapy` as the "pi" user, it installed it in the pi user's home directory, where root can't find it.