marcone/teslausb #695: tried to do update to latest version of teslausb

aaronac8, 2021-11-04:
I tired to update to latest version of teslausb and had this error:
[image]

aaronac8, 2021-11-04:
Also, my tesla cam files are NOT downloading to my PC now.  See diagnostics below.
[diagnostics.txt](https://github.com/marcone/teslausb/files/7474382/diagnostics.txt)

marcone, 2021-11-04:
hmm, upgrading from an older version works for me. What happens if you run `setup-teslausb upgrade` again?

aaronac8, 2021-11-04:
I tried it again, I see the following:
[image]

marcone, 2021-11-04:
The clock isn't set correctly, which is where those "not valid yet" messages come from.
Try running `sntp -S time.google.com` before running the upgrade command.

aaronac8, 2021-11-05:
I ran the clock set up command, and now get this error:
[KSErNl7.png (1263×961).pdf](https://github.com/marcone/teslausb/files/7482345/KSErNl7.png.1263x961.pdf)

aaronac8, 2021-11-05:
Also, I cannot upload my saved sentry clips to my PC.  This function worked fine before and now stopped.  That is why I wanted to try an upgrade.  Thanks for your help

marcone, 2021-11-05:
I think you might need to update linux first, e.g. do:
```
sudo -i
rm -rf /lib/modules/!($(uname -r)) # this frees up space by deleting unused kernel modules, not needed if you have sufficient space already
apt update
apt upgrade
```
Then reboot and try updating teslausb again.

aaronac8, 2021-11-05:
I did the updating:
[image]

I am still unable to have the saved sentry clips sent to my PC
[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/7489286/teslausb-headless-setup.log)
.

aaronac8, 2021-11-05:
here is my archiveloop.log
[archiveloop.log](https://github.com/marcone/teslausb/files/7489291/archiveloop.log)
:

marcone, 2021-11-05:
It looks you either didn't run all the commands I listed earlier, or they failed, since you're still at kernel 4.19.93.

marcone, 2021-11-06:
I was able to reproduce this. No idea what's going on yet. It works with the latest image, but not when upgrading from the 202000709 image.

marcone, 2021-11-06:
ok, I see: with the latest image, /sbin is a symbolic link to /usr/sbin, whereas with the older image(s), /sbin and /usr/sbin were separate folders.

closed 2021-11-06 by marcone

marcone, 2021-11-06:
If you run `setup-teslausb upgrade` again, it should work. Thanks for reporting the issue.

aaronac8, 2021-11-06:
Still getting STOP: Couldn't check image error
[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/7489976/teslausb-headless-setup.log)

aaronac8, 2021-11-06:
updated linux, but still getting error
[teslausb-headless-setup (1).log](https://github.com/marcone/teslausb/files/7490005/teslausb-headless-setup.1.log)

aaronac8, 2021-11-06:
I figured out the problem with my upgrade.....my PC IP address changed, so that is why I could not download my sentry clips.  Thank you.