marcone/teslausb #700: feature request: add music to the pi via SMB

czyz, 2021-11-11:
Hi, 

I've got my pi set to archive recordings to my computer via rsync, not CIF, so do not have the various SMB authentication and server options set in the configuration. I think this means I can't use the feature by which the pi will automatically sync over music into the music disk image on the pi.

But the TeslaCam image is shared over the network via smb, albeit in read-only mode. I looked at /etc/samba/smb.conf in the hopes that maybe I could share the Music image in a writeable form and then I'd just be able to copy files to it over the network. But it's sharing Teslacam from a location in /mutable and I see this mentioned in /root/bin/archiveloop but am not certain of how it works and thus don't know how to set up the Music image to be shared in this manner. But if it's technically possible to do so I'd love to not have to unplug the pi from my car and carry it in to the computer just to transfer music to it.

rmyadsk, 2021-11-11:
I think the target verb here would be "sync" rather than "add", but yes.

Also, to take it a step further, how about making the music slice rsync-able too?

marcone, 2021-11-11:
Someone started implementing music sync for rsync in https://github.com/marcone/teslausb/pull/641, but it was never finished.

Exposing music as a writable Samba share is tricky because in order to do it safely the drive needs to be removed from the car while it is being accessed by samba. This adds complexity that I don't think is worth it.

closed 2021-11-11 by marcone