marcone/teslausb #701: Rsync is not cleaning up old data

usuryadevara, 2021-11-15:
@marcone any suggestion on what needs to be fixed here? https://pastebin.com/xdn1thh8 

I have 128GB SD card and looks like it stops working after it hit 90%? 
1) Any recommendations on how to clean the old images without loosing the recordings? 
2) Currently this value is set to 100% and no music_size( export CAM_SIZE=100%), and the disk has Sentry and Recordings for more than a month i guess.

`root@teslausb:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       2.0G  1.3G  600M  68% /
devtmpfs        183M     0  183M   0% /dev
tmpfs           216M     0  216M   0% /dev/shm
tmpfs           216M   17M  199M   8% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           216M     0  216M   0% /sys/fs/cgroup
tmpfs           216M     0  216M   0% /var/tmp
tmpfs           216M  2.0M  214M   1% /tmp
tmpfs           216M     0  216M   0% /var/log/nginx
tmpfs           216M     0  216M   0% /var/lib/nginx
tmpfs           216M     0  216M   0% /var/spool
tmpfs           216M  948K  215M   1% /run/samba
tmpfs           216M  1.3M  214M   1% /var/cache/samba
/dev/mmcblk0p1  247M   49M  199M  20% /boot
/dev/mmcblk0p3  117G  107G  9.9G  92% /backingfiles
/dev/mmcblk0p4   95M   15M   74M  17% /mutable
/dev/loop1      106G   88G   19G  83% /tmp/snapshots/snap-000413
/dev/loop0      106G   88G   19G  83% /mnt/cam
overlay         216M  2.0M  214M   1% /tmp/cam/merged
tmpfs            44M     0   44M   0% /run/user/1000`

`root@teslausb:~# sudo /root/bin/force_sync.sh
Forcing archiveloop to sync by pretending to take archive host offline and back online.
Setting archive unreachable..
Waiting for archiveloop to see canary file..`

marcone, 2021-11-15:
What do you think is not working? The only thing I see in the log you posted is that the car apparently goes to sleep while archiving is in progress, so it never completes a full archive cycle.

usuryadevara, 2021-11-15:
Thanks for quick reply @marcone. I get USB too slow on the device on Tesla Screen so, i removed it from car after the car went to sleep and connected to computer to login and check the logs.

What should i do to get the data archived to backup location with Rsync?

marcone, 2021-11-16:
you could leave it connected to the computer (or other power source) until backup finishes. When plugged into the car, you could turn on sentry or climate while it's archiving. Unfortunately the mechanism by which teslausb itself can keep the car awake is currently not working.
There's a long discussion about "usb too slow" here: https://github.com/marcone/teslausb/issues/239

closed 2021-11-16 by marcone

usuryadevara, 2021-11-24:
@marcone any easy way to clear all data from the Raspberry pi? It takes forever to copy the data and now its 92% full. 

these files sentry_files_archived, sentry_files has files from couple months, even though i see the data copied using rsync, i think its not clearing the copied data.

marcone, 2021-11-24:
Looks like you specified a relatively large `CAM_SIZE` in your config. Teslausb will try to keep 20GB of free space on the drive image. It doesn't delete something just because it's been archived already. You can certainly delete the files manually if you want. You can do this either from the Pi itself, from the car, or by attaching to a PC and accessing the "CAM" drive.

usuryadevara, 2021-11-24:
Thanks for your quick response @marcone, i attached the pi to the computer to delete the files but it cannot mount to remove. Is there any best method to remove the files from ssh into the raspberry pi?

marcone, 2021-11-24:
> i attached the pi to the computer to delete the files but it cannot mount to remove

oh right, because it's still archiving, and the drive is not published during archiving.
In that case, just ssh into the Pi like you did before, `cd /mnt/cam/TeslaCam` and then delete what you want.