marcone/teslausb #702: Can't figure out why sentry and recorded clips are not getting archieved

balajeek, 2021-12-06:
Nice and cool project, thanks for making it easy for us.  great job.

I setup this project on raspberry pi zero 2w, all was well, plugged in and made some sentry sessions, return home to connect to wifi and it copied the sentry clips, all are awesome.  no recorded clips or saved clips though, next few days my model 3 keep saying "USB drive is not accessible by dashcam", after uplug/plug it works then an hour later would get the same message, sometime it also says "USB drive filesystem is corrupted" or "too slow" messages.

So to tackle that i thought i should first replace the sd card to a better one (I was using the sd card came with my BlackVue daskcam), I ordered a top performance samsung 128gb pro and re-did the project again.  This time i also uncommented the following so i can save recent clips in addition to sentry and saved.
```
export ARCHIVE_RECENTCLIPS=true
export CAM_SIZE=100%
```
When for test drive, made sure sentry and saved are created. Then return to wifi connect at home. Now the USB on tesla seem reliable and no issues.

It did archive Recent clips, great.  But no sentry clips.  Did a try again for another round of videos and still not sentry clips.  Its been few days and many tries and no luck.

I opened the web ui, i see all the folders and videos of sentry clips but for some reason its not archiving.

Then i opened the config and set also the the following:  but still it does not copy, all i get is Recent Clips.
```
export ARCHIVE_SAVEDCLIPS=true
export ARCHIVE_SENTRYCLIPS=true
```
On the final note, I was under impression that once the files are copied/archived it should delete form the sdcard on raspberry pi, but its not, all the recent clips are still in the usb sdcard.

Here are the relevant/recent archive log file:
```
Sun  5 Dec 22:17:20 EST 2021: Trying to set time...
Sun  5 Dec 22:35:40 EST 2021: Time adjusted by 1100.527180 seconds after 0.130000 seconds
Sun  5 Dec 22:35:40 EST 2021: not keeping car awake.
Sun  5 Dec 22:35:40 EST 2021: Archiving...
Sun  5 Dec 22:35:40 EST 2021: Ensuring cam archive is mounted...
Sun  5 Dec 22:35:40 EST 2021: Mounting /mnt/archive...
Sun  5 Dec 22:35:41 EST 2021: Mounted /mnt/archive.
Sun  5 Dec 22:35:41 EST 2021: Ensured cam archive is mounted.
Sun  5 Dec 22:35:41 EST 2021: Checking saved folder count...
Sun  5 Dec 22:35:41 EST 2021: Ensuring cam file is mounted...
Sun  5 Dec 22:35:41 EST 2021: Mounting /mnt/cam...
Sun  5 Dec 22:35:41 EST 2021: Mounted /mnt/cam.
Sun  5 Dec 22:35:41 EST 2021: Ensured cam file is mounted.
Sun  5 Dec 22:35:43 EST 2021: There are 61 event folder(s) with 586 file(s) and 0 track mode file(s) to move.  0 short recording(s) will be skipped.
Sun  5 Dec 22:35:43 EST 2021: Starting recording archiving
Sun  5 Dec 22:35:43 EST 2021: Archiving 61 event folder(s) with 586 file(s) starting at Sun  5 Dec 22:35:43 EST 2021
Sun  5 Dec 23:33:37 EST 2021: waiting up to 90 seconds for idle interval
Sun  5 Dec 23:33:38 EST 2021: mass storage process not active, OK to write
Sun  5 Dec 23:33:39 EST 2021: taking snapshot of cam disk in /backingfiles/snapshots/snap-000078
Sun  5 Dec 23:33:43 EST 2021: took snapshot
Sun  5 Dec 23:33:44 EST 2021: comparing new snapshot with /backingfiles/snapshots/snap-000077/snap.bin
Sun  5 Dec 23:33:44 EST 2021: new snapshot is identical to previous one, discarding
Mon  6 Dec 00:31:46 EST 2021: waiting up to 90 seconds for idle interval
```

Let me know if you need any more info.  Thank you.

marcone, 2021-12-06:
I think it's just taking a really long time to archive (the Pi's wifi isn't fast, especially when the signal is weak). You should be seeing _some_ files show up on your server, since it was archiving for almost 2 hours in the log you showed. If you ssh into the Pi while it's archiving, you can check the rsync logs in /tmp to see what it's doing.

balajeek, 2021-12-06:
Thanks for the quick response, if you think its taking too much time (or i did not have car on enough to copy), I can try tonight when sentry mode or climate on to see if it can finish copying and determine how long it took.  Would there be any log entry that shows that archive is complete ?

Also the rsync log would only be there while copying, i have the sdcard mounted on mac now and looking at the files, wondering if there would be any files to look for.  currently /tmp does not seem to have any such file of rsync.

marcone, 2021-12-06:
> Would there be any log entry that shows that archive is complete ?

Yes, it will log whether archiving was successful or not, and how many files it archived.

> currently /tmp does not seem to have any such file of rsync

Right, /tmp is a ramdisk when teslausb is booted.

balajeek, 2021-12-13:
as I mentioned i tried to put sentry on the whole night on Dec 6th to check if i can copy all the files.
Here is the log continued from the last i log i posted.

```
Mon  6 Dec 00:31:46 EST 2021: waiting up to 90 seconds for idle interval
Mon  6 Dec 00:31:47 EST 2021: mass storage process not active, OK to write
Mon  6 Dec 00:31:47 EST 2021: taking snapshot of cam disk in /backingfiles/snapshots/snap-000078
Mon  6 Dec 00:31:52 EST 2021: took snapshot
Mon  6 Dec 00:31:53 EST 2021: comparing new snapshot with /backingfiles/snapshots/snap-000077/snap.bin
Mon  6 Dec 00:31:53 EST 2021: new snapshot is identical to previous one, discarding
Mon  6 Dec 00:33:52 EST 2021: connection dead, killing archive-clips
Mon  6 Dec 00:33:53 EST 2021: Error during archiving. Archived 375 event files in 1h58m10s
2021/12/05 22:35:45 [798] building file list
2021/12/05 22:35:46 [798] done
2021/12/05 22:35:48 [798] >f+++++++++ RecentClips/2021-12-04/2021-12-04_13-47-42-back.mp4
2021/12/05 22:36:23 [798] >f+++++++++ RecentClips/2021-12-04/2021-12-04_13-47-42-front.mp4
2021/12/05 22:36:58 [798] >f+++++++++ RecentClips/2021-12-04/2021-12-04_13-47-42-left_repeater.mp4
    ...more of this lines (cut to simplify)
2021/12/05 23:17:46 [798] cd+++++++++ SavedClips/
2021/12/05 23:17:46 [798] cd+++++++++ SavedClips/2021-11-30_19-06-43/
2021/12/05 23:18:08 [798] >f+++++++++ SavedClips/2021-11-30_19-06-43/2021-11-30_18-56-01-back.mp4
    ...more of this lines (cut to simplify)
2021/12/05 23:53:04 [798] cd+++++++++ SentryClips/2021-11-30_12-34-32/
2021/12/05 23:53:11 [798] >f+++++++++ SentryClips/2021-11-30_12-34-32/2021-11-30_12-24-00-back.mp4
2021/12/05 23:53:32 [798] >f+++++++++ SentryClips/2021-11-30_12-34-32/2021-11-30_12-24-00-front.mp4
2021/12/05 23:53:52 [798] >f+++++++++ SentryClips/2021-11-30_12-34-32/2021-11-30_12-24-00-left_repeater.mp4
    ...more of this lines (cut to simplify)
2021/12/06 00:33:53 [798] sent 10439188739 bytes  received 10739 bytes  total size 16494899338
2021/12/06 00:33:53 [798] rsync error: received SIGINT, SIGTERM, or SIGHUP (code 20) at rsync.c(701) [sender=v3.2.3]
building file list ... done
rsync error: received SIGINT, SIGTERM, or SIGHUP (code 20) at io.c(503) [generator=v3.2.3]
rsync error: received SIGINT, SIGTERM, or SIGHUP (code 20) at rsync.c(701) [sender=v3.2.3]
Mon  6 Dec 00:33:53 EST 2021: cleaning cam mount
Mon  6 Dec 00:33:55 EST 2021: done cleaning cam mount
Mon  6 Dec 00:33:55 EST 2021: Trimming free space in /mnt/cam, which has 584 extents
Mon  6 Dec 00:33:55 EST 2021: Trim complete, image now has 489 extents
Mon  6 Dec 00:33:55 EST 2021: Unmounting /mnt/cam...
Mon  6 Dec 00:33:55 EST 2021: Unmounted /mnt/cam.
Mon  6 Dec 00:33:55 EST 2021: Finished archiving.
Mon  6 Dec 00:33:55 EST 2021: Music archive not configured or unreachable
Mon  6 Dec 00:33:55 EST 2021: unmounting /mnt/archive
Mon  6 Dec 00:33:55 EST 2021: Connecting usb to host...
Mon  6 Dec 00:33:55 EST 2021: Connected usb to host.
Mon  6 Dec 00:33:56 EST 2021: Waiting for archive to be unreachable...
Mon  6 Dec 00:33:56 EST 2021: Sleeping before retry...
Mon  6 Dec 00:33:57 EST 2021: Retrying...
Mon  6 Dec 00:33:57 EST 2021: Sleeping before retry...
Mon  6 Dec 00:33:58 EST 2021: Retrying...
Mon  6 Dec 00:33:58 EST 2021: Sleeping before retry...
Mon  6 Dec 00:33:59 EST 2021: Retrying...
Mon  6 Dec 00:33:59 EST 2021: Sleeping before retry...
Mon  6 Dec 00:34:00 EST 2021: Retrying...
Mon  6 Dec 01:29:55 EST 2021: waiting up to 90 seconds for idle interval
Mon  6 Dec 01:31:31 EST 2021: couldn't determine idle interval
Mon  6 Dec 01:31:31 EST 2021: taking snapshot of cam disk in /backingfiles/snapshots/snap-000078
Mon  6 Dec 01:31:34 EST 2021: took snapshot
Mon  6 Dec 01:31:36 EST 2021: comparing new snapshot with /backingfiles/snapshots/snap-000077/snap.bin
Mon  6 Dec 01:31:36 EST 2021: new snapshot is identical to previous one, discarding
Mon  6 Dec 02:29:37 EST 2021: waiting up to 90 seconds for idle interval
==============================================

seem like it took a really really long to copy.
so i unpluged the sdcard from pi and mounted into my mac and deleted all the folders of recordings under "Recent","Saved","Sentry". and then plugged back in to if it can handle a single day files to be copied.

Then i ended up in a situation where telsa says "USB drive malfunction, please remove it"
I removed it for few min and put it back, it stays on for sometime and then gives the same message.

What can i do now?
I got new pi zero 2W just for this and also got a good samsung 128gb sdcard.  Whats wrong with it now, should i wipe out and re-try again?

marcone, 2021-12-14:
> seem like it took a really really long to copy.

Looks like about 20 seconds per file. Your wifi may be a bit weak where you park your car. It's about twice as fast for me, but I put a small AP in my garage.

Then there's this:
```
Mon  6 Dec 00:33:52 EST 2021: connection dead, killing archive-clips
```
which basically means that teslausb couldn't ping your archive server for a while (possibly because the relatively poor connection was saturated with rsync traffic) and concluded that you moved away from wifi, which makes it stop the archive operation.

> should i wipe out and re-try again

That seems like a good idea. I would also suggest not setting CAM_SIZE to 100%.

balajeek, 2021-12-14:
Car is parked inside the garage and I have a 2nd wifi apple router hard wire mounted in the ceiling right above the car , i have outdoor ip cameras connect thru this. and it has good speed. 

I will do a fresh start again tonight. 
Hoping the usb malfunction error goes away.

closed 2021-12-25 by marcone

FA9US, 2022-01-03:
> Car is parked inside the garage and I have a 2nd wifi apple router hard wire mounted in the ceiling right above the car , i have outdoor ip cameras connect thru this. and it has good speed.
> 
> I will do a fresh start again tonight. Hoping the usb malfunction error goes away.

the pushover notification will also really help. i have it configured in the Teslausb and it notifies when the copy starts. i also have PLEX which knows about my TeslaUSB cam NAS folder and tautulli with pushover aswell. it will in realtime send me a notification about each recording with even an snapshot of the video thanks to tautulli.

it can sometimes be annoying if you have too many videos. i will soon look into mixing them and then pushing to Plex.

check here:
[image]
.