marcone/teslausb #703: Amazon SNS and Slack notifications aren't sending (or they're lost in the ether...)

wpattison74, 2021-12-09:
Hello, I was trying to setup Amazon SNS to send me an email notification, and was having difficulty getting that to work. As an alternative, I setup the webhook for Slack notifications, and that isn't working, either.

Log file shows entries stating "Sending SNS message". I don't see any entries related to the Slack notifications.

Here's my diagnostics output:  https://pastebin.com/raw/XWxG7jPD

I was trying to find a way to fire off a test message from the command line, but didn't see anything obvious.

Note - I'm running the latest update of the Pi Zero W 2 Rev 1.0.

Tonight I "rediscovered" that there's a web interface for TeslaUSB, I had forgotten about that. Perhaps there could be a way to surface a button or two for on-demand notification testing?

wpattison74, 2021-12-09:
Hi - thank you for the response.  Here's what I am getting:

```
root@teslausb:~# bin/send-push-message TestTitle TestMessage
TestMessage
Sending SNS message.
Traceback (most recent call last):
  File "/root/bin/send_sns.py", line 16, in <module>
    send_sns(args.topic, args.subject, args.message)
  File "/root/bin/send_sns.py", line 6, in send_sns
    response = sns.publish(TopicArn=topic, Message=message, Subject=subject)
  File "/usr/local/lib/python3.7/dist-packages/botocore/client.py", line 391, in _api_call
    return self._make_api_call(operation_name, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/botocore/client.py", line 719, in _make_api_call
    raise error_class(parsed_response, operation_name)
botocore.errorfactory.InvalidParameterException: An error occurred (InvalidParameter) when calling the Publish operation: Invalid parameter: Topic Name
```

wpattison74, 2021-12-09:
Update - I setup a Pushover account to test notifications, and that works.  Still unclear on why Slack and SNS aren't.

marcone, 2021-12-09:
It's probably skipping Slack because the SNS failure causes the `send-push-message` script to exit.

SNS might be failing because you're not using the correct "ARN". You need to use the "topic ARN", not the target ARN or any other ARN.

wpattison74, 2021-12-09:
Apologies, I had the Subscription ARN rather than the Topic ARN specified. All is working now, including Slack.  Thank you for the help.

closed 2021-12-09 by wpattison74